Koenkk/zigbee2mqtt #13440: [New device support]: Ikea LED bulb E26 250 lumen, wireless dimmable warm white/globe clear (TRADFRIbulbE26WWglobeclear250lm)

adgenet, 2022-08-06:
### Link

https://www.ikea.com/us/en/p/tradfri-led-bulb-e26-250-lumen-wireless-dimmable-warm-white-globe-clear-70439257/

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0x84fd27fffe0b843c","nwkAddr":20843,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRIbulbE26WWglobeclear250lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRIbulbE26WWglobeclear250lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20200826","swBuildId":"1.0.010"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20200826","swBuildId":"1.0.010","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659825368564,"defaultSendRequestWhen":"immediate"}

### Comments

Device currently shown as unsupported

### Supported color modes

brightness only

adgenet, 2022-08-07:
Model stamp on the base of the bulb is "LED2008G3"

Koenkk, 2022-08-08:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

adgenet, 2022-08-09:
The light does not support changing color temperature and returns 'UNSUPPORTED_ATTRIBUTE'

Koenkk, 2022-08-09:
Ah sorry, added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-09 by Koenkk

slochewie, 2022-09-26:
Switching to Docker latest-dev worked for me. Thanks!