Koenkk/zigbee2mqtt #13442: Zigbee2MDTT Herdman error when starting

ALPF5555, 2022-08-07:
### Is your feature request related to a problem? Please describe

I have tried 6 different Sonoof 3.0 dongles and 5 different USB cables as well as the USB2.0 and USB 3.0 ports on the Raspberry Pi I have even tried 2 different Raspberry Pi and every time I get the same error all the time.. I am running bullseye on the Raspberry Pi
zigbee2MQTT:error 2022-08-07 06:09:11: Error while starting zigbee-herdsman.

I have used the Serial find and each time it says it is USB0 and gives me the Sonoff dongle 3.0 and the serial number etc 
I used CC1352P2_CC2652P_other_coordinator_20220219.hex to write to the dongles and every time it was a success but every time I get the same failure the herdsman error.
I have checked online and cannot find anything that I have not tried to correct the problem. I cannot imagine that I have 6 bad dongles and 5 bad cables or two bad Raspberry Pi 4 B 4GB.

Zigbee2MQTT:error 2022-08-07 06:09:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-07 06:09:11: Failed to start zigbee
Zigbee2MQTT:error 2022-08-07 06:09:11: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-07 06:09:11: Exiting...
Zigbee2MQTT:error 2022-08-07 06:09:11: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
 
When I search for the dongle I always find it and here is one example 
i@raspberrypi:~ $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Aug  7 06:03 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_60737ce38819ec1181b839cc47486eb0-if00-port0 -> ../../ttyUSB0
I get the same every time the serial number is different but always find the Sonoff Dongle 3.0 plus.

### Describe the solution you'd like

I would like to be able to start Zigbee and use the Zigbee2mqtt to find the devices I have around the house 

### Describe alternatives you've considered

You can see above I have tried many options and trying to eliminate possible problems caused by me but everything I have tried seems to have the same failure results.

### Additional context

Is there something I am doing wrong I am loading the Zigbee2MQTT Why does it always fail at the starting Herdsman

Koenkk, 2022-08-07:
> Zigbee2MQTT:error 2022-08-07 06:09:11: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

ALPF5555, 2022-08-07:
Hello Sir              Thank you so much for your e-mail and I see that the Network key is the same as a one I have already got in use.How do I change or Generate a new Network key or change the Pan Id.
I have tried to change the numbers in the Network Key but that did not work how do I generate a new Network Key so it is the correct number Please
Or How do I change the PanID I cannot even see it in the Yaml file 
I am so sorry to be so stupid in this area but I am trying to set up a number of units for different properties and have different zigbee networks at the properties.
Regards Alan Turnbull

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel—

Koenkk, 2022-08-07:
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html shows how to set the `pan_id`

ALPF5555, 2022-08-12:
Hello Sir              I am still fighting with this Zigbee2MQTT. I have managed to get it installed or at least I think so I have been able to follow your instructions and the Herdsman starts etc I have included all the extra notes and files that needed to be added.I have even got to the end where I can see the Front End and it seems to be working but I cannot join any devices to the network.I have tried a number of new Aqara temperature and humidity sensors but none will join the network.
I have looked through the settings but cannot find anything that stand out as in correct.I have checked the Zigbee Config file and it is permit join true.
Can you help me or give me a clue as to why I cannot get a device to join the network?
Regards Alan Turnbull

Koenkk, 2022-08-13:
Check the points from https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair

ALPF5555, 2022-08-15:
Hello Sir              I want to take this opportunity for the help you have provided during my long struggle to get Zigbee2mqtt working. It was a struggle but a large learning process along the way and the help you gave me helped to keep me on a track where eventually all the hard work paid off.
Once again Thank you Have a great day 
All the best regards 
Alan Turnbull

closed 2022-08-16 by Koenkk