Koenkk/zigbee2mqtt #13443: Tuya TS601 _TZE200_locansqn Temperature and humidity sensor with LCD

esackbauer, 2022-08-07:
### Link

https://de.aliexpress.com/item/1005003634353180.html

### Database entry

{"id":16,"type":"EndDevice","ieeeAddr":"0xa4c138c26d8dfb58","nwkAddr":11532,"manufId":4417,"manufName":"_TZE200_locansqn","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":1,"appVersion":70,"modelId":"TS0601","manufacturerName":"_TZE200_locansqn","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0025e15d79","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1659880930934,"defaultSendRequestWhen":"immediate"}

### Comments

Hi all, 

I am a noob to Z2M, I tried to configure this device and it seems to work now. It reads all the values properly, however I had no success in changing values like "report_temperature_interval".

I needed to change the fromZigbee and toZigbee as well:

Additional code in fromZigbee.js:

```
    tuya_lcd_temperature_humidity_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        options: [exposes.options.precision('temperature'), exposes.options.calibration('temperature'),
            exposes.options.precision('humidity'), exposes.options.calibration('humidity')],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
                const dp = dpValue.dp;
                const value = tuya.getDataValue(dpValue);
                switch (dp) {
                case tuya.dataPoints.tlthTemperature:
                    result.temperature = calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature');
                    break;
                case tuya.dataPoints.tlthHumidity:
                    result.humidity = calibrateAndPrecisionRoundOptions(value, options, 'humidity');
                    break;
                case tuya.dataPoints.tlthBattery:
                    result.battery = value;
                    break;
                case tuya.dataPoints.tlthMaxTemp:
                    result.max_temperature = calibrateAndPrecisionRoundOptions(value / 10, options, 'max temperature');
                    break;
                case tuya.dataPoints.tlthMinTemp:
                    result.max_temperature = calibrateAndPrecisionRoundOptions(value / 10, options, 'min temperature');
                    break;
                case tuya.dataPoints.tlthMaxHumi:
                    result.max_humidity = calibrateAndPrecisionRoundOptions(value, options, 'max humidity');
                    break;
                case tuya.dataPoints.tlthMinHumi:
                    result.min_humidity = calibrateAndPrecisionRoundOptions(value, options, 'min humidity');
                    break;
                case tuya.dataPoints.tlthTempUnitConvert:
                    result.temperature_unit_convert = {0x00: 'celsius', 0x01: 'fahrenheit'}[value];
                    break;
                case tuya.dataPoints.tlthTempReportInterval:
                    result.report_temperature_interval = value;
                    break;
                 case tuya.dataPoints.tlthHumiReportInterval:
                    result.report_humidity_interval = value;
                    break;
                case tuya.dataPoints.tlthTempSensitivity:
                    result.temperature_sensitivity = calibrateAndPrecisionRoundOptions(value / 10, options, 'temperature');
                    break;
                case tuya.dataPoints.tlthHumiSensitivity:
                    result.humidity_sensitivity = calibrateAndPrecisionRoundOptions(value, options, 'humidity');
                    break;
               default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_lcd_temperature_humidity_sensor: NOT RECOGNIZED ` +
                        `DP #${dp} with data ${JSON.stringify(dpValue)}`);
                }
            }
            return result;
        },
    },

```
additional code for toZigbee.js:
```
    tuya_lcd_temperature_humidity_sensor: {
        key: [
            'temperature_sensitivity', 'humidity_sensitivity', 'temperature_unit_convert',
            'report_temperature_interval', 'report_humidity_interval',
        ],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'temperature_unit_convert':
                await tuya.sendDataPointEnum(entity, tuya.dataPoints.tlthTempUnitConvert, ['celsius', 'fahrenheit'].indexOf(value));
                break;
            case 'temperature_sensitivity':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.tlthTempSensitivity, Math.round(value * 10));
                break;
            case 'humidity_sensitivity':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.tlthHumiSensitivity, Math.round(value));
                break;
            case 'report_temperature_interval':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.tlthTempReportInterval, value);
                break;
            case 'report_humidity_interval':
                await tuya.sendDataPointValue(entity, tuya.dataPoints.tlthHumiReportInterval, value);
                break;
            default: // Unknown key
                meta.logger.warn(`Unhandled key ${key}`);
            }
        },
    },

```

Additional code in zigbee-herdman-converters/lib/tuya.ts:
```
    // Tuya SMart LCD Temperature and Humidity Sensor
    tlthTemperature: 1,
    tlthHumidity: 2,
    tlthBattery: 4,
    tlthTempUnitConvert: 9,
    tlthMaxTemp: 10,
    tlthMinTemp: 11,
    tlthMaxHumi: 12,
    tlthMinHumi: 13,
    tlthTempAlarm: 14,
    tlthHumiAlarm: 15,
    tlthTempSensitivity: 19,
    tlthHumiSensitivity: 20,
    tlthTempReportInterval: 17,
	tlthHumiReportInterval: 18,

```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_locansqn'
        },
    ],
    model: '_TZE200_locansqn', // Vendor model number, look on the device for a model number
    vendor: 'Tuya', // Vendor of the device (only used for documentation and startup logging)
    description: 'Tuya Smart Zigbee Temperature And Humidity Sensor With LCD Display ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
	   fz.tuya_lcd_temperature_humidity_sensor, fz.ignore_tuya_set_time,
	],
    toZigbee: [
	    tz.tuya_lcd_temperature_humidity_sensor,
	], // Should be empty, unless device can be controlled (e.g. lights, switches).
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
       e.temperature(), e.humidity(), e.battery(), // Here you should put all functionality that your device exposes
	   exposes.numeric('report_temperature_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(120).withValueStep(5)
           .withDescription('Report temperature interval'),
	   exposes.numeric('report_humidity_interval', ea.STATE_SET).withUnit('min').withValueMin(1).withValueMax(120).withValueStep(5)
           .withDescription('Report humidity interval'),
	   exposes.numeric('temperature_sensitivity', ea.STATE_SET).withUnit('°C').withValueMin(0.1).withValueMax(5).withValueStep(0.1)
           .withDescription('temperature sensitivity'),
	   exposes.numeric('humidity_sensitivity', ea.STATE_SET).withUnit('%').withValueMin(1).withValueMax(10).withValueStep(1)
           .withDescription('humidity sensitivity'),
       exposes.enum('temperature_unit_convert', ea.STATE_SET, ['celsius', 'fahrenheit']).withDescription('Current display unit'),
    ],
};

module.exports = definition;
```

Koenkk, 2022-08-07:
`_TZE200_locansqn` has been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4415/files and is supported in z2m 1.27.0

owangen, 2022-08-22:
_TZE200_locansqn has basic support. The device supports more features than implemented right now. For example setting reporting interval for temperature and humidity, would be nice. I also get some warnings in the logs with the current implementation:
NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,5]}}
NOT RECOGNIZED DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,7]}}

elgerg, 2022-09-09:
> _TZE200_locansqn has basic support. The device supports more features than implemented right now. For example setting reporting interval for temperature and humidity, would be nice. I also get some [...]

I also get the same errors:
zigbee-herdsman-converters:nous_lcd_temperature_humidity_sensor: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}
zigbee-herdsman-converters:nous_lcd_temperature_humidity_sensor: NOT RECOGNIZED DP #20 with data {"dp":20,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,6]}}

elgerg, 2022-09-09:
@owangen @esackbauer 
Are your devices reporting their changes? Mine reported once but hasnt in the last couple of hours.
Thanks

esackbauer, 2022-09-09:
Yes its reporting, however I cannot say what is triggering it to report. Sometimes 0,1 degree changes are reported, mostly changes around 0.2-0-3 degrees are reported, rarely its >0,5 degree without a change. But works pretty well in general, but no comparison to my Sonoff Sensors, they are hastily updating even smallest changes.
I had no luck in changing the reporting frequency. According to the vendor you can set frequency to anywhere between 1 minute and 120 minutes.

elgerg, 2022-09-09:
@Koenkk 

I'm keen to be able to help in getting the reporting intervals changed. Ideally I would like it to check in every 2 minutes regardless of temp change.

Any ideas how I can go about this?

Thanks

Koenkk, 2022-09-10:
@elgerg TuYa uses manufacturer specific commands so we cannot use the standard ones. I see it already supports `temperature_sensitivity`, does changing this help?

owangen, 2022-09-10:
@elgerg I was developing support for every feature it supports https://github.com/Koenkk/zigbee-herdsman-converters/pull/4375
But I had some problems setting temp and humi report interval. I was planning to dig more into, when someone else added it. It works setting report interval when paired to the tuya gateway, so my plan was to sniff traffic between device and coordinator and see what it does.

elgerg, 2022-09-25:
@owangen did you make any progress/is there anything I can do to help?

Thanks!

owangen, 2022-09-25:
@elgerg I'm sorry, but I have not prioritized working on it.

closed 2022-11-02 by github-actions[bot]