Koenkk/zigbee2mqtt #13446: [New device support]: WZ5 Zigbee Tuya

z0mb1ek, 2022-08-07:
### Link

https://aliexpress.ru/item/1005003145273010.html

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xa4c138efa7265da6","nwkAddr":1677,"manufId":4417,"manufName":"_TZB210_3zfp8mki","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659888780282,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, I bought this controller before and it was already in z2m, but now manufName changed, please add it

Koenkk, 2022-08-07:
Please check if it works with the following external converter.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZB210_3zfp8mki'}],
  model: 'TS0505B',
  vendor: 'TuYa',
  description: 'Zigbee RGB+CCT light',
  extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500], disableColorTempStartup: true}),
  meta: {applyRedFix: true, enhancedHue: false},
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

z0mb1ek, 2022-08-07:
It works, but not so correctly, because in white mode i see color settings. Earlier it defined as:
[image]

and works perfectly

z0mb1ek, 2022-08-07:
I check all my 8 devices, and 1 of them is:
`{"id":14,"type":"Router","ieeeAddr":"0xa4c138fa3b0e9cff","nwkAddr":30167,"manufId":4417,"manufName":"_TZB210_rkgngb5o","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0501B","manufacturerName":"_TZB210_rkgngb5o","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659893403721,"defaultSendRequestWhen":"immediate"}`

add it please too

Koenkk, 2022-08-08:
Try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [
      {modelID: 'TS0601', manufacturerName: '_TZE200_mde0utnv'},
      {modelID: 'TS0601', manufacturerName: '_TZE200_aa9awrng'},
      {modelID: 'TS0501B', manufacturerName: '_TZB210_rkgngb5o'},
  ],
  model: 'WZ5_rgbcct',
  vendor: 'Skydance',
  description: 'Zigbee & RF 5 in 1 LED controller (RGB+CCT mode)',
  fromZigbee: [fz.tuya_light_wz5],
  toZigbee: [tz.tuya_dimmer_state, tz.tuya_light_wz5],
  exposes: [
      exposes.light().withBrightness().setAccess('state', ea.STATE_SET).setAccess('brightness',
          ea.STATE_SET).withColor(['hs']).withColorTemp([250, 454]).setAccess('color_temp',
          ea.STATE_SET),
      exposes.numeric('white_brightness', ea.STATE_SET).withValueMin(0).withValueMax(254).withDescription(
          'White brightness of this light'),
  ],
  meta: {separateWhite: true},
};

module.exports = definition;

```

Koenkk, 2022-08-17:
`_TZB210_rkgngb5o` has been added now in the dev branch. #13490 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-17 by Koenkk

ralawa, 2022-09-02:
Hi,

Like @z0mb1ek I also have a _TZB210_3zfp8mki device. It works with your first external converter and it also shows color settings in white mode. Do you plan to support it? I can help if necessary.

I currently only use white mode, I don't have RGB led strips, but is it supposed to announce the same name (_TZB210_3zfp8mki) regardless of the operating mode? Whatever the operating mode I set, my device always announces _TZB210_3zfp8mki.

MiloshCZ, 2022-10-05:
Hi,
I have problem with Skydance WZ5_rgbcct. Can I change order of parameters in JSON payload?

If I sent:
```
{
  "state" : "ON",
  "brightness" : "10"
}
```
Brightness is changed and mqtt answer is
```
{
  "brightness" : "10",
  "color" : {
    "h" : 60,
    "hue" : 60,
    "s" : 1,
    "saturation" : 1
  },
  "color_mode" : "hs",
  "color_temp" : 454,
  "last_seen" : 1664982691982,
  "linkquality" : 87,
  "state" : "ON",
  "white_brightness" : 0
}
```
debug:
```
Debug 2022-10-05 17:16:16Received MQTT message on 'zigbee2mqtt/LED pasek bar/set' with data '{ "state": "ON", "brightness": "10" }'
Debug 2022-10-05 17:16:16Publishing 'set' 'brightness' to 'LED pasek bar'
Debug 2022-10-05 17:16:17Publishing 'set' 'state' to 'LED pasek bar'
Info 2022-10-05 17:16:17MQTT publish: topic 'zigbee2mqtt/LED pasek bar', payload '{"brightness":"10","color":{"h":60,"hue":60,"s":1,"saturation":1},"color_mode":"hs","color_temp":454,"last_seen":1664982851213,"linkquality":87,"state":"ON","white_brightness":0}'
Debug 2022-10-05 17:16:17Received Zigbee message from 'LED pasek bar', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":3328}' from endpoint 1 with groupID 0
Info 2022-10-05 17:16:17MQTT publish: topic 'zigbee2mqtt/LED pasek bar', payload '{"brightness":"10","color":{"h":60,"hue":60,"s":1,"saturation":1},"color_mode":"hs","color_temp":454,"last_seen":1664982977095,"linkquality":87,"state":"ON","white_brightness":0}'
Debug 2022-10-05 17:16:17Received Zigbee message from 'LED pasek bar', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[48,48,51,99,48,48,48,97,48,48,50,55],"type":"Buffer"},"datatype":3,"dp":5}],"seq":3584}' from endpoint 1 with groupID 0
Info 2022-10-05 17:16:17MQTT publish: topic 'zigbee2mqtt/LED pasek bar', payload '{"brightness":10,"color":{"h":60,"hue":60,"s":1,"saturation":1},"color_mode":"hs","color_temp":454,"last_seen":1664982977142,"linkquality":87,"state":"ON","white_brightness":0}'
```

But when I send parameters in reverse order:
```
{
  "brightness" : "255",
  "state" : "ON"
}
```
Answer is correct, but no change in real world:
```
{
  "brightness" : "255",
  "color" : {
    "h" : 60,
    "hue" : 60,
    "s" : 1,
    "saturation" : 1
  },
  "color_mode" : "hs",
  "color_temp" : 454,
  "last_seen" : 1664982851213,
  "linkquality" : 87,
  "state" : "ON",
  "white_brightness" : 0
}
```
debug:
```
Debug 2022-10-05 17:17:31Received MQTT message on 'zigbee2mqtt/LED pasek bar/set' with data '{ "brightness": "255", "state": "ON" }'
Debug 2022-10-05 17:17:31Publishing 'set' 'state' to 'LED pasek bar'
Debug 2022-10-05 17:17:31Publishing 'set' 'brightness' to 'LED pasek bar'
Info 2022-10-05 17:17:31MQTT publish: topic 'zigbee2mqtt/LED pasek bar', payload '{"brightness":"255","color":{"h":60,"hue":60,"s":1,"saturation":1},"color_mode":"hs","color_temp":454,"last_seen":1664983021951,"linkquality":87,"state":"ON","white_brightness":0}'
Debug 2022-10-05 17:17:31Received Zigbee message from 'LED pasek bar', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":3840}' from endpoint 1 with groupID 0
Info 2022-10-05 17:17:31MQTT publish: topic 'zigbee2mqtt/LED pasek bar', payload '{"brightness":"255","color":{"h":60,"hue":60,"s":1,"saturation":1},"color_mode":"hs","color_temp":454,"last_seen":1664983051308,"linkquality":76,"state":"ON","white_brightness":0}'
```