Koenkk/zigbee2mqtt #13448: Moes ZTS-US_3gang wrongly paired as TuYa TS0013

relombardo, 2022-08-07:
### What happened?

After pairing my wall switch Moes ZTS-US_3gang trough zigbee2mqtt in HA, the interface wrongly detects as TuYa TS0013 switch. Even though this is a wrong model, I am able to toggle ON and OFF, but I do not have access to exposes Indicate_light (enum) and Power_on_behavior (enum). I know ZTS-US is not present in zigbee2mqtt documentation, but I am confident that ZTS-EU model should be fine. 

Is it possible to force my switch to be recognized as Moes ZTS-EU_3gang ([Link](https://www.zigbee2mqtt.io/devices/ZTS-EU_3gang.html#moes-zts-eu-3gang)) instead of  TuYa TS0013 ([Link](https://www.zigbee2mqtt.io/devices/TS0013.html#tuya-ts0013)) ?

Here is the database.db line corresponding to my device. I tried to breakdown lines to make it easier to read:

{"id":2,
"type":"EndDevice",
"ieeeAddr":"0xa4c138a63b37b597",
"nwkAddr":8907,
"manufId":4417,
"manufName":"_TZ3000_qewo8dlz",
"powerSource":"Mains (single phase)",
"modelId":"TS0013",
"epList":[1,2,3],
"endpoints":{"1":{"profId":260,
"epId":1,
"devId":256,
"inClusterList":[3,4,5,6,57344,57345,0],
"outClusterList":[25,10],
"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},
"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},
"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e12c","endpointID":1}],
"configuredReportings":[],
"meta":{}},
"2":{"profId":260,
"epId":2,"devId":256,
"inClusterList":[4,5,6,57345],
"outClusterList":[],
"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},
"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e12c","endpointID":1}],
"configuredReportings":[],
"meta":{}},
"3":{"profId":260,
"epId":3,
"devId":256,
"inClusterList":[4,5,6,57345],
"outClusterList":[],
"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},
"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e12c","endpointID":1}],
"configuredReportings":[],
"meta":{}}},
"appVersion":69,
"stackVersion":0,
"hwVersion":1,
"dateCode":"",
"zclVersion":3,
"interviewCompleted":true,
"meta":{"configured":2},
"lastSeen":1659876436634,
"defaultSendRequestWhen":"immediate"}

### What did you expect to happen?

I expected that my device would be found as a similar model, such as Moes ZTS-US_3gang in such a way I have access to the exposes  Indicate_light (enum) and Power_on_behavior (enum)

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

Koenkk, 2022-08-08:
Could you check if it works with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0013', manufacturerName: '_TZ3000_qewo8dlz'}],
    model: 'ZTS-US_3gang',
    vendor: 'Moes',
    description: 'Wall touch light switch (3 gang)',
    exposes: [e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET),
        exposes.enum('indicate_light', ea.STATE_SET, Object.values(tuya.moesSwitch.indicateLight))
            .withDescription('Indicator light status'),
        exposes.enum('power_on_behavior', ea.STATE_SET, Object.values(tuya.moesSwitch.powerOnBehavior))
            .withDescription('Controls the behavior when the device is powered on')],
    fromZigbee: [fz.ignore_basic_report, fz.tuya_switch, fz.moes_switch],
    toZigbee: [tz.tuya_switch_state, tz.moes_switch],
    onEvent: tuya.onEventSetLocalTime,
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        // Endpoint selection is made in tuya_switch_state
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        if (device.getEndpoint(2)) await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        if (device.getEndpoint(3)) await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

relombardo, 2022-08-08:
Hi Koenkk. Thanks for the prompt reply.  I followed your instructions but now, the exposes do not work at all, even to turn on and off. The exposes tab interface also shows a question mark (?) instead of toggle button at the first click. After that it replaces by the button.  The picture at the About tab is not showing a model similar to mine, which is a touch screen. Not sure if this is an indicator to be using a wrong model. This is the closest I found in zigbee2mqtt Moes ZTS-EU_3gang ([Link](https://www.zigbee2mqtt.io/devices/ZTS-EU_3gang.html#moes-zts-eu-3gang))

Koenkk, 2022-08-08:
This is the converter from the ZTS-EU_3gang, but given that that uses modelID `TS0601` I expect this is a completely different device (since yours uses modelID `TS0013`). I'm not sure if your device supports the additional features, you need to check if it does and find the data points of them with the original tuya gateway: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

relombardo, 2022-08-08:
I got it. Would it be possible to manually add Indicate_light (enum) and Power_on_behavior (enum) to TS0013? I am wondering if trying this out assuming they exist in my hardware would work. I do not have Tuya gateway to test right now.

Koenkk, 2022-08-08:
> Would it be possible to manually add Indicate_light (enum) and Power_on_behavior (enum) to TS0013

Only is supported by the original gateway, but at least it is different from the ` ZTS-EU_3gang`

relombardo, 2022-08-08:
Is there a chance to have my device officially listed on zigbee2mqtt supported devices in the future? I am pretty sure these additional exposes should be available. I saw people configuring indicator light through Tuya app using manufacture's gateway as you said.

relombardo, 2022-08-08:
I also found this: 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3822 Support power_on_behavior and indicate_light for Moes ZTS-EU_1gang, ZTS-EU_2gang and ZTS-EU_3gang

Should it be applied to my case?

Koenkk, 2022-08-09:
>  Is there a chance to have my device officially listed on zigbee2mqtt supported devices in the future?

If someone reverse the data points, yes

>  I am pretty sure these additional exposes should be available. I saw people configuring indicator light through Tuya app using manufacture's gateway as you said.

If that is indeed the exact same device, then it should be possible. But as I mentioned the data points differ from the ZTS-EU_*gang so these should be reverse engineered with the TuYa gateway 

> Should it be applied to my case?

This was exactly what was in the external converter from https://github.com/Koenkk/zigbee2mqtt/issues/13448#issuecomment-1208257242 but as you mentioned this doesn't work.

relombardo, 2022-08-09:
I got it, @Koenkk . Thank you very much for the clarifications. I am a beginner learner in HA automation, and things are starting to make sense for me now. I am glad to learn more. If a have the opportunity to get a Tuya gateway, I will do my best to collaborate with community and share my findings.

ericb48, 2022-08-26:
I have the same problem with the Moes ZTS-EU 1gang. below is an extract from database.db

{"id":25,"type":"EndDevice","ieeeAddr":"0xa4c13805a802249b","nwkAddr":55330,"manufId":4417,"manufName":"_TZ3000_hhiodade","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TZ3000_hhiodade","powerSource":3,"zclVersion":3,"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00039d956d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1482240604},"lastSeen":1661536809749,"defaultSendRequestWhen":"immediate"}

closed 2022-10-03 by github-actions[bot]

relombardo, 2022-10-04:
Hi @Koenkk , this issue was fixed in a mysterious way. Even though z2m still detects my devices as Tuya TS0013, there are new entities available for "_backlight_mode" and "_power_on_behavior" respectively. And I finally managed to configure the backlight. However, this feature is only working for the 3-gang version. I am opening a new request for 1-gang.

Koenkk, 2022-10-04:
@relombardo this has been added recently I believe.