Koenkk/zigbee2mqtt #13449: LELLKI plug showing as unsupported

icherfas, 2022-08-07:
### What happened?

After adding the plug to the network it shows as unsupported.
I have 2 of the same, both showing the same in the network.
The devices appear in the supported devices. 

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

CC2531_DEFAULT_20211115

### Adapter

Texas Instruments CC2531

### Debug log

`Zigbee2MQTT:warn  2022-08-07 21:47:34: Device '0x003c84fffe85c324' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_ps3dmato' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:error 2022-08-07 21:47:39: Failed to interview '0x003c84fffe85b696', device has not successfully been paired
Zigbee2MQTT:warn  2022-08-07 21:49:46: Device '0x003c84fffe85b696' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_ps3dmato' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Danit2, 2022-08-08:
You must update your Z2M. 

https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/134#issuecomment-855363394

icherfas, 2022-08-08:
> You must update your Z2M.
> 
> [zigbee2mqtt/hassio-zigbee2mqtt#134 (comment)](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/134#issuecomment-855363394)

thank you!
for some reason, I was under the impression it will be updated automatically.
updated the issue is gone

closed 2022-08-08 by icherfas