Koenkk/zigbee2mqtt #13451: Changes in DEV tree with postfixWithEndpointName has broken custom converters

RJSDevel, 2022-08-08:
### What happened?

debug 2022-08-07 22:12:27: TypeError: Cannot read property 'device' of undefined
    at postfixWithEndpointName (C:\Users\Developer\Desktop\zigbee2mqtt\node_modules\zigbee-herdsman-converters\lib\utils.js:118:45)
    at Object.convert (C:\Users\Developer\Desktop\zigbee2mqtt\dist\util\externally-loaded.js:48:12)
    at Receive.onDeviceMessage (C:\Users\Developer\Desktop\zigbee2mqtt\lib\extension\receive.ts:143:51)
    at EventEmitter.emit (events.js:412:35)
    at EventBus.emitDeviceMessage (C:\Users\Developer\Desktop\zigbee2mqtt\lib\eventBus.ts:102:22)
    at Controller.<anonymous> (C:\Users\Developer\Desktop\zigbee2mqtt\lib\zigbee.ts:106:27)
    at Controller.emit (events.js:400:28)
    at Controller.selfAndDeviceEmit (C:\Users\Developer\Desktop\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:495:14)
    at Controller.onZclOrRawData (C:\Users\Developer\Desktop\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:706:18)
    at ZStackAdapter.<anonymous> (C:\Users\Developer\Desktop\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:143:70)

### How to reproduce it (minimal and precise)

old - postfixWithEndpointName('volume', msg, model)
new - postfixWithEndpointName('volume', msg, model, **meta**)

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

000000000

### Adapter

CC2531

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9279005/log.txt)

Koenkk, 2022-08-08:
You should update your external converter to also pass the `meta` property.

RJSDevel, 2022-08-08:
All convectors? Could you override one method for save compatibility with alert log? I don't think that people will read readme

Koenkk, 2022-08-08:
This was not a refactor [but a fix](https://github.com/Koenkk/zigbee2mqtt/issues/13138) (otherwise the `postfixWithEndpointName` method doesn't work correctly) so I suggest to update the external converter.

RJSDevel, 2022-08-08:
Are you going to warn other developers or add explanation to code if the meta hasn't passed to the method? Because I was findind the reason about half hour. I suggest there are extremely changes for "simple people"

Koenkk, 2022-08-08:
Valid point, pushed a fix please let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RJSDevel, 2022-08-09:
This has fixed the error, but device aren't working at all, events doesn't receive.

han-alink, 2022-08-17:
I think this issue is causing an error in at least one of my external definitions. But I do not understand what should be changed in the following device description:
    {
    zigbeeModel: ['onOffSwitch'],
    model: 'OnOffTouchController',
    vendor: 'Alink', 
    description: 'Touch on off step switch',
    fromZigbee: [fzLocal.touch_input, fz.battery],
    toZigbee: [],
    exposes: [e.contact().withEndpoint('left'), e.contact().withEndpoint('right'), e.battery(), e.battery_voltage()], 
    meta: { multiEndpoint: true, battery: {voltageToPercentage: '3V_2500_3200'}},
    endpoint: (device) => {return {'left': 1, 'right': 2}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const overides = {min: 1, max: 3600};
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genPowerCfg','genBinaryInput']);  
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genBinaryInput']);
        await reporting.presentValue(endpoint1, overides);
        await reporting.presentValue(endpoint2, overides);
        await reporting.batteryVoltage(endpoint1, overides);
        }
    },

closed 2022-09-24 by github-actions[bot]