Koenkk/zigbee2mqtt #13457: TuYa IH012-RT01 spamming lqi updates after attempting to set parameters

dwf, 2022-08-08:
### What happened?

The reported keep time on this sensor (of 120s) did not match what I was seeing in Home Assistant (clearing in just over a minute) so I attempted to change it, as well as alter the sensitivity.

Attempting to change the parameters of this motion sensor from the UI resulted in the UI being "stuck" with two values of e.g. sensitivity selected, but more importantly, somehow this caused the sensor (or Z2M in response to the sensor, hard to place the blame) to start spamming updates (notably in LQI) more than once a second. This caused it to jump around in the Device list such that it was difficult to even select. Cutting power to the sensor by removing the battery stopped it, but it continued once powered again; removing and re-adding it did not stop it. The combination of cut power -> remove device -> re-power -> re-pair seems to have fixed it.

### What did you expect to happen?

I expected the device parameter to get set and then for it to behave normally. 

### How to reproduce it (minimal and precise)

It's happened twice now, and messing around with the exposed parameters (sensitivity and keep time) seems to be the common thread, however I don't have an exact reproduction recipe, nor can I provide debug logs just at the moment (I will post them once I can reproduce). Also happy to donate a sensor to the authors for debugging, if this would be helpful.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Plus dongle

closed 2022-09-16 by github-actions[bot]

dwf, 2022-09-16:
Sorry, this is still relevant.

FlexxFR, 2022-10-30:
Did you ever found a solution, I have the exact same issue.
I have 4 of these, 2 operational and 1 is having this issue after trying to adjust the sensitivity.

Fresh, out-of-the box I was able to set sensitivity to high (default medium?) and keep time to 60 (default 30). Didn't check the logs back then. But now I want to change it again, and this started happening. 

I will start with the repair/power-off/on procedure

chicco83, 2022-12-22:
same issue happened to me ten minutes ago.