Koenkk/zigbee2mqtt #13459: Xiaomi WXKG15LM (Aqara Remote Switch H1) - No "release" event

fredck, 2022-08-08:
### What happened?

There is any kind if "release" event being fired when the device buttons are released. The following (basic) automations are impossible because of this:

 * No continuous hold automations that perform until the button is released (e.g. dim lights up/down)
 * It's not possible to simulate "hold" events when the "fast" `click_mode` is enabled.

### What did you expect to happen?

If `click_mode` is "multi" (or default), the actions `hold_left`, `hold_right`, `hold_both` should be followed by `release` once the button(s) are released.

If `click_mode` is "fast", the action single_left, single_right, single_both should be followed by `release` once the button(s) are release.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210708

### Adapter

Sonoff USB

fredck, 2022-09-13:
Unfortunately I'm not able to provide a patch for it.

What's the correct course of action to have this issue resolved? Is there any known path within Xiomi / Aqara that we could try to use?

Koenkk, 2022-09-14:
@fredck does the switch send a release event? You can check by activating the debug logging and see if a zigbee message is received when you release the switch.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

z-x-f, 2022-09-14:
This wireless switch will not announce the release event, no matter it is fast mode or multi mode, fast mode only has click event, because fast response is required, multi mode has double click and hold event.  Considering that holding for too long will cause the device to initialize, I don't think it is suitable for use as hold-release automation.

fredck, 2022-09-19:
It seems that this device is simply limited as the one and only I've met so far to not dispatch a release event. We can't do much to work this around, which drastically reduces the use cases for this device.

closed 2022-09-19 by fredck

fredck, 2022-11-10:
Just for documentation, I've tried contacting Aqara on this issue and this is everything I got:

> We make no claims about being compatible with other brands (Home Assistant). Some enthusiasts have managed to get Aqara devices working with different brands but that is not something we support.
> 
> You will need to have an Aqara Hub in order to use our devices.

Basically, they don't care.