Koenkk/zigbee2mqtt #13460: [New device support]: _TZE200_0u3bj3rc Tuya ZigBee Human Presence Sensor MmWave Radar

ferbulous, 2022-08-08:
### Link

https://www.aliexpress.com/item/1005004542293208.html?spm=a2g0o.productlist.0.0.59ce419dmZy7nm&algo_pvid=d3f9f8f0-453a-443f-9905-10de46569ba9&algo_exp_id=d3f9f8f0-453a-443f-9905-10de46569ba9-2&pdp_ext_f=%7B%22sku_id%22%3A%2212000029541349486%22%7D&pdp_npi=2%40dis%21MYR%21223.24%21104.93%21%21%21%21%21%40210318be16599806638003852ef3c9%2112000029541349486%21sea&curPageLogUid=JVvjick50yaf

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0xa4c13859309f72a7","nwkAddr":40097,"manufId":4417,"manufName":"_TZE200_0u3bj3rc","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�w�*i�w�*i�w�*i�w�*i�w�*f�w�*i�w�*i�w�*\u0012","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1659980554955,"defaultSendRequestWhen":"immediate"}

### Comments

I've also tried dev branch (z2m edge) but I'm still getting the same error message it's not supported yet. 
I haven't tried using external converter since I'm not sure what I should use for this device.
It keeps looping error message so I've unpaired it for now

[image]

nmaster2042, 2022-08-12:
Have one and same issue.

ferbulous, 2022-08-14:
> Have one and same issue.

Do you also get those odd symbols in the database entry?

nmaster2042, 2022-08-14:
> Do you also get those odd symbols in the database entry?

Yes, it's strange.

burken-, 2022-08-16:
Not really sure about the values for dp 101 and 102. I figure they are occupancy time and away time expressed in minutes. The other values are occupancy and indicator light (on/off). Got things working by adding this external converter (work in progress). 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    human_radar_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'human_radar_sensor');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
//            meta.logger.debug(`from human_radar_sensor, dp=[${dp}], datatype=[${dpValue.datatype}], value=[${value}]`);
            let result = null;
            switch (dp) {
            case tuya.dataPoints.trsPresenceState:
                result = {occupancy: {0: false, 1: true}[value]};
                break;
            case 101:
                result = {occupancy_minutes: value};
                break;
            case 102:
                result = {away_minutes: value};
                break;
            case 103:
                result = {indicator_light: value ? 'ON' : 'OFF'};
                break;
			default:
//                meta.logger.warn(`fromZigbee.human_radar_sensor: NOT RECOGNIZED DP ${dp} with data ${JSON.stringify(dpValue)}`);
            }
            return result;
        },
    },
};

const tzLocal = {
    human_radar_sensor: {
        key: ['indicator_light'],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'indicator_light':
                await tuya.sendDataPointBool(entity, 103, value === 'ON');
                break;
            default: // Unknown key
                throw new Error(`tz.human_presence_sensor: Unhandled key ${key}`);
            }
        }
    }
};

const definition = {
    fingerprint: [
     {
        modelID: 'TS0601',
        manufacturerName: '_TZE200_0u3bj3rc'
     }
    ],
    model: 'ZN368199_01',
    vendor: 'Tuya',
    description: 'Human Presence Sensor',
    fromZigbee: [
//        fz.ignore_basic_report,
//        fz.tuya_data_point_dump,
        fzLocal.human_radar_sensor,
    ],
    toZigbee: [
//        tz.tuya_data_point_test,
        tzLocal.human_radar_sensor,
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.occupancy(), exposes.binary('indicator_light', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable indicator light'),
    exposes.numeric('occupancy_minutes', ea.STATE), exposes.numeric('away_minutes', ea.STATE)],
};

module.exports = definition;
```

cbrightly, 2022-08-21:
It seems like 105 may be a setting for target distance using 0-9 to set distance in 0.5 meter intervals so that 0 = 0.5 m, 1 = 1.0 m, 2 = 1.5 m … 9 = 5.0 m. 
Can someone more familiar than I with external converters test this please?

burken-, 2022-08-22:
> @burken- Thanks for the external converter! Working great for me.
> 
> Of note, it seems that 105 may be a setting for target distance using 0–9 to set distance in 0.5 meter intervals so that 0 = 0.5 m, 1 = 1.0 m, 2 = 1.5 m … 9 = 5.0 m. Can someone more familiar than I with external converters test this please?

@cbrightly Are you getting 105:s in the zigbee2mqtt-log or have you found them in the tuya-log? I haven't seen any in my logs. Do you have a tuya gateway and a tuya app? It would be easier to find all dp-values that way.

cbrightly, 2022-08-23:
> @cbrightly Are you getting 105:s in the zigbee2mqtt-log or have you found them in the tuya-log? I haven't seen any in my logs. Do you have a tuya gateway and a tuya app? It would be easier to find [...]

I don't believe so; at least I don't think what I have is branded as being Tuya, though the commercial hubs I own may well be Tuya based.

I have seen (and read on the seller's description & other places on GitHub) that this particular human presence sensor should expose a sensitivity parameter as well as a 'degree of motion' setting (which as I understand it shows things like unoccupied, occupied but no movement, occupied with small movements present, and occupied with large amounts of motion detected. I am aware that some work has been done to support these data points within other coordinator solutions, though I don't know how much progress has been made.

If it is of any help, I have a Sonoff Zigbee USB dongle (the non-'Plus' version with a PCB-antenna instead of an external one) running Z2M via Home Assistant OS on a Raspberry Pi 4, a ConBee II USB dongle running deConz + Phoscon on a separate setup, and access to both a Linkind and Sonoff branded Zigbee / WiFi hub. 

Please let me know if I can contribute in any way.

burken-, 2022-08-23:
@cbrightly If you can point me in the direction of the seller's description or the github places you've found I can check if there's anything I can use to complete the converter. I found another implementation for this device in a different project but it was basically the same as I have i.e. for data point 1, 101, 102 and 103.

fuatakgun, 2022-08-23:
I have the same device and used tuya cloud to dump the registers. It is only 1, 101, 102 and 103

burken-, 2022-08-24:
I changed a few things in my external converter. This device is really chatty and I failed to figure out how (and if possible) to configure reporting for specific data points in the manuSpecificTuya-cluster. If anyone knows how to do this please do tell! 

For now I added an option to make it less chatty over mqtt even though it's still very chatty over zigbee. If "quiet" is set to true (default) the device will only broadcast messages when a value changes.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const fzLocal = {
    human_radar_sensor: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'human_radar_sensor');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            let result = null;
            switch (dp) {
            case tuya.dataPoints.trsPresenceState:
                result = {occupancy: {0: false, 1: true}[value]};
                break;
            case 101:
                result = {occupancy_minutes: value};
                break;
            case 102:
                result = {away_minutes: value};
                break;
            case 103:
                result = {indicator_light: value ? 'ON' : 'OFF'};
                break;
            default:
                meta.logger.warn(`fromZigbee.human_radar_sensor: NOT RECOGNIZED DP ${dp} with data ${JSON.stringify(dpValue)}`);
                break;
            }

            var broadcast = !(options.quiet??true);
            if(options.quiet ?? true){
                const last = globalStore.getValue(msg.endpoint, 'dp_' + dp, null);
                const current = result[Object.keys(result)[0]];

                if(last !== current)
                {
                    globalStore.putValue(msg.endpoint, 'dp_' + dp, current);
                    broadcast = true;
                }
            }
            
            //Broadcast if chatty or if value changed
            if(broadcast)
                return result;
        },
    },
};

const tzLocal = {
    human_radar_sensor: {
        key: ['indicator_light'],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'indicator_light':
                await tuya.sendDataPointBool(entity, 103, value === 'ON');
                break;
            default: // Unknown key
                throw new Error(`tz.human_presence_sensor: Unhandled key ${key}`);
            }
        },
    }
};

const definition = {
    fingerprint: [
     {
        modelID: 'TS0601',
        manufacturerName: '_TZE200_0u3bj3rc'
     }
    ],
    model: 'ZN368199_01',
    vendor: 'Tuya',
    description: 'Black Square Human Presence Sensor',
    options: [exposes.binary('quiet', ea.SET, true, false).withDescription('When enabled the device will only broadcast changed values. Recommended to set this to true since this device is really chatty. Default = true. Also recommended to set "Debounce" to 1 and filter out the attributes you don\'t need')],
    fromZigbee: [
//        fz.ignore_basic_report,
//        fz.tuya_data_point_dump,
        fzLocal.human_radar_sensor,
    ],
    toZigbee: [
        tz.tuya_data_point_test,
        tzLocal.human_radar_sensor,
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
        await endpoint.configureReporting('manuSpecificTuya', {});
        await tuya.sendDataPointBool(endpoint, 103, false);
    },
        exposes: [e.occupancy(), exposes.binary('indicator_light', ea.STATE_SET, 'ON', 'OFF').withDescription('Enable indicator light'),
        exposes.numeric('occupancy_minutes', ea.STATE), exposes.numeric('away_minutes', ea.STATE)],
};

module.exports = definition;
```

cbrightly, 2022-09-07:
> I changed a few things in my external converter. This device is really chatty and I failed to figure out how (and if possible) to configure reporting for specific data points in the manuSpecificTuya-cluster. If anyone knows how to do this please do tell! 

This works very well for me; thank you very much for sharing! I'm not particularly familiar with Z2M -- is this ready for a PR?

Mirarkitty, 2022-09-15:
Got one too, but it presents itself as **_TZE200_v6ossqfy** - however the datapoint match very well with above.

burken-, 2022-09-27:
I'll try to find the time to structure everything in a PR to get it added to the code base. If someone else feels the urge to get it done quicker, please use whatever you want/need from the external converter.

Soukyuu, 2022-10-07:
> @burken- Thanks for the external converter! Working great for me.
>
> Of note, it seems that 105 may be a setting for target distance using 0–9 to set distance in 0.5 meter intervals so that 0 = 0.5 m, [...]

Could this be the reason that the sensor works more like a PIR sensor in its current support state (Zigbee2mqtt 1.28.0-1)? It fails to detect me at 1.5m distance in direct line of sight once I stop moving.

talwinter, 2022-11-04:
The sensor always shows that a presence is detected in Zigbee2mqtt .. i dont know why.
I tried to add an external converter and adding it to the configuration yml but it seems that it doesn't work.
Any ideas why it says "Occupied" all the time.
How much is the cool down time?

Mirarkitty, 2022-12-18:
Yes, occupied has a cooldown of a few seconds. It will notice at once if you point the sensor to a wall.

closed 2023-01-26 by github-actions[bot]