Koenkk/zigbee2mqtt #13463: Xiaomi power plug ZNCZ04LM - after upgrade OTA fw to 01-23-2022 energy amount has been reset to zero

KitHubek, 2022-08-08:
### What happened?

As a title

lumi.plug.mmeu01 Xiaomi power plug ZigBee ZNCZ04LM
After upgrade firmware via zigbee2mqtt web panel OTA tab, to newest 01-23-2022 my previous energy amounthas been reset to zero

### What did you expect to happen?

Energy amount stay old counter

### How to reproduce it (minimal and precise)

go to z2mqt panel, go to OTA tab, check fw ugrade avialable for this device, and click: upgrade. Wait about 30 mins, and energy go to zero :/

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

[image]
[image]
[image]
[image]
[image]

Koenkk, 2022-08-08:
I guess the FW update does this, not something that can be fixed from z2m.

KitHubek, 2022-08-08:
I wonder if this counter reset after updating the software, if it broke and will no longer work. so far it still shows zero but there is very little consumption on this socket, it consumes about 2-3W so it will take a while before there is any consumption. 

Is there any way to send a command to the device that will record the state of consumed energy for example 20kWh and the device will now send this measurement + new consumption to add to it. And is there any option to downgrade the software to the previous one that worked properly?

KitHubek, 2022-08-08:
Update just reset energy counter. After some time it starts counting from zero. 

[image]

Is there any way to send a command to the device that will record the state of consumed energy for example 20kWh and the device will now send this measurement + new consumption to add to it.

Koenkk, 2022-08-09:
> Is there any way to send a command to the device that will record the state of consumed energy for example 20kWh and the device will now send this measurement + new consumption to add to it.

The is no standard command available for this. In case there is such a command (should be checked with the original Xiaomi gateway) a sniff of this is necessary to add support for it. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

KitHubek, 2022-08-09:
ok. thanks for support

closed 2022-08-09 by KitHubek