Koenkk/zigbee2mqtt #13464: [Feature request]: Set temp display units for Xiaomi VOCKQJK11LM

ptr727, 2022-08-08:
### Is your feature request related to a problem? Please describe

Default display temp units is C, would like to change units to F.
Could not find any info on changing units using Z2M.
Have to use Aqara Home app on iOS with M2 hub to change display units from C to F, then rebind to Z2M.

### Describe the solution you'd like

Would like ot be able to configure display temp units in Z2M app.

### Describe alternatives you've considered

Use Aqara Home app and M2 hub to change units.

### Additional context

https://www.reddit.com/r/homeassistant/comments/rkyllu/aqara_tvoc_air_quality_monitor/

ptr727, 2022-09-13:
See: https://forum.phoscon.de/t/aqara-tvoc-zhaairquality-data/1160/21

```
cluster 0xfcc0 / attribute 0x0114 (display on the device):
bit 1: mg/m³ or ppb (unset, set)
bit 2: temperature °C/°F (unset, set)
```

Koenkk, 2022-09-13:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    VOCKQJK11LM_display_unit: {
        key: ['display_unit'],
        convertSet: async (entity, key, value, meta) => {
            // bit 1: mg/m³ or ppb (unset, set)
            // bit 2: temperature °C/°F (unset, set)
            const lookup = {
                0b00000000: 'mgm3_celsius',
                0b00000001: 'ppb_celsius',
                0b00000010: 'mgm3_fahrenheit',
                0b00000011: 'ppb_fahrenheit',
            }
            await entity.write('aqaraOpple', {0x0114: {value: lookup[value], type: 0x20}}, {manufacturerCode: 0x115F, disableDefaultResponse: true});
            return {state: {display_unit: value}};
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.airmonitor.acn01'],
    model: 'VOCKQJK11LM',
    vendor: 'Xiaomi',
    whiteLabel: [{vendor: 'Xiaomi', model: 'AAQS-S01'}],
    description: 'Aqara TVOC air quality monitor',
    fromZigbee: [fz.xiaomi_tvoc, fz.battery, fz.temperature, fz.humidity, fz.aqara_opple],
    toZigbee: [tzLocal.VOCKQJK11LM_display_unit],
    meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
    exposes: [e.temperature(), e.humidity(), e.voc(), e.device_temperature(), e.battery(), e.battery_voltage(),
        exposes.enum('display_unit', ea.STATE_SET, ['mgm3_celsius', 'ppb_celsius', 'mgm3_fahrenheit', 'ppb_fahrenheit'])
                .withDescription('Units to show on the display')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genAnalogInput'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.humidity(endpoint);
        await reporting.temperature(endpoint);
        const payload = reporting.payload('presentValue', 10, constants.repInterval.HOUR, 5);
        await endpoint.configureReporting('genAnalogInput', payload);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- use z2m frotnend -> device -> exposes to change the units

ptr727, 2022-09-13:
1)  
The UI shows the settings, I managed to change the temp from F to C by clicking the ui button, then clicking the device button once, and the display switched.  
After that I could not get the display to change back to F, no matter what option I select, click, partial select, press button, ui completes select, but the device does not change.  
I do assume that the command is sent as soon as the device wakes up, which could be self scheduled, or when I press the button?

2)  
Looking at the epaper UI, I am not sure where the change in ppb vs. mg/m3 is suposed to be displayed, I just see leaves?

3)  
I am not familiar with node, but wondering if the `reporting.humidity()` and `reporting.temperature()` should not also include reporting of voc? But, I don't really know what it is supposed to do, even looking at some other examples, maybe the `genAnalogInput` is for battery?

logs:

```
debug 2022-09-13 15:33:56: Received MQTT message on 'zigbee2mqtt/Aqara-Air-1A1893/set' with data '{"display_unit":"mgm3_fahrenheit"}'
debug 2022-09-13 15:33:56: Publishing 'set' 'display_unit' to 'Aqara-Air-1A1893'
debug 2022-09-13 15:34:04: Received Zigbee message from 'Aqara-Air-1A1893', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.airmonitor.acn01"}' from endpoint 1 with groupID 0
info  2022-09-13 15:34:04: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":25504,"powerSource":"Battery","softwareBuildID":"2020\u0000www.","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":25,"display_unit":"mgm3_fahrenheit","humidity":59.1,"last_seen":"2022-09-13T22:34:04.526Z","linkquality":83,"power_outage_count":8,"temperature":25.05,"update":{"state":null},"update_available":null,"voc":89,"voltage":3117}'
debug 2022-09-13 15:34:04: Received Zigbee message from 'Aqara-Air-1A1893', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,45,12,3,40,25,4,33,168,67,5,33,9,0,6,36,4,0,0,0,0,8,33,29,1,10,33,0,0,12,32,1,100,41,202,9,101,33,10,23,102,33,93,0,103,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0
debug 2022-09-13 15:34:04: lumi.airmonitor.acn01: Processed buffer into data {"1":3117,"3":25,"4":17320,"5":9,"6":4,"8":285,"10":0,"12":1,"100":2506,"101":5898,"102":93,"103":1}
debug 2022-09-13 15:34:04: lumi.airmonitor.acn01: unknown key 6 with value 4
debug 2022-09-13 15:34:04: lumi.airmonitor.acn01: Processed data into payload {"voltage":3117,"battery":100,"device_temperature":25,"power_outage_count":8,"temperature":25.06,"humidity":58.98}
debug 2022-09-13 15:34:04: lumi.airmonitor.acn01: Processed data into payload {"voltage":3117,"battery":100,"device_temperature":25,"power_outage_count":8,"temperature":25.06,"humidity":58.98}
info  2022-09-13 15:34:04: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":25504,"powerSource":"Battery","softwareBuildID":"2020\u0000www.","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":25,"display_unit":"mgm3_fahrenheit","humidity":58.98,"last_seen":"2022-09-13T22:34:04.533Z","linkquality":83,"power_outage_count":8,"temperature":25.06,"update":{"state":null},"update_available":null,"voc":89,"voltage":3117}'
debug 2022-09-13 15:34:05: Received Zigbee message from 'Aqara-Air-1A1893', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":96}' from endpoint 1 with groupID 0
info  2022-09-13 15:34:05: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":25504,"powerSource":"Battery","softwareBuildID":"2020\u0000www.","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":25,"display_unit":"mgm3_fahrenheit","humidity":58.98,"last_seen":"2022-09-13T22:34:05.852Z","linkquality":58,"power_outage_count":8,"temperature":25.06,"update":{"state":null},"update_available":null,"voc":96,"voltage":3117}'
info  2022-09-13 15:34:05: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":25504,"powerSource":"Battery","softwareBuildID":"2020\u0000www.","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":25,"display_unit":"mgm3_fahrenheit","humidity":58.98,"last_seen":"2022-09-13T22:34:05.885Z","linkquality":65,"power_outage_count":8,"temperature":25.06,"update":{"state":null},"update_available":null,"voc":96,"voltage":3117}'
info  2022-09-13 15:34:05: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturer [... cut ...]
info  2022-09-13 15:36:58: MQTT publish: topic 'zigbee2mqtt/Aqara-Air-1A1893', payload '{"battery":100,"device":{"applicationVersion":29,"dateCode":"20220222","friendlyName":"Aqara-Air-1A1893","hardwareVersion":1,"ieeeAddr":"0x54ef4410001a1893","manufacturerID":4447,"manufacturerName":"LUMI","model":"VOCKQJK11LM","networkAddress":25504,"powerSource":"Battery","softwareBuildID":"2020\u0000www.","stackVersion":2,"type":"EndDevice","zclVersion":3},"device_temperature":25,"display_unit":"ppb_fahrenheit","humidity":58.87,"last_seen":"2022-09-13T22:36:58.946Z","linkquality":76,"power_outage_count":8,"temperature":25.07,"update":{"state":null},"update_available":null,"voc":111,"voltage":3117}'
```

ptr727, 2022-09-14:
I captured wireshark traces.
I updated the device to firmware v29, latest reported by aqara app.
I set units to F, to C, to ppb, and to mgm3, capturing individual logs.
The M2 app asks to press the button on making a settings change, and reported success after every change, individual logs captured.
The device UI does not change with ppb or mgm3, only the F or C makes a UI change.

The network M2 info:
M2 PAN ID: `0xBD30`
M2 Channel: `25`
M2 Network Key: `38:DD:FE:6B:B4:29:DA:97:F0:35:76:83:EC:01:D2:2F`

[M2_VOC_27_to_29.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9562259/M2_VOC_27_to_29.pcapng.gz)
[M2_VOC_ppb.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9562260/M2_VOC_ppb.pcapng.gz)
[M2_VOC_mgm3.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9562261/M2_VOC_mgm3.pcapng.gz)
[M2_VOC_F.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9562262/M2_VOC_F.pcapng.gz)
[M2_VOC_C.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9562263/M2_VOC_C.pcapng.gz)

Koenkk, 2022-09-14:
1. I cannot decrypt the sniff with key `38:DD:FE:6B:B4:29:DA:97:F0:35:76:83:EC:01:D2:2F`, can you provide me a screenshot of the data payloads of F -> C, C -> F? 
2. I don't know either
3. The ZCL (Zigbee) standard defines no VOC cluster, therefore Xiaomi seems to misuse the `genAnalogInput` for VOC reporting

ptr727, 2022-09-14:
Here are my zigbee_pc_keys, I am just removing my home Z2M key:

```
# This file is automatically generated, DO NOT MODIFY.
"5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39","Normal","Trust Center"
"x","Normal","Z2M Network Key"
"81:42:86:86:5D:C1:C8:B2:C8:CB:C5:2E:5D:65:D1:B8","Normal","ZLL Trust Center"
"9F:55:95:F1:02:57:C8:A4:69:CB:F4:2B:C9:3F:EE:31","Normal","ZLL Master Key"
"38:DD:FE:6B:B4:29:DA:97:F0:35:76:83:EC:01:D2:2F","Normal","M2 Network Key"
```

There are several write attributes commands per change, if the keys don't work, I'll take screenshots of all of them.

For 1, as above, for 2, maybe the units should not be messed with if the change is not understood, for 3, got it.

ptr727, 2022-09-14:
Set ppb, write attributes repeated 3 times, same value:  
[image]

Set mgm3, write attribute repeated 4 times, same value:  
[image]

Set F, write attributed, repeated 3 times, same value:  
[image]

Set C, write attribute, repeated 3 times, same value:  
[image]

Koenkk, 2022-09-15:
I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13464#issuecomment-1245025845, please check again

ptr727, 2022-09-15:
I see you changed the bit sets to account for 0x01 vs 0x10.

I had a quick test before going to work and it does not work.  
I will capture differences in attributes between making unit changes in z2m and the aqara app tonight, and post the results.

Thank you for continuing to look.

ptr727, 2022-09-15:
Ok, couldn't resist, did a quick capture.  
Looks like the attributes are always written as 0x00?

[Z2M_VOC_mgm3_c.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9576624/Z2M_VOC_mgm3_c.pcapng.gz)
[Z2M_VOC_ppb_c.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9576626/Z2M_VOC_ppb_c.pcapng.gz)
[Z2M_VOC_ppb_f.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9576627/Z2M_VOC_ppb_f.pcapng.gz)
[Z2M_VOC_mgm3_f.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/9576628/Z2M_VOC_mgm3_f.pcapng.gz)

[image]
[z2mlog.zip](https://github.com/Koenkk/zigbee2mqtt/files/9576638/z2mlog.zip)

ptr727, 2022-09-15:
This is showing promising results:

```
            const lookup = {
                'mgm3_celsius': 0b00000,
                'ppb_celsius': 0b00001,
                'mgm3_fahrenheit': 0b10000,
                'ppb_fahrenheit': 0b10001
            }
```

Koenkk, 2022-09-16:
Ah my bad, nice that you were able to fix the converter.

> What about reading the active state and reporting that, i.e. inverse of the lookup?

Does the device send an update when you change it on the device? You should see a received zigbee message when you change it in the z2m debug log.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Please also post your fixed external converter.

ptr727, 2022-09-16:
Here is working code:

```js
const constants = require('zigbee-herdsman-converters/lib/constants');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    VOCKQJK11LM_display_unit: {
        key: ['display_unit'],
        convertSet: async (entity, key, value, meta) => {
            // Match key names from exposes.enum
            const lookup = {
                'mgm3_celsius': 0x00, // mg/m³,  (default)
                'ppb_celsius': 0x01, // ppb, °C
                'mgm3_fahrenheit': 0x10, // mg/m³, °F
                'ppb_fahrenheit': 0x11 // ppb, °F
            }
            await entity.write('aqaraOpple', { 0x0114: { value: lookup[value], type: 0x20 } }, { manufacturerCode: 0x115F, disableDefaultResponse: true });
            return { state: { display_unit: value } };
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.airmonitor.acn01'],
    model: 'VOCKQJK11LM',
    vendor: 'Xiaomi',
    whiteLabel: [{ vendor: 'Xiaomi', model: 'AAQS-S01' }],
    description: 'Aqara TVOC air quality monitor',
    fromZigbee: [fz.xiaomi_tvoc, fz.battery, fz.temperature, fz.humidity, fz.aqara_opple],
    toZigbee: [tzLocal.VOCKQJK11LM_display_unit],
    meta: { battery: { voltageToPercentage: '3V_2850_3000' } },
    exposes: [e.temperature(), e.humidity(), e.voc(), e.device_temperature(), e.battery(), e.battery_voltage(),
    exposes.enum('display_unit', ea.STATE_SET, ['mgm3_celsius', 'ppb_celsius', 'mgm3_fahrenheit', 'ppb_fahrenheit'])
        .withDescription('Units to show on the display')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genAnalogInput'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.humidity(endpoint);
        await reporting.temperature(endpoint);
        const payload = reporting.payload('presentValue', 10, constants.repInterval.HOUR, 5);
        await endpoint.configureReporting('genAnalogInput', payload);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

I'm looking at captures and I never see a report attributes for 0x0114 (units), just a report attributes 0x00f7, with an octet string.  
There is no on-device change of units available.  
I do not know if there is a query attributes command to read the current value for units?

Koenkk, 2022-09-19:
With the converter below a refresh icon should show up the z2m frontend -> device -> exposes page, what is logged in the debug log once you click it?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

```js
const constants = require('zigbee-herdsman-converters/lib/constants');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    VOCKQJK11LM_display_unit: {
        key: ['display_unit'],
        convertSet: async (entity, key, value, meta) => {
            // Match key names from exposes.enum
            const lookup = {
                'mgm3_celsius': 0x00, // mg/m³, °C (default)
                'ppb_celsius': 0x01, // ppb, °C
                'mgm3_fahrenheit': 0x10, // mg/m³, °F
                'ppb_fahrenheit': 0x11 // ppb, °F
            }
            await entity.write('aqaraOpple', { 0x0114: { value: lookup[value], type: 0x20 } }, { manufacturerCode: 0x115F, disableDefaultResponse: true });
            return { state: { display_unit: value } };
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('aqaraOpple', [0x0114], { manufacturerCode: 0x115F, disableDefaultResponse: true });
        },
    }
}

const definition = {
    zigbeeModel: ['lumi.airmonitor.acn01'],
    model: 'VOCKQJK11LM',
    vendor: 'Xiaomi',
    whiteLabel: [{ vendor: 'Xiaomi', model: 'AAQS-S01' }],
    description: 'Aqara TVOC air quality monitor',
    fromZigbee: [fz.xiaomi_tvoc, fz.battery, fz.temperature, fz.humidity, fz.aqara_opple],
    toZigbee: [tzLocal.VOCKQJK11LM_display_unit],
    meta: { battery: { voltageToPercentage: '3V_2850_3000' } },
    exposes: [e.temperature(), e.humidity(), e.voc(), e.device_temperature(), e.battery(), e.battery_voltage(),
    exposes.enum('display_unit', ea.STATE_SET, ['mgm3_celsius', 'ppb_celsius', 'mgm3_fahrenheit', 'ppb_fahrenheit'])
        .withDescription('Units to show on the display')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genAnalogInput'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.humidity(endpoint);
        await reporting.temperature(endpoint);
        const payload = reporting.payload('presentValue', 10, constants.repInterval.HOUR, 5);
        await endpoint.configureReporting('genAnalogInput', payload);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

ptr727, 2022-09-20:
No refresh icon?

[image]

Koenkk, 2022-09-20:
Whoops; updated https://github.com/Koenkk/zigbee2mqtt/issues/13464#issuecomment-1251165859

ptr727, 2022-09-20:
Ok, press refresh, press button on device.
[z2mlog.log](https://github.com/Koenkk/zigbee2mqtt/files/9609252/z2mlog.log)
[_zigbee2mqtt_logs.log](https://github.com/Koenkk/zigbee2mqtt/files/9609253/_zigbee2mqtt_logs.log)

Koenkk, 2022-09-21:
Fixed that now and integrated it in the dev branch.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ptr727, 2022-10-01:
Confirmed working in ` 1.28.0 commit: 03ba647`.  
Thank you

closed 2022-10-01 by ptr727