Koenkk/zigbee2mqtt #13465: Support for Tuya USB Zigbee Repeater - Device wrongly detected analog 4329

CPlusPlus17, 2022-08-08:
### What happened?

The device is detected as https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector but should be https://www.zigbee2mqtt.io/devices/TS0207_repeater.html. This problem was noted here and fixed later: https://github.com/Koenkk/zigbee2mqtt/issues/4329

### What did you expect to happen?

Device should be recognized as TuYa TS0207_repeater.

### How to reproduce it (minimal and precise)

Pair a TuYa TS0207_repeater and it's displayed as follow:

Device type: Router
Zigbee Model: TS0207
Zigbee Manufacturer: _TZ3000_nkkl7uzv
Description: Water leak detector
Support status: Supported

IEEE Address: 0xb4e3f9fffe3e0669
Network address: 0x9F54

### Zigbee2MQTT version

 1.27.0-1

### Adapter firmware version

0x26720700 

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-08-09:
Added!

closed 2022-08-09 by Koenkk

KoenChiau, 2022-09-08:
Same issue with newly received USB Zigbee  Repeater.

I believe _TZ3000_gszjt2xx also needs to be added to TS027_repeater.

After interview, this is the result:
- Friendly name:               0x04cd15fffe0a2496
- Description:
- Last seen:                     N/A
- Availability:                    [Disabled] 
- Device type:                  Router
- Zigbee Model:               TS0207
- Zigbee Manufacturer:   _TZ3000_gszjt2xx
- Description:                   Water leak detector
- Support status:              Supported
- IEEE Address:                 0x04cd15fffe0a2496
- Network address:           0x3035
- Manufacturer:                 [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
- Model:                             [TS0207_water_leak_detector](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector)
- Power:
- Interview completed       True

janfuhlrott, 2023-01-12:
Same here today: Repeater recognized as Water Leak sensor (sorry for german labels)

Geräte-Typ: Router
Zigbee Modell: TS0207
Zigbee Hersteller: _TZ3000_misw04hq
Beschreibung: Water leak detector
Unterstützungsstatus: Unterstützt
IEEE Addresse: 0x1c34f1fffeb75f7b
Netzwerk Adresse: 0x35C6
Hersteller: [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
Modell: [TS0207_water_leak_detector](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector)
Spannungsversorgung: 
Interview erfolgreich: Wahr

MonacledMonkey, 2023-01-12:
I have also today received the "TZ3000_misw04hq" variant of TS0207, being recognised as a water leak sensor in the HA Edge version.
IEEE address: 0x1c34f1fffeb76556
Network Address: 0xDC2F