Koenkk/zigbee2mqtt #13466: [New device support]: Skydance WZ5 - TS0503B based

jmbinette, 2022-08-09:
### Link

https://www.zigbee2mqtt.io/devices/WZ5_rgb.html

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c1389a71adc225","nwkAddr":50188,"manufId":4417,"manufName":"_TZB210_zdvrsts8","powerSource":"Mains (single phase)","modelId":"TS0503B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503B","manufacturerName":"_TZB210_zdvrsts8","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":9}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1660014904464,"defaultSendRequestWhen":"immediate"}

### Comments

I have succesfully used the Skydance WZ5-RGB converter.

I received a new batch with the TS0503B model and _TZB210_zdvrsts8 manufacturer.

I tried adding and external converter that is a copy of skydance.js but with adding the "_TZB210_zdvrsts8" signature under the RGB section. _This does not work_

I tried adding and external converter that is a copy of tuya.js but with adding the "_TZB210_zdvrsts8" signature under the TS0503B-RGB section. ON/OFF and brightness works but colors seems off.

### External converter

```shell
Added after line in #40 skydance.js
            {modelID: 'TS0503B', manufacturerName: '_TZB210_zdvrsts8'}

Added after line #299 in tuya.js :
            {modelID: 'TS0503B', manufacturerName: '_TZB210_zdvrsts8'}],
```

### Supported color modes

color rgb

jmbinette, 2022-08-09:
Do I need to modify the converter ?
Seems like Hue/Sat is not exposed : 

effect_list: blink, breathe, okay, channel_change, finish_effect, stop_effect
supported_color_modes: xy
friendly_name: WZ5_7
supported_features: 61

jmbinette, 2022-08-09:
I don't have hs control in the frontend. Something I am doing wrong ?

[image]

Koenkk, 2022-08-09:
Can you check with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0503B', manufacturerName: '_TZB210_zdvrsts8'}],
  model: 'TS0503B',
  vendor: 'TuYa',
  description: 'Zigbee RGB light',
  extend: extend.light_onoff_brightness_color({supportsHS: true}),
  // Requires red fix: https://github.com/Koenkk/zigbee2mqtt/issues/5962#issue-796462106
  meta: {applyRedFix: true, enhancedHue: false},
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

jmbinette, 2022-08-09:
Of course, will try tonight

Thanks !

jmbinette, 2022-08-10:
I tested tonight with my controller.

What works :
- State
- brightness
- color_hs
- link quality

What does not work :
- color_xy (wrong colors)
- effect (No effects)

Koenkk, 2022-08-10:
> color_xy (wrong colors)

Seems the fw does not implement the x/y colors correctly, nothing we can do from z2m here

> effect (No effects)

Looks that fw does not support it, disabled them now.

Updated https://github.com/Koenkk/zigbee2mqtt/issues/13466?notification_referrer_id=NT_kwDOACwkNbI0MTU5MDg1NjA0OjI4OTI4NTM#issuecomment-1209662389

jmbinette, 2022-08-10:
Thanks

Any way to have HA control it through HS as default ?
Seems like it changes in xy mode when testing via HA developper mode

Koenkk, 2022-08-10:
@jmbinette yes, I've updated https://github.com/Koenkk/zigbee2mqtt/issues/13466#issuecomment-1209662389

jmbinette, 2022-08-10:
Works well ! Thanks again !

Do I need to to a PR or will be integrated in the future ?

closed 2022-08-11 by Koenkk

jmbinette, 2022-08-11:
Thank you !

As a side note : what would be needed to support effects? The datapoints ? I could do some testing

Koenkk, 2022-08-11:
> As a side note : what would be needed to support effects? The datapoints ? I could do some testing

I expect this device doesn't support effects (I havent seen any TuYa light which does), but to confirm it has to be checked with the original hub.

jmbinette, 2022-08-11:
Thanks for your help

I will order myself a tuya hub and do some testing as I plan to use a couple of these WZ5 devices.

Thanks !