Koenkk/zigbee2mqtt #13467: Network devices unable to connect after reboot

StevenJonSmith, 2022-08-09:
### What happened?

All bar one device stopped functioning entirely.

### What did you expect to happen?

Devices to continue functioning as they had been after network build.

### How to reproduce it (minimal and precise)

Have a working network, reboot Home Assistant OS on Raspberry Pi.

Please note I am aware the RaspBee II firmware is old, but I have recently had an issue rebuilding my network from channel 11 to channel 25 where the devices would not connect because the RaspBee was not starting with the right channel. The fix which worked was to rollback the firmware and change the channel, I was then able to reconnect my devices and they all functioned fine until the Pi rebooted due to a power outage.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

26610700

### Adapter

Raspbee II

### Debug log

```
Zigbee2MQTT:error 2022-08-09 10:04:35: Publish 'set' 'state' to 'Floor lamp' failed: 'Error: Command 0x000d6ffffe2cfdcd/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)'
Zigbee2MQTT:debug 2022-08-09 10:04:35: Error: Command 0x000d6ffffe2cfdcd/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:654:23)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:772:28)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:283:17)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Zigbee2MQTT:info  2022-08-09 10:04:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Floor lamp' failed: 'Error: Command 0x000d6ffffe2cfdcd/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received)'","meta":{"friendly_name":"Floor lamp"},"type":"zigbee_publish_error"}'
```

adorobis, 2022-09-06:
Did you manage to resolve this problem? My zigbee network also started to behave this way for all cable powered devices (plugs, switches etc).

StevenJonSmith, 2022-09-06:
Unfortunately not.

Bizarrely some sensors continue to work fine, some not at all, and some only part of the device reports correctly. For example an outdoor Phillips Hue PIR reports luminosity fine, but not occupation or temperature. It doesn't make any sense, and I don't even really know where to start with troubleshooting.

I tried removing and reading a device to the network. It seemed to remove fine, but now won't add back in. I'm really annoyed, but I'm starting to think it might be a HA OS thing as it's also not working in ZHA. I have been considering getting rid of the RaspBee and swapping to a USB stick coordinator, but I'm hesitant as I can't narrow it down to the RaspBee yet, and since some work it makes me think it must be functioning.

adorobis, 2022-09-08:
My problem got resolved after I have upgraded Conbee II firmware and switch z2m to a dev branch. I've made two changes at the same time so not sure which one resolved the problem though...

closed 2022-10-16 by github-actions[bot]