Koenkk/zigbee2mqtt #13471: [New device support]:

Millennium-bug-82, 2022-08-09:
### Link

https://www.lidl.nl/p/livarno-home-outdoor-led-band-zigbee-smart-home/p100341806

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x04cd15fffe789435","nwkAddr":16102,"manufId":4098,"manufName":"_TZ3210_zbabx9wh","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_zbabx9wh","powerSource":1,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentHue":0,"colorMode":2,"options":0,"currentX":0,"tuyaRgbMode":0,"startUpColorTemperature":153,"colorCapabilities":25}},"genIdentify":{"attributes":{"identifyTime":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"manuSpecificTuya_3":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1660066227018,"defaultSendRequestWhen":"immediate"}

### Comments

I bought a few of the new LIDL/Livorno outdoor LED strips (or bands as they call them) hoping they'd work with zigbee2mqtt and be of decent quality. They pair easily but without support they only do cold white and nothing else, no hardware buttons to play with. They seem quite nice actually so I was hoping we could get them supported. Unfortunately I don't know how to add the support myself, made worse by the fact that they appear to be of some Tuya variant. I don't have a LIDL/Tuya bridge so I can't get much out of them. They do respond to some read requests in the Dev console so that might be of some help. Any chance of support or should I return them?

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2022-08-09:
Could you check if it works with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_zbabx9wh'}],
  model: 'TS0505B',
  vendor: 'TuYa',
  description: 'Zigbee RGB+CCT light',
  extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500], disableColorTempStartup: true}),
  meta: {applyRedFix: true, enhancedHue: false},
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

Millennium-bug-82, 2022-08-11:
Thanks for your fast response. I wasn't able to test yesterday but tried just now and everything works as it should. This is great! Thank you very much!

The LED bands are actually quite nice like a lot of LIDL smart home stuff so worth checking out.

Koenkk, 2022-08-11:
Great, what is the model number of this device? I expect it starts with `HG` (e.g. `HG06104A`)

Millennium-bug-82, 2022-08-11:
Okay, it was hidden on the small sticker and found nowhere else but I finally found it: HG08007.

Koenkk, 2022-08-12:
Great, added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-12 by Koenkk

PuckStar, 2022-08-27:
Hi @Koenkk . I got it added with your workaround but it only shows as normal light. So I can only turn on/off and can't change any colors. Am I doing something wrong?  Thanks

Koenkk, 2022-08-27:
@PuckStar can you check via the exposes tab in the z2m frontend?

PuckStar, 2022-08-28:
> @PuckStar can you check via the exposes tab in the z2m frontend?

There it looks good. There is an option to change colors.
I then looked again at the light icon in Home Assistant and strangely there I can now also change to all colors.
Not sure if my looking at the exposed part triggered this in Home Assistant or if it just took longer in any case to show correctly in Home Assistant but it's now all good! :)

Also now that I think of it. I see you can only change colors after turning on the light and only by clicking on the 3 dots.
Maybe earlier I was only looking at changing colors by clicking on the light icon which then of course turned the light off (and on if pressed again etc.). So maybe I was looking wrong.

Thanks for your reply as it made me look again and better :)

RolandGoud, 2024-08-26:
Does anybody know how to reset this strip? Tried switching it on and of by cutting the power but it doesnt seem to be workjing

RolandGoud, 2024-08-26:
omg, it worked. The trick is to wait 2-3 seconds on each switch on/off