Koenkk/zigbee2mqtt #13472: [New device support]: Legrand 067646L

samohthou, 2022-08-09:
### Link

https://www.legrand.fr/pro/catalogue/44692-version-celiane-with-netatmo/commande-sans-fils-sans-pile-self-e-celiane-with-netatmo-pour-volets-roulants-connectes-enjoliveur-blanc

### Database entry

{"id":6,"type":"GreenPower","ieeeAddr":"0x0000000000529efa","nwkAddr":40698,"manufId":null,"modelId":"GreenPower_3","epList":[242],"endpoints":{"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1660070599526,"defaultSendRequestWhen":"immediate"}

### Comments

Hello,
I can't add the Legrand switch for rolling shutter 067646L from Legrand. 
I tried to add it by taking example on the existing switches like Legrand 067646 or Legrand ZLGP15 by following the documentation "Support new devices" but without success. 
Can you help me?
Thanks in advance.

samohthou, 2022-09-09:
Can someone help me solve my problem ?

Loic691, 2022-09-11:
hi
Do you have a 067646 device ? Is it working ?

samohthou, 2022-09-11:
Hello, 
And thank you for your feedback. 
No I don't have a 067646 device. I have 067646L device and is appears as "not supported". But I don't have any problem for devices like 067723L and 067724L.

samohthou, 2022-09-13:
Hello,
To complete my request, I tried to create a file 067646.js with the following information:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GreenPower_3'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '067646', // Vendor model number, look on the device for a model number
    vendor: 'Legrand', // Vendor of the device (only used for documentation and startup logging)
    description: 'Shutters central remote switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.identify, fz.ignore_basic_report, fz.command_cover_open, fz.command_cover_close, fz.command_cover_stop, fz.legrand_binary_input_moving], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.action(['identify', 'open', 'close', 'stop', 'moving', 'stopped'])],
};

module.exports = definition;
```

After calling this file in the configuration.yaml file and restarting, the device appears as "supported", but here is what is returned in the log:

- When I press the "stop" button:

`Debug 2022-09-14 01:20:20Received Zigbee message from '0x0000000000529efa', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":52,"frameCounter":668,"options":0,"payloadSize":0,"srcID":5414650}' from endpoint 242 with groupID 2948
Debug 2022-09-14 01:20:20No converter available for '067646' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{},"commandID":52,"frameCounter":668,"options":0,"payloadSize":0,"srcID":5414650}'`

- When I press the "up" button:

`Debug 2022-09-14 01:21:00Received Zigbee message from '0x0000000000529efa', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":53,"frameCounter":669,"options":0,"payloadSize":0,"srcID":5414650}' from endpoint 242 with groupID 2948
Debug 2022-09-14 01:21:00No converter available for '067646' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{},"commandID":53,"frameCounter":669,"options":0,"payloadSize":0,"srcID":5414650}'`

- When I press the "down" button:

`Debug 2022-09-14 01:21:43Received Zigbee message from '0x0000000000529efa', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{},"commandID":54,"frameCounter":670,"options":0,"payloadSize":0,"srcID":5414650}' from endpoint 242 with groupID 2948
Debug 2022-09-14 01:21:43No converter available for '067646' with cluster 'greenPower' and type 'commandNotification' and data '{"commandFrame":{},"commandID":54,"frameCounter":670,"options":0,"payloadSize":0,"srcID":5414650}'`

samohthou, 2022-10-02:
Is there a possible solution to my problem ?

Koenkk, 2022-10-04:
Try with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const utils = require('zigbee-herdsman-converters/lib/utils');
const ea = exposes.access;

const fzLocal = {
    conv: {
        cluster: 'greenPower',
        type: ['commandNotification', 'commandCommisioningNotification'],
        convert: (model, msg, publish, options, meta) => {
            const commandID = msg.data.commandID;
            if (utils.hasAlreadyProcessedMessage(msg, msg.data.frameCounter, `${msg.device.ieeeAddr}_${commandID}`)) return;
            if (commandID === 224) return; // Skip commisioning command.

            // Button 1: A0 (top left)
            // Button 2: A1 (bottom left)
            // Button 3: B0 (top right)
            // Button 4: B1 (bottom right)
            const lookup = {
               // 0x10: 'press_1', 0x14: 'release_1', 0x11: 'press_2', 0x15: 'release_2', 0x13: 'press_3', 0x17: 'release_3',
            };

            if (!lookup.hasOwnProperty(commandID)) {
                meta.logger.error(`067646: missing command '${commandID}'`);
            } else {
                return {action: lookup[commandID]};
            }
        },
    },
}

const definition = {
    zigbeeModel: ['GreenPower_3'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '067646', // Vendor model number, look on the device for a model number
    vendor: 'Legrand', // Vendor of the device (only used for documentation and startup logging)
    description: 'Shutters central remote switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fzLocal.conv], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.action(['identify', 'open', 'close', 'stop', 'moving', 'stopped'])],
};

module.exports = definition;
```

When clicking the button you will get `missing command..` in the logs, this is what you need to add to the `const lookup = {`

samohthou, 2022-10-04:
Hello, 
And thank you for your feedback, it works perfectly !

closed 2022-10-04 by samohthou