Koenkk/zigbee2mqtt #13473: All new devices left the network immediately after click permit join

Nikolay-Lipovtsev, 2022-08-09:
### What happened?

I create new Home Assistant server and can't add any devices, because all new devises left the network immediately after click permit join.

My server:

Pi4B
Home Assistant 2022.8.1
Supervisor 2022.07.0
Operating System 8.4
Mosquitto broker 6.1.2
Zigbee2MQTT 1.27.0-1
Coordinator:
CC2652P
zStack3x0
firmware version 20220726
firmware: [Z-Stack-firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin)

/config/zigbee2mqtt/configuration.yaml

homeassistant: true
mqtt:
  server: mqtt://core-mosquitto:1883
  user: ***
  password: ***
  keepalive: 60
  reject_unauthorized: true
  version: 4
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099
Device:
BlitzWolf mart Socket BW-SHP13

### What did you expect to happen?

Join new device

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20220726

### Adapter

Egony Stick V4 (Ebyte ver.) CC2652P

### Debug log

Zigbee2MQTT:info 2022-08-06 14:33:08: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info 2022-08-06 14:33:08: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:info 2022-08-06 14:33:20: zigbee-herdsman started (resumed)
Zigbee2MQTT:info 2022-08-06 14:33:20: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220726,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info 2022-08-06 14:33:20: Currently 0 devices are joined:
Zigbee2MQTT:info 2022-08-06 14:33:20: Zigbee: disabling joining new devices.
Zigbee2MQTT:info 2022-08-06 14:33:20: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:info 2022-08-06 14:33:20: Connected to MQTT server
Zigbee2MQTT:info 2022-08-06 14:33:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info 2022-08-06 14:33:20: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info 2022-08-06 14:33:20: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220726,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.27.0"}'
Zigbee2MQTT:info 2022-08-06 14:33:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info 2022-08-06 14:33:59: Zigbee: allowing new devices to join.
Zigbee2MQTT:info 2022-08-06 14:33:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"v1pl0-1"}'
Zigbee2MQTT:warn 2022-08-06 14:34:07: Device '0x60a423fffead67a4' left the network
Zigbee2MQTT:info 2022-08-06 14:34:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffead67a4","ieee_address":"0x60a423fffead67a4"},"type":"device_leave"}'
Zigbee2MQTT:info 2022-08-06 14:34:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x60a423fffead67a4"},"type":"device_removed"}'
Zigbee2MQTT:warn 2022-08-06 14:34:07: Device '0x60a423fffead67a4' left the network
Zigbee2MQTT:info 2022-08-06 14:34:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffead67a4","ieee_address":"0x60a423fffead67a4"},"type":"device_leave"}'
Zigbee2MQTT:info 2022-08-06 14:34:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x60a423fffead67a4"},"type":"device_removed"}'
Zigbee2MQTT:warn 2022-08-06 14:34:08: Device '0x60a423fffead67a4' left the network
Zigbee2MQTT:info 2022-08-06 14:34:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffead67a4","ieee_address":"0x60a423fffead67a4"},"type":"device_leave"}'
Zigbee2MQTT:info 2022-08-06 14:34:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x60a423fffead67a4"},"type":"device_removed"}'

Koenkk, 2022-08-10:
It seems that only `0x60a423fffead67a4` leaves, right?

Nikolay-Lipovtsev, 2022-08-10:
tested on it mainly

Nikolay-Lipovtsev, 2022-08-11:
I can't join other devices. Today I try Aqara SJCGQ11LM and MOES BHT-002-GCLZB. Nothing happens after I click on the pairing button in the z2m interface and on the device

Koenkk, 2022-08-12:
Assuming you have checked all points of https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair already, can you provide me a sniff when trying to pair a device?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Nikolay-Lipovtsev, 2022-08-13:
Today I try Aqara B1 curtain motor ZNCLDJ12LM.
At first it was identified as an unknown device. Then I tried to delete it, but it was deleted only with the force option.
Then I tried to reconnect it, but it didn't connect.

```s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[16:42:35] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[16:42:40] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:info  2022-08-07 16:42:54: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-07.16-42-54' filename: log.txt
Zigbee2MQTT:info  2022-08-07 16:42:54: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-07 16:42:54: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:info  2022-08-07 16:43:39: zigbee-herdsman started (restored)
Zigbee2MQTT:info  2022-08-07 16:43:39: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-08-07 16:43:39: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-08-07 16:43:39: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-07 16:43:40: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:info  2022-08-07 16:43:41: Connected to MQTT server
Zigbee2MQTT:info  2022-08-07 16:43:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-08-07 16:43:42: Started frontend on port 0.0.0.0:8099
Zigbee2MQTT:info  2022-08-07 16:43:42: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.27.0"}'
Zigbee2MQTT:info  2022-08-07 16:43:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2022-08-07 16:44:33: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-08-07 16:44:33: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ebd82-1"}'
Zigbee2MQTT:warn  2022-08-07 16:44:45: Device '0x60a423fffead67a4' left the network
Zigbee2MQTT:info  2022-08-07 16:44:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffead67a4","ieee_address":"0x60a423fffead67a4"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-08-07 16:44:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x60a423fffead67a4"},"type":"device_removed"}'
Zigbee2MQTT:warn  2022-08-07 16:44:46: Device '0x60a423fffead67a4' left the network
Zigbee2MQTT:info  2022-08-07 16:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x60a423fffead67a4","ieee_address":"0x60a423fffead67a4"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-08-07 16:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x60a423fffead67a4"},"type":"device_removed"}'
Zigbee2MQTT:info  2022-08-07 16:45:11: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-07 16:45:11: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"ebd82-2"}'
Zigbee2MQTT:info  2022-08-07 16:47:13: Start Touchlink scan
Zigbee2MQTT:info  2022-08-07 16:47:24: Finished Touchlink scan
Zigbee2MQTT:info  2022-08-07 16:47:24: MQTT publish: topic 'zigbee2mqtt/bridge/response/touchlink/scan', payload '{"data":{"found":[]},"status":"ok","transaction":"ebd82-3"}'
Zigbee2MQTT:info  2022-08-11 21:09:17: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-08-11 21:09:18: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"zmbyp-1"}'
Zigbee2MQTT:info  2022-08-11 21:11:01: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-11 21:11:01: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"f8kfs-1"}'
Zigbee2MQTT:info  2022-08-11 22:06:15: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-08-11 22:06:15: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"aqq2r-1"}'
Zigbee2MQTT:info  2022-08-11 22:07:01: Zigbee: disabling joining new devices.
[... cut ...]
Zigbee2MQTT:info  2022-08-13 11:01:22: Successfully removed device '0x04cf8cdf3c8f0edb' (block: false, force: true)
Zigbee2MQTT:info  2022-08-13 11:01:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x04cf8cdf3c8f0edb"},"status":"ok","transaction":"9rcce-2"}'
Zigbee2MQTT:info  2022-08-13 11:01:28: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-08-13 11:01:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"9rcce-3"}'```

guinola, 2022-08-25:
Hi, I have the same problem. Have you been able to solve it?

closed 2022-10-02 by github-actions[bot]