Koenkk/zigbee2mqtt #13474: NAS-AB02B0 (NeoSiren) keeps blinking after pairing (no command 'resetDevice', no converter available)

Piscator74, 2022-08-09:
### What happened?

I already had a NAS-AB02B0 (NeoSiren) lying around for a year and since I saw on the forum issues were resolved and fixed (https://github.com/Koenkk/zigbee2mqtt/issues/7051, https://github.com/Koenkk/zigbee2mqtt/issues/7407, https://github.com/Koenkk/zigbee2mqtt/issues/12225), I expected it to work now.

In the logging, I see it has successfully paired but the device keeps blinking and does not send any MQTT publish (also not link quality)

I enabled debug logging (see log) and I see various error after repairing the device.
1st attempt. Error: Cluster 'manuSpecificTuya' has no command 'resetDevice'
2nd attempt. No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":8192}'

Any ideas what I can try to debug the problem / fix the problem?

### What did you expect to happen?

The device stops blinking after successful pairing and publishes MQTT topics.

### How to reproduce it (minimal and precise)

NA

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB stick

### Debug log

debug 2022-08-09 20:16:17: Loaded state from file /app/data/state.json
info  2022-08-09 20:16:17: Logging to console and directory: '/app/data/log/2022-08-09.20-16-17' filename: log.txt
debug 2022-08-09 20:16:17: Removing old log directory '/app/data/log/2022-07-27.13-19-42'
info  2022-08-09 20:16:17: Starting Zigbee2MQTT version 1.27.0 (commit #a9b8808)
info  2022-08-09 20:16:17: Starting zigbee-herdsman (0.14.46)
debug 2022-08-09 20:16:17: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[26],"extendedPanID":[33,117,141,25,0,75,18,0],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB1"}}'
warn  2022-08-09 20:16:25: Extended PAN ID is reversed (expected=21758d19004b1200, actual=00124b00198d7521)
info  2022-08-09 20:16:28: zigbee-herdsman started (resumed)
info  2022-08-09 20:16:28: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"}'
debug 2022-08-09 20:16:28: Zigbee network parameters: {"channel":26,"extendedPanID":"0x00124b00198d7521","panID":6754}
info  2022-08-09 20:16:28: Currently 61 devices are joined:
<snip>
info  2022-08-09 20:16:28: 0x804b50fffe047535 (0x804b50fffe047535): NAS-AB02B0 - Neo Temperature & humidity sensor and alarm (EndDevice)
warn  2022-08-09 20:16:28: `permit_join` set to  `true` in configuration.yaml.
warn  2022-08-09 20:16:28: Allowing new devices to join.
warn  2022-08-09 20:16:28: Set `permit_join` to `false` once you joined all devices.
info  2022-08-09 20:16:28: Zigbee: allowing new devices to join.
info  2022-08-09 20:16:28: Connecting to MQTT server at mqtt://localhost
debug 2022-08-09 20:16:28: Using MQTT anonymous login
info  2022-08-09 20:16:28: Connected to MQTT server
info  2022-08-09 20:16:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
error 2022-08-09 20:16:28: Failed to call 'OTAUpdate' 'start' (TypeError: zigbeeOTA_1.default.setDataDir is not a function
    at OTAUpdate.start (/app/lib/extension/otaUpdate.ts:58:19)
    at Controller.callExtensions (/app/lib/controller.ts:315:40)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at Controller.start (/app/lib/controller.ts:154:9)
    at start (/app/index.js:109:5))
debug 2022-08-09 20:16:28: Received MQTT message on 'zigbee2mqtt/bridge/config' with data '{"commit":"a9b8808","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":26,"extendedPanID":"0x00124b00198d7521","panID":6754},"permit_join":true,"version":"1.27.0"}'
info  2022-08-09 20:16:28: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"a9b8808","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":26,"extendedPanID":"0x00124b00198d7521","panID":6754},"permit_join":true,"version":"1.27.0"}'

<snip, pressed reset 1st time>
warn  2022-08-09 20:17:18: Device '0x804b50fffe047535' left the network
info  2022-08-09 20:17:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe047535","ieee_address":"0x804b50fffe047535"},"type":"device_leave"}'
info  2022-08-09 20:17:18: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x804b50fffe047535"},"type":"device_removed"}'
<snip>
info  2022-08-09 20:17:30: Device '0x804b50fffe047535' joined
[... 57 lines cut ...]

warn  2022-08-09 20:25:28: Device '0x804b50fffe047535' left the network
info  2022-08-09 20:25:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe047535","ieee_address":"0x804b50fffe047535"},"type":"device_leave"}'
info  2022-08-09 20:25:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x804b50fffe047535"},"type":"device_removed"}'
<snip>
info  2022-08-09 20:25:40: Device '0x804b50fffe047535' joined
info  2022-08-09 20:25:40: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe047535","ieee_address":"0x804b50fffe047535"},"type":"device_joined"}'
info  2022-08-09 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x804b50fffe047535"},"type":"device_connected"}'
info  2022-08-09 20:25:41: Starting interview of '0x804b50fffe047535'
info  2022-08-09 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe047535","ieee_address":"0x804b50fffe047535","status":"started"},"type":"device_interview"}'
info  2022-08-09 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x804b50fffe047535"},"type":"pairing"}'
debug 2022-08-09 20:25:41: Device '0x804b50fffe047535' announced itself
info  2022-08-09 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe047535","ieee_address":"0x804b50fffe047535"},"type":"device_announce"}'
info  2022-08-09 20:25:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x804b50fffe047535"},"type":"device_announced"}'
debug 2022-08-09 20:25:44: Received Zigbee message from '0x804b50fffe047535', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_d0yu2xgi","modelId":"TS0601"}' from endpoint 1 with groupID 0
debug 2022-08-09 20:25:44: Received Zigbee message from '0x804b50fffe047535', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2022-08-09 20:25:45: Received Zigbee message from '0x804b50fffe047535', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2022-08-09 20:25:45: Received Zigbee message from '0x804b50fffe047535', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2022-08-09 20:25:45: Successfully interviewed '0x804b50fffe047535', device has successfully been paired
info  2022-08-09 20:25:45: Device '0x804b50fffe047535' is supported, identified as: Neo Temperature & humidity sensor and alarm (NAS-AB02B0)
info  2022-08-09 20:25:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor and alarm","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":3,"name":"humidity_alarm","property":"humidity_alarm","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"temperature_alarm","property":"temperature_alarm","type":"binary","value_off":false [... cut ...]
debug 2022-08-09 20:25:46: Received Zigbee message from '0x804b50fffe047535', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2022-08-09 20:25:51: Received Zigbee message from '0x804b50fffe047535', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":8192}' from endpoint 1 with groupID 0
debug 2022-08-09 20:25:51: No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":8192}'

Piscator74, 2022-09-07:
Anyone (after the summer holidays) has any idea / suggestion what I can try to debug this problem?

iqbalaydrus, 2022-09-27:
My siren was disconnected after 1 year, trying to re-pair the siren indeed didn't work as described above. But when I deleted the device from zigbee2mqtt and start pairing, it worked.

Piscator74, 2022-10-04:
I force deleted it with "zigbee2mqttassistant", but that does not make a difference (it gives the same error messages). How did you delete the device?

iqbalaydrus, 2022-10-07:
@Piscator74 I did it via the GUI. I forgot though whether I used normal delete or force delete.

closed 2022-11-15 by github-actions[bot]