Koenkk/zigbee2mqtt #13478: Popp/Danfoss TRVs become unresponsive with CC2652P

moster11, 2022-08-09:
### What happened?

I have a setup with several Popp-TRVs (identical to Danfoss Ally TRVs), lots of Aqara sensors and two routers, all connected to a Sonoff USB Dongle 3.0 Plus. I run Home Assistant and have automations that send temperatures measured by the Aqara sensors to the external_measured_room_sensor endpoint of the TRVs.

Everything runs smoothly for some time - until the TRVs stop reacting to any command sent to them. They still report their states without any issue, it's just that I cannot send any command to them any more. Most notably this ONLY happens to TRVs which are connected directly to the coordinator, never if connected to a router.

Looking at the devices tab shows recent "Last seen" values. I highlighted "Thermostat Badezimmer Fenster" as that's the one that's unresponsive (did an initial re-pairing of all devices this morning):

[image]

Looking at the map gives me a clear sign that this TRV will not react to my commands anymore. LQI is shown as 0 here, while the previous picture showed LQI 72:

[image]

I tried Z2M versions 1.27.0 and 1.26.0. The Sonoff dongle was on ZStack 20220219, now upgraded to dev branch 20220726. I resetted and repaired all devices numerous times, even changed to position of the coordinator to the other side of the house. Nothing helps, it's always the devices connected to the coordinator that stop reacting after some hours, at max one or two days. Only way to make them react again for some time is to take the batteries in and out.

I completely reinstalled Z2M 1.27.0 and connected all devices to the coordinator directly. The result is the same - after some hours the first TRVs become unresponsive but happily send their states.

In the debug log can see some "data confirm errors" which might give a clue.

### What did you expect to happen?

I expect the devices always react to the commands I send to them and not become unresponsive.

### How to reproduce it (minimal and precise)

Just setup the TRVs, make sure they are connected to the coordinator, and wait some hours.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219 and dev 20220726

### Adapter

Sonoff USB Dongle 3.0 plus

### Debug log

2022-08-09T22:17:35.975Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-08-09T22:17:35.975Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-08-09T22:17:35.975Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-08-09T22:17:35.975Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:37.682Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,13,6]
2022-08-09T22:17:37.682Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,13,6]
2022-08-09T22:17:37.684Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,13] - 6
2022-08-09T22:17:37.684Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":13}
2022-08-09T22:17:37.684Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:37.685Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xb4e3f9fffedf0571:10453,205,2)
2022-08-09T22:17:37.685Z zigbee-herdsman:adapter:zStack:adapter Request network address of '0xb4e3f9fffedf0571'
2022-08-09T22:17:37.686Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nwkAddrReq - {"ieeeaddr":"0xb4e3f9fffedf0571","reqtype":0,"startindex":0}
2022-08-09T22:17:37.686Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,10,37,0,113,5,223,254,255,249,227,180,0,0,43]
2022-08-09T22:17:37.701Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,0,0,100]
2022-08-09T22:17:37.701Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,0,0,100]
2022-08-09T22:17:37.701Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 0 - [0] - 100
2022-08-09T22:17:37.701Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nwkAddrReq - {"status":0}
2022-08-09T22:17:37.701Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:40.813Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,0,0,32,0,113,235,1,242,0,87,0,126,88,7,0,0,3,9,96,0,113,235,29,0]
2022-08-09T22:17:40.814Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,32,0,113,235,1,242,0,87,0,126,88,7,0,0,3,9,96,0,113,235,29,0]
2022-08-09T22:17:40.814Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,32,0,113,235,1,242,0,87,0,126,88,7,0,0,3,9,96,0,113,235,29] - 0
2022-08-09T22:17:40.815Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":32,"srcaddr":60273,"srcendpoint":1,"dstendpoint":242,"wasbroadcast":0,"linkquality":87,"securityuse":0,"timestamp":481406,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[9,96,0]}}
2022-08-09T22:17:40.816Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":96,"manufacturerCode":null,"commandIdentifier":0},"Payload":{},"Command":{"ID":0,"parameters":[],"name":"checkin"}},"address":60273,"endpoint":1,"linkquality":87,"groupID":0,"wasBroadcast":false,"destinationEndpoint":242}'
2022-08-09T22:17:40.818Z zigbee-herdsman:controller:device:log check-in from 0x0c4314fffe45317c: declining fast-poll
2022-08-09T22:17:40.818Z zigbee-herdsman:controller:endpoint Command 0x0c4314fffe45317c/1 genPollCtrl.checkinRsp({"startFastPolling":false,"fastPollTimeout":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-08-09T22:17:40.818Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0c4314fffe45317c:60273/1 (0,0,2)
2022-08-09T22:17:40.819Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":60273,"destendpoint":1,"srcendpoint":1,"clusterid":32,"transid":14,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,6,0,0,0,0]}}
2022-08-09T22:17:40.819Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,113,235,1,1,32,0,14,0,30,6,1,6,0,0,0,0,158]
[... 47 lines cut ...]
2022-08-09T22:17:46.675Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:46.675Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x0c4314fffe45317c:60273,233,2)
2022-08-09T22:17:46.675Z zigbee-herdsman:adapter:zStack:adapter Request network address of '0x0c4314fffe45317c'
2022-08-09T22:17:46.675Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nwkAddrReq - {"ieeeaddr":"0x0c4314fffe45317c","reqtype":0,"startindex":0}
2022-08-09T22:17:46.676Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,10,37,0,124,49,69,254,255,20,67,12,0,0,125]
2022-08-09T22:17:46.682Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,0,0,100]
2022-08-09T22:17:46.682Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,0,0,100]
2022-08-09T22:17:46.682Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 0 - [0] - 100
2022-08-09T22:17:46.682Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nwkAddrReq - {"status":0}
2022-08-09T22:17:46.683Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:47.703Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xb4e3f9fffedf0571:10453/1 (0,3,2)
2022-08-09T22:17:47.704Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":10453,"destendpoint":1,"srcendpoint":1,"clusterid":32,"transid":17,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,5,0,0,0,0]}}
2022-08-09T22:17:47.704Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,213,40,1,1,32,0,17,0,30,6,1,5,0,0,0,0,229]
2022-08-09T22:17:47.720Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-08-09T22:17:47.721Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-08-09T22:17:47.721Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-08-09T22:17:47.721Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-08-09T22:17:47.721Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
2022-08-09T22:17:49.165Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,17] - 26
2022-08-09T22:17:49.165Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":17}
2022-08-09T22:17:49.165Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-08-09T22:17:49.165Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0xb4e3f9fffedf0571:10453,205,3)
2022-08-09T22:17:49.166Z zigbee-herdsman:adapter:zStack:adapter Wait 2000ms

TheLongAndOnly, 2022-09-08:
I can second this. Similar issue, Danfoss Ally, Popp and Eurotronic TRV stop communicating after a while of flawless operation. I'm using a ConBeeII. Only way top fix I found so far is to rejoin the Zigbee network again. At least in my case the connections also run through routers.

wormiedk, 2022-09-27:
I have the same problem. I was hoping it was a cc2652 problem so that I could solve it by switching adapter. I might have to go back to zha instead then

moster11, 2022-10-03:
I found a solution for my use case. I had Zigbee2MQTT installed in Home Assistant OS which runs on a VM on my Synology NAS (DSM 7).

No matter if I used a Conbee II or the Sonoff Dongle I encountered the bespoken dropouts. As soon as I installed Zigbee2MQTT as a Docker image and set up my TRVs from there everything suddenly started to run without any flaws. Rock solid setup for more than 4 weeks already.

Seems to me that the USB passthrough to the VM was the root cause.

TheLongAndOnly, 2022-10-04:
Interesting, but I'm running z2m directly installed on a Raspi4. Maybe I should try docker or check, if there are options to improve USB behavior. Figured out that removing batteries from the devices also fixed the problem.

TheLongAndOnly, 2022-10-04:
I have a ZigBee sniffer available, but I'm not very familiar with the protocol. Is there anything I should watch for?

DonDonatello, 2022-10-05:
Experienced this as well with both the Sonoff and Conbee II stick. Newest stable Z2M running in Docker and all devices connected directly to the coordinator.

Happy to produce any relevant logs if needed.

xeri-sk, 2022-10-17:
I have same problem with Danfoss Ally TVR. After some time (hours, days) TVR still reporting states, but not responding for any command sent to TVR (through openhab or Z2M frontend).

When this happened to TVR i have to remove it from Z2M, factory reset TVR and re-join.
I hope that somebody will fix it soon couse i bought 3 TVR devices and are not usable with Z2M now.

My enviroment: OS Linux (Debian), Sonoff USB Dongle 3.0 Plus (with USB extension) FW 20220219, Z2M 1.28.0, openhab 3.3.0.

nikolai5slo, 2022-10-17:
Also happening to me. I have two Ally TRV and both stop responding after few hours. Restarting Z2M doesn't help. Only factory reset of TRV + re-join helps. I am not having any connection or signal issues with any other devices (around 20 devices on the network).

wormiedk, 2022-10-17:
I switched to ZHA and I don't have the problem there

stachjankowski, 2022-10-19:
In my case, it helped to reduce the range of the coordinator by removing the external antenna (Sonoff ZBDongle CC2652P).
Now all TRVs connect through routers (IKEA bulbs, TUYA Smart plug, etc).

My enviroment: QNAP (docker), Sonoff ZBDongle CC2652P FW 20220219, Z2M 1.28.0.

TheLongAndOnly, 2022-10-20:
I did some more research. I sniffed the network while the Danfoss Ally (0x4d8e) did not work anymore. I see data requests from the device reaching out to 0xd1ac, which is a GLEDOPTO GL-C-009 (Low Voltage LED Driver)

1118	251.639075	0x4d8e	0xd1ac	IEEE 802.15.4	12	Data Request
1138	258.690621	0x4d8e	0xd1ac	IEEE 802.15.4	12	Data Request
1167	265.741399	0x4d8e	0xd1ac	IEEE 802.15.4	12	Data Request
1185	272.792932	0x4d8e	0xd1ac	IEEE 802.15.4	12	Data Request

When I try to change a value through HA, I see a lot of route requests, but there is no success to reach the device

1195	275.161548	0xe6ae	Broadcast	ZigBee	51	Route Request, Dst: 0x4d8e, Src: 0xe6ae
1196	275.177605	0xe6ae	Broadcast	ZigBee	51	Route Request, Dst: 0x4d8e, Src: 0xe6ae
1197	275.184444	0xe6ae	Broadcast	ZigBee	51	Route Request, Dst: 0x4d8e, Src: 0xe6ae

I removed the battery and restarted the Danfoss (leaving everything else untouched) and I started to communicate again (just a short snippet:
1928	195.633365	0x4d8e	0xda68	IEEE 802.15.4	12	Data Request
1930	195.652961	0x0000	0x4d8e	ZigBee	49	APS: Ack, Dst Endpt: 1, Src Endpt: 1
1932	195.656802	0x4d8e	0xda68	IEEE 802.15.4	12	Data Request
1934	195.669037	0x0000	0x4d8e	ZigBee	49	APS: Ack, Dst Endpt: 1, Src Endpt: 1
1936	195.673227	0x4d8e	0xda68	IEEE 802.15.4	12	Data Request
1938	195.684501	0x0000	0x4d8e	ZigBee HA	56	ZCL: Default Response, Seq: 14

The biggest obvious difference seems to be the router which is used. It is now device 0xda68, which is an OSRAM Zigbee socket.

My suspect is that the GL-C-009 claims to be a router as it is permanently power with 24V, but does not do the job. Sometimes the TRV switches the router to the GL-C-009 and then it is disconnected as the router does not work. I will verify next time the TRV disconnects.

Question: IS there a way to exclude nodes from routing?

xeri-sk, 2022-10-21:
I tried to add two routers (Sonoff  S26R2ZB) to my Zigbee network and now all of my 3 Danfoss TVRs still working for 5 days.
Before i had only endpoint devices connected directly to coordination and Danfoss TVR works for 1 - 3 days.
Now 2 TVRs are connected through routers and 1 still directly to coordinator.
I will continue with monitoring and share results here.

[image]

xeri-sk, 2022-10-25:
After 7days 2 from 3 TVR not responding. This time i solved by rebooting my linux server.
I like my heating solution with Danfoss Ally but this issue is killing it totally.

Now testing 5m USB extension cable to avoid any interference from WiFi AP.

lukas-mc, 2022-10-25:
I changed the setup for my HA installation. Have VM now running HAOS and debian with ser2net. This setup is running fine so far -- except some TRV from Danfoss/Popp.

Now I have 6 TRV running without problems and exact one I can not communicate to.

I receive messages from the trv:

`2022-10-25 17:12:10Received Zigbee message from 'Heizung_Wohnzimmer_Anna', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":2187}' from endpoint 1 with groupID 0
Info`

The same as soon I changed the ,,occupied temp" at the TRV:

`2022-10-25 17:11:43Received Zigbee message from 'Heizung_Wohnzimmer_Anna', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1750,"setpointChangeSource":0}' from endpoint 1 with groupID 0
Info`

But when I try to change any settings using the GUI or M2Z this is not published to or received by the trv:

`2022-10-25 17:21:39Received MQTT message on 'zigbee2mqtt/Heizung_Wohnzimmer_Anna/set' with data '{"occupied_heating_setpoint":19.5}'
Debug 2022-10-25 17:21:39Publishing 'set' 'occupied_heating_setpoint' to 'Heizung_Wohnzimmer_Anna'
Debug 2022-10-25 17:21:46Received MQTT message on 'zigbee2mqtt/Heizung_Wohnzimmer_Anna/set/external_measured_room_sensor' with data '2100'
Debug 2022-10-25 17:21:46Publishing 'set' 'external_measured_room_sensor' to 'Heizung_Wohnzimmer_Anna'`

The value on the TRV is not changed and resetted to the original value within the GUI,

I tried to restart the TRV , HAOS and ser2net also as repairing the TRV  nothing helped to ,,awak" the beast.

Regards Lukas

DonDonatello, 2022-10-25:
I did some fiddling in the Dev console of Z2M and experienced something which might be strange - I'm not sure.

I tried reading some attributes and sending commands to one of my TRVs which currently works. However, after i sent a command that did not work/was not recognized by the TRV, i did not get any error from Z2M. I then tried change the heating setpoint from Z2M and nothing happened - the TRV did not react and the slider reset itself back to the previous setpoint (just as when the TRV is not functioning as expected).

After a minute or so Z2M suddenly gave multiple error messages from the wrong commands/get requests, and then changed the setpoint to the value I had been trying previously. And now the TRV is working as expected again. 

I am not sure if this is expected behaviour or not, however including log if relevant (0x847127fffef2bc71 is the relevant TRV).

[TRV.txt](https://github.com/Koenkk/zigbee2mqtt/files/9861873/TRV.txt)

xeri-sk, 2022-10-25:
> > After 7days 2 from 3 TVR not responding. This time i solved by rebooting my linux server. I like my heating solution with Danfoss Ally but this issue is killing it totally.
> 
> Well , which of the TRV's are not respondig? Both connected to the routers?

Exactly. But i am not sure if routers make problem, before i had Zigbee network without router and same problem with TVR occoured.

TheLongAndOnly, 2022-10-25:
I think I can now confirm the the GLEDOPTO LED drivers cause my problems. Another TRV went out of service and it is again trying to send data request to a GELDOPTO device:
861	174.081297	0x0549	0xe6ae	IEEE 802.15.4	12	Data Request
898	181.159780	0x0549	0xe6ae	IEEE 802.15.4	12	Data Request
966	188.238124	0x0549	0xe6ae	IEEE 802.15.4	12	Data Request
993	195.316823	0x0549	0xe6ae	IEEE 802.15.4	12	Data Request

0x0549 is the Popp TRV, 0xe6ae is the GLEDOPTO device.

I either need to find a way to tell the system not to route through the GLEDOPTO device or remove those devices. I read another article that those devices do not work properly as routers.

lukas-mc, 2022-10-25:
What I figured out now fits to yours DonDonatello,

> I am not sure if this is expected behaviour or not, however including log if relevant (0x847127fffef2bc71 is the relevant TRV).
> 
> I wonder if something like this might be happening "under the hood" without us knowing, i.e. a queue of commands of some sort is created but never sent to the TRV because something else is blocking it? To me, this fits nicely with the fact that Z2M continues to receive state updates from the TRV when the "disconnection" happens.
> 
When I delete the not responsive TRV completely from Z2M, restart the System and add it again, it will become responsive again. But You have to restart the System before pairing the TRV again. Other ways the TRV will remain unresponsive. 

This looks like something like a ,,queue" waiting to be send to the TRV, witch ill never be cleaned up on ,,error".

Regards Lukas

wormiedk, 2022-10-26:
I switched to ZHA a month ago and my thermostats have kept responding since. Maybe something is handled differently in ZHA?

MattWestb, 2022-10-26:
First OSRAM plug is made for the "black box of bad Zigbee device" then its not working as routers in the mesh so if having problem with them, taking them out of the network.
GLEDOPTO is using the same Zigbee chip so its using the same Zigbee stack with the same bugs = problems.

If some brave user have one not TI or de(F)CONZ coordinator and can testing im 110% sure its working having those TRV connected to the coordinator.

One coordinator is one "normal" router with little extras like the Trust Center that is manage the security in the mesh network.

Latest finding is that TI coordinators is having problems in the router part and real Zigbee 3 end device is having problem then the router is not setting up "end device timeout request" and is leaving or falling out of the network if using the coordinator as there parent.

Also possible adding some good Zigbee 3 router in the network and trying getting the TRVs using them shall working OK.

TheLongAndOnly, 2022-10-26:
@MattWestb how can I prevent devices from changing the router? My problem is that devices seem to change the route after a while, then picking the GLEDOPTO.

Koenkk, 2022-10-26:
Can someone provide me a sniff starting from the point where controlling the thermostat still works until it fails?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

TheLongAndOnly, 2022-10-26:
I think we have different issue here, I have sniffs from a TRV being unresponsive and from the restart after battery disconnect, but none which is tracking the moment of failure. It takes days to weeks to happen.  But I'm pretty sure my issue is caused by the GLEDOPTP devices.

MattWestb, 2022-10-26:
@TheLongAndOnly You cant the its the self healing of one mesh network.
Best practice using good Zigbee 3 router and not "OSRAM Plug" types.

@Koenkk Can you pleas making the TI coordinator replaying to Command Frame: `End Device Timeout Request` from joining end devices with one `End Device Timeout Response, Success` ? 

If not its not working with most Zigbee 3 end devices.

Its one part of the Zigbee PRO = Zigbee 3 moderate.

Koenkk, 2022-10-27:
@MattWestb 

> Can you pleas making the TI coordinator replaying to Command Frame: End Device Timeout Request from joining end devices with one End Device Timeout Response, Success ?

With a Zigbee 3 coordinator it should already do this.

Koenkk, 2022-10-27:
@MattWestb what firmware are you running on the ZZH?

MattWestb, 2022-10-27:
The user have testing zzh on dev firmware.20220928 and february 19th. on a SonOff key.
I dont have TI coordinators i exclusive EZSP.

paszczaq, 2022-10-31:
Seems that i'm this club too.
I'm using z2m as HA OS addon with Sonoff-P USB dongle since 1month - i've switched from deconz+conbee2 about month ago. And after couple of days one of three danfoss ally stopped to accept commands from z2m/HA but still sending data if I change temp manully. The one with problem seems to be connected directly to coordinator. The other 2 are connected via routers (tuya 16a power plug) - and working fine.

What i've noticed that on bindings tab the target endpoint is different - "1" in the Salon Termostat which stopped to accept commands and "242" in Pokój Termostat which works fin
[image]

[image]

xeri-sk, 2022-11-03:
I am sharing results from testing 5m USB extension cable.

Today after 9 days one TVR stopped responding. Another two still respond commands.
First i tried to remove batteries from that one TVR and wait 15min. This solves problem, but another two TVRs stopped responding immediatelly. Can reseting one TVR affect anothers?

Finally i disconnected coordinator from USB a stopped z2m service for 30min.
After connect coordinator and start service again, all TVR working again.

Let me note i didnt connect any router during test.
End devices:
3x Danfoss Ally TVR [014G2461](https://www.zigbee2mqtt.io/devices/014G2461.html#danfoss-014g2461)
3x Sonoff temp sensor [SNZB-02](https://www.zigbee2mqtt.io/devices/SNZB-02.html#sonoff-snzb-02)
1x Sonoff door sensor [SNZB-04](https://www.zigbee2mqtt.io/devices/SNZB-04.html#sonoff-snzb-04)

wormiedk, 2022-11-04:
@xeri-sk which zigbee stick / coordinator are you using?

Is z2m running as an addon or standalone?

tatec22, 2022-11-04:
I have the exact same issue in my network. (Conbee2 and z2m in a docker conatiner)
3 of my 5 Danfoss Ally do not respond when I change the temperature in HA or the Z2M WebUi.
Only after I repeat the request to an unresponsive device serveral times it might finally respond and set the temperature value.

xeri-sk, 2022-11-04:
@wormiedk 
My enviroment: OS Linux (Debian), Sonoff USB Dongle 3.0 Plus (with USB extension) FW 20220219, Z2M 1.28.1, openhab 3.3.0.
Z2M is installed manually, not as addon.

lukas-mc, 2022-11-05:
Today I installed version 1.28.2.1 of z2m and since then only 2 of my 10 Danfoss TRV are responsive any more. 
I see the status is transmitted from TRV to Corodinator by I can not change any value  back to thr TRV any more.

Before Version 1.28.2-1 only some of the TRV got unresponsive after a couple of days, and after restarting TRV, other devices or HA the TRV got resonsive again. 

Thi8s is more bad ever before.

paszczaq, 2022-11-05:
@lukas-mc I think that's justo coincidence I've updated to 1.28.2.1 yesterday and all 3 Danfoss TRV's are reacting on commands from HA/z2m. But the time will tell when one of them will stop responding. It must be somthing with "routing" or binding...thats my guess. Previously I was on conbee2 with deconz and all 3 trv's worked fine for months

lukas-mc, 2022-11-05:
Thank You @paszczaq.
Did you repair the devices after upgrade? I recogniced some new parameters after upgrade.

I also noticed some delay in sending parameters to the trv. Some times you have to repeat the change of parameter in the Gui to take effect.

After I removed most of the routers from my setup, most of the TRV do send data all the time.

So it could be routing and/or buffers/binding.  

> @lukas-mc I think that's justo coincidence I've updated to 1.28.2.1 yesterday and all 3 Danfoss TRV's are reacting on commands from HA/z2m. But the time will tell when one of them will stop [...]

paszczaq, 2022-11-06:
@lukas-mc I'm not 100% sure but my procedure to get it back was:
1. stop z2m
2. remove battery from TRV
3. start z2m
4. put battery back to trv

Also can you check whats your Binding config?

[image]

moritz-john, 2022-11-08:
I have the same issue. 

I only have two Zigbee devices in total: 2x Popp-TRVs (identical to Danfoss Ally TRVs)
and I am using a SONOFF Zigbee 3.0 USB Dongle Plus, TI CC2652P + CP2102(N)

I am running _zigbee2mqtt_, _Home-Assistant-Core_ and _mosquitto_ all in a different Docker on Unraid.

The TRVs become unresponsive after a few hours. 
Resetting and re-adding them to zigbee2mqtt make them work for a few hours until they become unresponsive again.

___

I just got one Philips Hue light and I added it to my zigbee2mqtt mesh network.

My plan is to route one of my Popp-TRVs through the light and one directly through my SONOFF Dongle as before and see what happens. I hope that the Popp-TRV that is routed through the Philips light stays responsive as descirbed in comments above.

[image]

lukas-mc, 2022-11-11:
> @lukas-mc I'm not 100% sure but my procedure to get it back was:
> 
> 1. stop z2m
> 2. remove battery from TRV
> 3. start z2m
> 4. put battery back to trv
> 
> Also can you check whats your Binding config?
> 
> [image]

@paszczaq Yes, this works for me also.  
And my bindings are the same as yours.

But I now upgrades to HAOS 22.11.2 and Z2M Version 1.28.2-1. Since then all of my TRVs are working fine for now 6 days.

Will report more when they got unresponsive again.

moritz-john, 2022-11-17:
> My plan is to route one of my Popp-TRVs through the light and one directly through my SONOFF Dongle as before and see what happens. I hope that the Popp-TRV that is routed through the Philips light stays responsive as described in comments above.

So I bought some more Zigbee devices and tried many different configurations.

[image]

I came to the following conclusion:

**My CC2652P can only handle exactly one direct connection to a Popp/Danfoss TRV.**

As soon as I connect another Popp/Danfoss TRV directly to my CC2652P one of them become unresponsive. 

___

So my solution is to connect only  '1>' Popp/Danfoss TRVs to my CC2652P Dongle and connect all the other TRVs through different routers like IKEA / Hue lights...

Sadly I can't provide any logs or explanations as to why this issue / bug happens.

wormiedk, 2022-11-17:
@moritz-john the strange thing is that the TRVs work fine in zha where they don't become unresponsive...

MattWestb, 2022-11-17:
@wormiedk One sleeping or not sleeping end device can only have on parent router and its up to the device selecting on and also jumping to one it think is better and the algorithm is locked in the zigbee stack (= firmware) and is different for manufacture / version.

The best way is having one good router in the near the end device so its keeping its parent but some device is not doing it (like Philips HUE motion sensor that is one jumper to the bad routers it can finding).

In ZHA use "add device via this device" from the router device card for forcing it joining thru it but if is not like it it jumping to one "better" and we cant doing any thing its how one mesh network is working.

xeri-sk, 2022-11-17:
Today morning all 3 TVRs stopped working at the same time. All were connected directly to coordinator (no routers in network) for 13 days. I thing problem must be somewere in Z2M.

MattWestb, 2022-11-17:
If some one have one sniffer its being very interesting see how paring with the coordinator in Z2M **and** ZHA is being done.
Also different coordinator is of large interest (IT, Corn/RaspBee I / II and EZSP).

i-wintermute, 2022-11-23:
I can confirm the bug.
Zigbee2MQTT version **1.28.2** 
Coordinator type **zStack3x0**
Coordinator revision **20211219**
Total installed 10 danfoss ally TRV Firmware version 01.18.0008 01.18
The closest TRV set stopped responding after a week of connection.
Others are still working correctly.

DonDonatello, 2022-11-23:
My Ally’s have been running smoothly for a month now after I made some changes to my setup. I stopped my Home Assistant, removed one of my other devices from the network (a Aqara temp sensor) and changed the coordinator passthrough in Docker to Z2M from ttyACM0 to ttyUSB0, so I cannot be sure what did the trick. Also I cannot see why any of the three would affect the setup in the first place.

I will try changing all three back one at a time and report if my Ally’a stop working again

tatec22, 2022-12-07:
I also suffered from Ally's which became unresponsive or took very long to respond.
A few days I just discovered something interesting which I haven't really thought about yet (even though the documentation clearly mentions it).

I'm running Home Assistant and of course some automations to set different values of my Ally's.
One automation does set the temperature value of an external room sensor to the Ally's "external_measured_room_sensor" parameter every 5 minutes or whenever a temp change occurs.
It seems that this automation caused the Ally's to get stuck in a loop which made them unresponsive.
I just changed my automation to only update the "external_measured_room_sensor" parameter every 10 minutes. 
All my Ally's "healed" themelves within minutes and became super responsive again. 

I had no problems since.

Maybe this helps anyone to fix their issues as well..

MattWestb, 2022-12-07:
Firmware update its in the pipe https://csa-iot.org/csa_product/danfoss-ally-radiator-thermostat-3/.

moritz-john, 2022-12-08:
> Multiple Ally's behind IKEA, Ubisys and Silvercrest routers and all working fine for multiple weeks now, while they kept dropping out when directly connected to the coordinator.

Are multiple Ally's connected to the same router in your setup? Or is each Ally TRV connected to a different router?

My setup has been working for over two weeks now but only if each TRV is connected to a different router. 
If two or more TRVs are connected to the same router I get the same problem as when more than 1 TRV is connected to the coordinator.

[image]

xeri-sk, 2022-12-10:
> By reducing transmit power on my coordinator to the bare minimum and power cycling all the thermostats, I convinced them to connect through routers instead and my dozen or so thermostats are all stable now.
> 

What is your current transmit power setting? I would like to descrease it, but i dont have specified any value, so probably default (unknown) value is used.

BBJake, 2022-12-15:
It seems that I have the same problem with Bosch Thermostat II devices...
I have one thermostat which is directly connected to the nearby coordinator (Sonoff Dongle-P).
After some times this device does not accept any commands sent to it. It still reports the actual temperature or changes that I do directly on the thermostat.
Other Bosch thermostats which are connected over a router are not affected. Also other thermostats from a different brand (Saswell) which are connected to the coordinator directly are not affected.
Most of the time I do not get error messages, so I guess the message is stuck somewhere in z2m.

BBJake, 2022-12-15:
btw.  I can resolve the problem by removing the batteries of the thermostat. After that it accepts new commands...
So maybe also the device blocks something, or the reconnecting releases something in z2m (or maybe the stick).

KuChar91, 2022-12-22:
Same problem here with not accepting commands to Danfoss Ally after some time.

louiskram, 2022-12-25:
I had the same problem where I'd face random unresponsiveness from the trvs. I have five popp trvs which are based on the Danfoss Ally. The workaround for me was the same that some people here already mentionend: to connect the trvs over a router like a lamp or a power plug. 

Three of them are behind the same router and the other two each have a router to their own. 

I didn't change anything else beside that and everything is now working flawlessly!

Char-r, 2022-12-26:
Seems that a Router is not the silver bullet to solve this issue:
[image]

While the router has a working route, it is now connected at 0 to the Coordinator and the devices are refusing to receive commands

wormiedk, 2022-12-26:
There is definitely some problem in z2m. Switched back to zha 2 months ago and no dropouts since then

pmoser1976, 2022-12-29:
I can confirm that issue of not responding POPP TRVs after a while.

My Setup is:
- Zigbee2MQTT via Docker and USB Passthroug
- Mosquitto via Dockeer
- IOBroker via Debian VM
- Different Routers
- POPP TRVs (some with original firmware, some with updated firmware, some with firmware Problem 

Many of the TRVs work without issues, but some are affected in the described way (they send their status, but don't react when sending a new Temp-Value to them).

Maybe this observation can help: 

a Log entry of a **successful command** (send occupiedHeatingSetpoint to TRV):
_Received Zigbee message from 'TOP4_TRV_KU', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2450,"setpointChangeSource":2}' from endpoint 1 with groupID 0_

a Log entry of sending occupiedHeatingSetpoint to TRV **not reacting**:
_Received Zigbee message from 'TOP4_TRV_KZ', type 'commandCheckIn', cluster 'genPollCtrl', data '{}' from endpoint 1 with groupID 0_

[image]

[image]

xeri-sk, 2022-12-30:
I am fighting with this issue for 2months and can confirm followings.

TVR stop responding:
- after few days when connected directly to coordinator
- immediately when changed connection from router to coordinator

Simple workaround can be forcing TVR to connect specific router, but i didn't found the way yet.
Tried reducing tx power on z2m, installed more routers, but sometime TVR still change connection to coordinator.

When TVR stop responding i can fix by manually forcing TVR to connect to nearest router:

1.  Disconnect coordinator from USB
2. Push button on TVR - this will immediately connect to nearest router (coordinator is not available)
3. Connect coordinator to USB
4. Everything is working again

pmoser1976, 2023-01-02:
After changing Temperature manually on affected TRV it starts reacting to Zigbee-commands again.
Sometimes within seconds, in other cases after 30 or 60 minutes.

gdschut, 2023-01-07:
Sadly I experience the same connecting problems.
My config: Home Assistant(docker) + Zigbee2MQTT(docker) + conbee II (5m extention cable) + 2 Popp TRV (and a few other devices).
One of the TRV's is on old firmware (01.12.0008 01.12) and one of them with updated firmware (01.18.0008 01.18).
The one with old firmware works fine, the one updated looses connection after a while.
After restarting all of them to reconnect it always returns to this situation.
Both are directly connected to my coordinator.
Some questions:
- is it possible that the firmware of the TRV is the problem?
- is it possible to downgrade the firmware of the TRV to test this?

If there is anything I can do to help solving this issue, please let me know!

brusiniak, 2023-01-08:
I am experiencing same issue. I think it might be related somehow with  #11049, however one Popp trv that is connected with Sonoff Dongle hangs ever couple of hours and other that goes thru the router is working stable (up 2 now)

[143 more replies]

closed 2023-06-08 by Koenkk