Koenkk/zigbee2mqtt #13479: [New device support]: Aqara E1 human body movement and illuminance sensor

vh13294, 2022-08-10:
### Link

https://www.tokopedia.com/bramcom/aqara-human-body-sensor-e1?extParam=ivf%3Dfalse

### Database entry

{"id":4,"type":"EndDevice","ieeeAddr":"0x54ef441000491f98","nwkAddr":63817,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.motion.acn001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,64704],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"appVersion":8,"modelId":"lumi.motion.acn001","zclVersion":3,"stackVersion":2,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":8,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1660106123966,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to use the Aqara P1 motion sensor configuration. But it doesn't seem to work.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['llumi.motion.acn001'],
    // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'WSDCGQ01LM',
    // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi',
    // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara human body movement and illuminance sensor',
    // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
    fromZigbee: [fz.aqara_opple, fz.battery, fz.aqara_occupancy_illuminance, fz.RTCGQ11LM_illuminance],
    // We will add this later
    toZigbee: [],
    // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.occupancy(), e.device_temperature(), e.battery_voltage(), e.illuminance_lux().withProperty('illuminance'),
            e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'), e.power_outage_count(false)],
    // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Otnow, 2022-08-10:
This sensor is more likely analog _T1_ than _P1_, so try using the following [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters):

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.motion.acn001'],
    model: 'RTCGQ15LM',
    vendor: 'Xiaomi',
    description: 'Aqara E1 human body movement and illuminance sensor',
    fromZigbee: [fz.aqara_occupancy_illuminance, fz.aqara_opple, fz.battery],
    toZigbee: [tz.aqara_detection_interval],
    exposes: [e.occupancy(), e.illuminance_lux().withProperty('illuminance'),
        e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
        exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
        .withDescription('Time interval for detecting actions'),
        e.device_temperature(), e.battery(), e.battery_voltage(), e.power_outage_count(false)],
    meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genPowerCfg', ['batteryVoltage']);
        await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
    },
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

_Please note that the firmware of the coordinator must be at least a [certain version](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware) in order for the sensor to work correctly._

vh13294, 2022-08-10:
> This sensor is more likely analog _T1_ than _P1_, so try using the following [external converter](https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters):
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['lumi.motion.acn001'],
>     model: 'RTCGQ15LM',
>     vendor: 'Xiaomi',
>     description: 'Aqara E1 human body movement and illuminance sensor',
>     fromZigbee: [fz.aqara_occupancy_illuminance, fz.aqara_opple, fz.battery],
>     toZigbee: [tz.aqara_detection_interval],
>     exposes: [e.occupancy(), e.illuminance_lux().withProperty('illuminance'),
>         e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'),
>         exposes.numeric('detection_interval', ea.ALL).withValueMin(2).withValueMax(65535).withUnit('s')
>         .withDescription('Time interval for detecting actions'),
>         e.device_temperature(), e.battery(), e.battery_voltage(), e.power_outage_count(false)],
>     meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await endpoint.read('genPowerCfg', ['batteryVoltage']);
>         await endpoint.read('aqaraOpple', [0x0102], {manufacturerCode: 0x115f});
>     },
>     ota: ota.zigbeeOTA,
> };
> 
> module.exports = definition;
> ```
> 
> _Please note that the firmware of the coordinator must be at least a [certain version](https://www.zigbee2mqtt.io/devices/RTCGQ12LM.html#adapter-firmware) in order for the sensor to work correctly._

Hi, it works better than the P1. But I discover that there is a config for the same "model" but different "zigbee model".

```javascript
zigbeeModel: ['lumi.sensor_motion.aq2'],
        model: 'RTCGQ11LM',
        vendor: 'Xiaomi',
        description: 'Aqara human body movement and illuminance sensor',
        meta: {battery: {voltageToPercentage: '3V_2850_3200'}},
        fromZigbee: [fz.xiaomi_basic, fz.occupancy_with_timeout, fz.RTCGQ11LM_illuminance],
        toZigbee: [],
        exposes: [e.battery(), e.occupancy(), e.device_temperature(), e.battery_voltage(), e.illuminance_lux().withProperty('illuminance'),
            e.illuminance().withUnit('lx').withDescription('Measured illuminance in lux'), e.power_outage_count(false)],
```

shall I submit a merge request with "lumi.motion.acn001" added to the array?

Otnow, 2022-08-10:
> Shall I submit a pull request?

No, you rushed with the PR, because we also need to analyze the heartbeat-message reported by the sensor.

Please provide a [debug log](https://www.zigbee2mqtt.io/guide/usage/debug.html#zigbee2mqtt-debug-logging) (`log_level: debug`), which will contain the following actions:

- sensor pairing
- press the pairing button once (heartbeat-message (`{"247":{"data":...`) from the sensor should appear in the log)

vh13294, 2022-08-10:
- sensor pairing
```
2022-08-10 20:45:23Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"rj4lc-2","value":true}'
2022-08-10 20:45:29Device '0x54ef441000491f98' announced itself
2022-08-10 20:45:34Received Zigbee message from '0x54ef441000491f98', type 'attributeReport', cluster 'genBasic', data '{"appVersion":8,"modelId":"lumi.motion.acn001"}' from endpoint 1 with groupID 0
2022-08-10 20:45:35Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
2022-08-10 20:45:36Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
2022-08-10 20:45:37Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
2022-08-10 20:45:38Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
2022-08-10 20:45:39Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
2022-08-10 20:45:40Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
2022-08-10 20:45:41Received Zigbee message from '0x54ef441000491f98', type 'readResponse', cluster 'aqaraOpple', data '{"258":60}' from endpoint 1 with groupID 0
2022-08-10 20:45:41lumi.motion.acn001: Processed data into payload {"detection_interval":60}
2022-08-10 20:45:54Received Zigbee message from '0x54ef441000491f98', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,21,12,3,40,25,4,33,0,0,5,33,1,0,8,33,8,0,10,33,0,0,12,32,30,100,16,0,101,33,9,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
2022-08-10 20:45:54lumi.motion.acn001: Processed buffer into data {"1":3093,"3":25,"4":0,"5":1,"8":8,"10":0,"12":30,"100":0,"101":9}
2022-08-10 20:45:54lumi.motion.acn001: Processed data into payload {"voltage":3093,"battery":69,"device_temperature":25,"power_outage_count":0,"state":"OFF"}
2022-08-10 20:45:54lumi.motion.acn001: Processed data into payload {"voltage":3093,"battery":69,"device_temperature":25,"power_outage_count":0,"state":"OFF"}
2022-08-10 20:46:13Received Zigbee message from '0x54ef441000491f98', type 'attributeReport', cluster 'aqaraOpple', data '{"illuminance":65546}' from endpoint 1 with groupID 0
2022-08-10 20:46:13lumi.motion.acn001: Processed data into payload {}
```

- press the pairing button once
```
2022-08-10 20:46:35Received Zigbee message from '0x54ef441000491f98', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.motion.acn001"}' from endpoint 1 with groupID 0
2022-08-10 20:46:40Received Zigbee message from '0x54ef441000491f98', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,31,12,3,40,25,4,33,0,0,5,33,1,0,8,33,8,0,10,33,0,0,12,32,30,100,16,1,101,33,6,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
2022-08-10 20:46:40lumi.motion.acn001: Processed buffer into data {"1":3103,"3":25,"4":0,"5":1,"8":8,"10":0,"12":30,"100":1,"101":6}
2022-08-10 20:46:40lumi.motion.acn001: Processed data into payload {"voltage":3103,"battery":72,"device_temperature":25,"power_outage_count":0,"state":"ON"}
2022-08-10 20:46:40lumi.motion.acn001: Processed data into payload {"voltage":3103,"battery":72,"device_temperature":25,"power_outage_count":0,"state":"ON"}
```

- When trying to remove the device from GUI page
```
2022-08-10 20:40:54Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data '{"block":false,"force":false,"id":"0x54ef441000491f98","transaction":"rj4lc-1"}'
2022-08-10 20:41:04Error: Failed to remove device '0x54ef441000491f98' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms) at Bridge.removeEntity (/app/lib/extension/bridge.ts:563:19) at Bridge.onMQTTMessage (/app/lib/extension/bridge.ts:122:34)
```

- when holding reset button after trying to remove the device from GUI page (failed)
```
2022-08-10 20:43:06lumi.motion.acn001: unknown key 252 with value 0
```

Otnow, 2022-08-10:
Ok, I created an additional [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4522) based on the analysis of the heartbeat-message reported by the sensor.

---

> * When trying to remove the device from GUI page
>
>
> ```
> 2022-08-10 20:40:54Received MQTT message on 'zigbee2mqtt/bridge/request/device/remove' with data [...]

To delete a sensor or change its settings, you need to wake it up by pressing the pairing button once so that it accepts the command.

---

>     * when holding reset button after trying to remove the device from GUI page (failed)
> 
> 
> ```
> 2022-08-10 20:43:06lumi.motion.acn001: unknown key 252 with value 0
> ```

This message was [noticed](https://github.com/Koenkk/zigbee2mqtt/issues/13252#issuecomment-1194568803) on the _E1 leak sensor (SJCGQ13LM)_, but so far it has not been possible to understand what it means.

Koenkk, 2022-08-11:
Since this device has been added now I assume this can be closed.

closed 2022-08-11 by Koenkk