Koenkk/zigbee2mqtt #13480: Trust ZPIR-8000 Motion sensor is always Occupied

Xpl0jd, 2022-08-10:
### What happened?

After adding Trust ZPIR-8000 to the network is reporting all the time to be Occupied. It is getting updates from the sensor when it detects motion but it is never set to Clear

This sensor works in ZHA and deCONZ.

### What did you expect to happen?

Occupancy attribute should change to false after some time (60sec) when there is no motion detected and the status of the sensor is changed to Clear

### How to reproduce it (minimal and precise)

Add ZPIR-8000 to the network
Detect motion with a sensor
Leave it idle, it will never become "Clear".

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26720700

### Adapter

Conbee2

Koenkk, 2022-08-10:
Can you provide the debug log starting from the point where you trigger occupancy until you expect occupancy to turn false again?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Xpl0jd, 2022-08-10:
> Can you provide the debug log starting from the point where you trigger occupancy until you expect occupancy to turn false again?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Sorry i´m new to HA and zigbee2mqtt and are not sure what logs you wanted.
I did have a look in Zigbee2MQTT > Logs with debug level.

Once i trigger motion i get the following.
Debug 2022-08-10 18:08:09Received Zigbee message from '0x00158d0001953355', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":2}' from endpoint 1 with groupID null
Info 2022-08-10 18:08:10MQTT publish: topic 'zigbee2mqtt/0x00158d0001953355', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":true,"tamper":false}'

Koenkk, 2022-08-10:
That is what I'm looking for, do you see a similar message when the sensor should report occupancy false? (after e.g. 60 seconds?)

Xpl0jd, 2022-08-10:
> That is what I'm looking for, do you see a similar message when the sensor should report occupancy false? (after e.g. 60 seconds?)

No i dont see anything when it should go back to false.
Did also check with MQTT explorer and always say true.

Koenkk, 2022-08-11:
@arcsyst  are you still using z2m/this sensor, if yes does it still work for you?

arcsyst, 2022-08-11:
Still using z2m but not this device any longer, couldn't get it to work reliably and scrapped it quite a while ago.

Keep up good work!

Brian

closed 2022-09-18 by github-actions[bot]

AppleFlapp, 2023-03-15:
I have two with the same/similar issues. I moved them from my KaKu ICS-2000 to HA but now one is continuously on "occupied" and the other one is continuously on "clear". They are paired correctly by Z2M and when there is movement the led blinks but occupancy doesn't change. 
Unless I did something wrong, debugging is also not reporting anything about those PIRs.
They don't show up in debug either.

AppleFlapp, 2023-03-15:
Bummer. Might need to swap to Aqara P1 then. 

Thanks for sharing this info. Not what I wanted to hear but at least I now know how to proceed.

nukleuz80, 2023-06-16:
> > I have two with the same/similar issues. I moved them from my KaKu ICS-2000 to HA but now one is continuously on "occupied" and the other one is continuously on "clear". They are paired correctly by Z2M and when there is movement the led blinks but occupancy doesn't change. Unless I did something wrong, debugging is also not reporting anything about those PIRs. They don't show up in debug either.
> 
> This device is not working with Zigbee2mqtt, it states as supported but its not. You need ZHA or Deconz to get this device working.

It´s working correctly with Firmware 0.1.4.
I have two of ZPIR-8000:

with 0.1.0 --> always occupied
with 0.1.4 --> like expected

The question is: How to update without Trust Bridge?

THX

nukleuz80, 2023-09-20:
I can confirm that FW 0.1.0 is working in ZHA!