Koenkk/zigbee2mqtt #13484: Groups remove_all Error

AIM289, 2022-08-10:
### What happened?

The remove_all command (to remove a device from all groups) does not appear to be working correctly and produces an error. (Error: Cannot read properties of null (reading 'isDevice'))

When I first attempted to use this command, the device was not removed from any groups and an error was given. However, since deleting the groups, restarting Zigbee2MQTT, and re-creating and populating groups, the remove_all command does now appear to remove the device from all groups correctly, but the error is still given. I have been unable to replicate the initial behaviour of a device not being removed from the groups.

The same issue occurs with an OSRAM LIGHTIFY Outdoor Flex RGBW, and an IKEA TRADFRI control outlet.

Occurs on both master and dev branches.

### What did you expect to happen?

I expected the device to be removed from all groups, and no error.

### How to reproduce it (minimal and precise)

Add a device to 1 or more groups (although the error also occurs when a device is in no groups)

Publish to the topic "zigbee2mqtt/bridge/request/group/members/remove_all" with payload "{"device": "my_device"}"

### Zigbee2MQTT version

 1.27.0 commit: a9b8808

### Adapter firmware version

20210708

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle Plus Gateway with Antenna

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9299015/log.txt)

Koenkk, 2022-08-10:
Fixed the issue, please let me know if it also works for you.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

AIM289, 2022-08-11:
Yes, it works for me now too. Thanks for fixing this so quickly!

closed 2022-08-11 by AIM289