Koenkk/zigbee2mqtt #13488: Using hue_move causes invalid color mode warnings in Home Assistant

deviantintegral, 2022-08-10:
### What happened?

If I use `hue_move` to set a looping color on a light, and I have reporting set up on the light to report colour changes, Home Assistant will log warnings every time the colour change is reported.

> Logger: homeassistant.components.mqtt.light.schema_json
> Source: components/mqtt/light/schema_json.py:336 
> Integration: MQTT (documentation, issues) 
> First occurred: 10:23:02 AM (1214 occurrences) 
> Last logged: 12:26:00 PM
> 
> Invalid color mode received

### What did you expect to happen?

I did not expect warnings in Home Assistant.

### How to reproduce it (minimal and precise)

1. Set the light to report both hue and saturation changes.

[image]

2. Publish a message like the following to `/get` on the light to start rotating colours:

```json
{"state": "on", "hue_move": 1, "color": {"saturation": 100}}
```

Notice the errors in the Home Assistant log, until the light is turned off or the color is manually changed via the Home Assistant UI:

```json
{
  "topic": "zigbee2mqtt/Desk Backlight",
  "payload": {
    "brightness": 254,
    "color": {
      "h": 167,
      "hue": 167,
      "s": 100,
      "saturation": 100,
      "x": 0.1535,
      "y": 0.3853
    },
    "color_mode": "hs",
    "color_temp": 153,
    "last_seen": "2022-08-10T16:27:44.889Z",
    "linkquality": 98,
    "state": "ON",
    "update": {
      "state": "available"
    }
  },
  "qos": "AT_MOST_ONCE",
  "receivedAt": "2022-08-10 12:27:44",
  "retain": false
}
```

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

Electrolama zig-zig-ah! zzh

Koenkk, 2022-08-10:
This is a limitation of HA, it can only support XY or HS (not both at the same time)

deviantintegral, 2022-08-10:
Is there a way to configure reporting so it reports in XY instead of HS? Or would it make sense to convert HS to XY in the message sent to Home Assistant?

Koenkk, 2022-08-11:
You can configure HA to use HS instead of XY by adding the following to the z2m `configuration.yaml`:

```yaml
devices:
  "0x12345678":
    friendly_name: my_switch
    homeassistant:
      light:
        supported_color_modes: ['hs','color_temp']
```

deviantintegral, 2022-08-12:
Interesting. I applied that to the light and restarted both zigbee2mqtt and Home Assistant. I'm still getting the warnings in the logs.

To confirm, the config I've set is:

```yaml
  '0x001788010b48c3f4':
    friendly_name: Desk Backlight
    homeassistant:
      light:
        supported_color_modes: ['hs', 'color_temp']
```

The light is this one: https://www.zigbee2mqtt.io/devices/8718699703424.html#philips-8718699703424

Also, I noticed that `supported_color_modes` doesn't have any results in the zigbee2mqtt docs, so once we get this sorted we should figure out some sort of docs update too.

Koenkk, 2022-08-13:
Can you provide me the debug log when starting z2m?

deviantintegral, 2022-08-21:
Here you go. I captured to the point where I saw normal traffic and it appeared startup was done: https://gist.github.com/deviantintegral/ef3845963757f060f4e755ebc964ab70

I also get this when I start `hue_move` - interesting that somehow `color_temp` is going above 500?

```
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:debug 2022-08-20 21:22:41: Received MQTT message on 'zigbee2mqtt/Desk Backlight/set' with data '{"state": "on", "hue_move": 1, "color": {"saturation": 100}}'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:debug 2022-08-20 21:22:41: Publishing 'set' 'state' to 'Desk Backlight'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:info  2022-08-20 21:22:41: MQTT publish: topic 'zigbee2mqtt/Desk Backlight', payload '{"brightness":254,"color":{"h":26,"hue":26,"s":100,"saturation":100,"x":0.5473,"y":0.4292},"color_mode":"hs","color_temp":500,"last_seen":"2022-08-21T01:22:41.690Z","linkquality":109,"state":"OFF","update":{"state":"available"},"update_available":null}'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:debug 2022-08-20 21:22:41: Publishing 'set' 'hue_move' to 'Desk Backlight'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:info  2022-08-20 21:22:41: MQTT publish: topic 'zigbee2mqtt/Desk Backlight', payload '{"brightness":254,"color":{"h":26,"hue":26,"s":100,"saturation":100,"x":0.5473,"y":0.4292},"color_mode":"hs","color_temp":500,"last_seen":"2022-08-21T01:22:41.733Z","linkquality":109,"state":"OFF","update":{"state":"available"},"update_available":null}'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:debug 2022-08-20 21:22:41: Publishing 'set' 'color' to 'Desk Backlight'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:info  2022-08-20 21:22:41: MQTT publish: topic 'zigbee2mqtt/Desk Backlight', payload '{"brightness":254,"color":{"h":26,"hue":26,"s":100,"saturation":100,"x":0.5473,"y":0.4292},"color_mode":"hs","color_temp":500,"last_seen":"2022-08-21T01:22:41.777Z","linkquality":109,"state":"OFF","update":{"state":"available"},"update_available":null}'
Aug 20 21:22:41 zigbee2mqtt npm[16639]: Zigbee2MQTT:debug 2022-08-20 21:22:41: Requested color_temp 509 is higher than maximum supported 500, using maximum!
```

Koenkk, 2022-08-21:
> Here you go. I captured to the point where I saw normal traffic and it appeared startup was done: https://gist.github.com/deviantintegral/ef3845963757f060f4e755ebc964ab70

Please provide me the complete log, I want to see whether `supported_color_modes` is applied correctly. In this log; can you also trigger the error in HA by:
- Go to the HA log and press clear (settings -> system -> logs -> clear)
- Start z2m
- change color of the bulb via HA
- go back to HA log, press refresh and confirm the error is now there

> I also get this when I start hue_move - interesting that somehow color_temp is going above 500?

z2m has some functionality to sync the hue/saturation with the color temp (just the published value in the payload, this is not send to the bulb). But this conversion is not perfect (given that you go from color -> color temperature) therefore it can happen that the computed color temp value exceeds the max support color temp of the bulb which triggers this debug logging.

deviantintegral, 2022-09-02:
This error showed up twice in the HA logs after restarting zigbee2mqtt:

> Logger: homeassistant.components.mqtt.light.schema_json
> Source: components/mqtt/light/schema_json.py:333 
> Integration: MQTT (documentation, issues) 
> First occurred: 3:10:34 PM (1 occurrences) 
> Last logged: 3:10:34 PM
> 
> Invalid color mode received

It appears that `supported_color_modes` is being set correctly:

```
Sep 02 15:10:53 zigbee2mqtt npm[3979]: Zigbee2MQTT:info  2022-09-02 15:10:53: MQTT publish: topic 'homeassistant/light/0x001788010b48c3f4/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Desk Backlight/availability"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Desk Backlight/set","device":{"identifiers":["zigbee2mqtt_0x001788010b48c3f4"],"manufacturer":"Philips","model":"Hue white and color ambiance LightStrip plus (8718699703424)","name":"Desk Backlight","sw_version":"1.93.11"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":500,"min_mireds":150,"name":"Desk Backlight","schema":"json","state_topic":"zigbee2mqtt/Desk Backlight","supported_color_modes":["hs","color_temp"],"unique_id":"0x001788010b48c3f4_light_zigbee2mqtt"}'
Sep 02 15:10:54 zigbee2mqtt npm[3979]: Zigbee2MQTT:debug 2022-09-02 15:10:54: Received MQTT message on 'homeassistant/light/0x001788010b48c3f4/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Desk Backlight/availability"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Desk Backlight/set","device":{"identifiers":["zigbee2mqtt_0x001788010b48c3f4"],"manufacturer":"Philips","model":"Hue white and color ambiance LightStrip plus (8718699703424)","name":"Desk Backlight","sw_version":"1.93.7"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":500,"min_mireds":150,"name":"Desk Backlight","schema":"json","state_topic":"zigbee2mqtt/Desk Backlight","supported_color_modes":["hs","color_temp"],"unique_id":"0x001788010b48c3f4_light_zigbee2mqtt"}'
Sep 02 15:10:56 zigbee2mqtt npm[3979]: Zigbee2MQTT:debug 2022-09-02 15:10:56: Received MQTT message on 'homeassistant/light/0x001788010b48c3f4/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Desk Backlight/availability"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/Desk Backlight/set","device":{"identifiers":["zigbee2mqtt_0x001788010b48c3f4"],"manufacturer":"Philips","model":"Hue white and color ambiance LightStrip plus (8718699703424)","name":"Desk Backlight","sw_version":"1.93.11"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":500,"min_mireds":150,"name":"Desk Backlight","schema":"json","state_topic":"zigbee2mqtt/Desk Backlight","supported_color_modes":["hs","color_temp"],"unique_id":"0x001788010b48c3f4_light_zigbee2mqtt"}
```

Here's the logs running from startup through to the first five minutes or so: https://gist.github.com/deviantintegral/0a9f85c831567752e6a09b6d6e7bce8b

Koenkk, 2022-09-06:
@deviantintegral I don't the the error originates from `Desk Backlight`. The published `color_mode` matches the `supported_color_modes`, e.g.:

```
Sep 02 15:10:53 zigbee2mqtt npm[3979]: Zigbee2MQTT:info  2022-09-02 15:10:53: MQTT publish: topic 'zigbee2mqtt/Desk Backlight', payload '{"brightness":254,"color":{"h":26,"hue":26,"s":100,"saturation":100,"x":0.5473,"y":0.4292},"color_mode":"hs","color_temp":500,"last_seen":"2022-09-02T19:10:34.668Z","linkquality":72,"state":"ON","update":{"state":"idle"},"update_available":null}'
```

deviantintegral, 2022-09-12:
I was all excited to step debug this when I noticed the log message changed in Home Assistant to include the entity ID: https://github.com/home-assistant/core/pull/77909

That showed that while I could trigger the error by setting `hue_move` on a single device, that it was a _group_ being logged in Home Assistant. In this case, I can trigger the error more simply by having conflicting color mode settings between a device and a group.

To avoid these errors, it seems like all devices and all groups would have to be set to use `hs`. Is there a reason why `xy` is the default?

Koenkk, 2022-09-12:
Ah great that you found the problem.

> Is there a reason why xy is the default?

Because X/Y is the default in Zigbee, according to the ZCL spec XY is mandatory, HS optional therefore we picked XY as the default for z2m

deviantintegral, 2022-10-08:
Setting `supported_color_modes` on all my groups containing hue colour bulbs has fixed this error. However, it doesn't look like this is documented anywhere. Would it make sense to add this to https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html ? Are there other keys to document too?

Koenkk, 2022-10-09:
There is a bit of docs on overriding discovery properties: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/guide/usage/integrations/home_assistant.md#exposing-switch-as-a-light but it would be good to add more. Would you mind making a PR?

Koenkk, 2022-10-11:
Yes any property can be overridden, but I suggest to add a special note for your use case.

closed 2022-11-11 by deviantintegral