Koenkk/zigbee2mqtt #13490: [New device support]: Tuya Zigbee 2CH LED controller

jacekmarek, 2022-08-10:
### Link

https://videosec.com/products/WZ1_Datasheet.pdf

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0xa4c138f8ec0e9858","nwkAddr":2569,"manufId":4417,"manufName":"_TZB210_rkgngb5o","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0501B","manufacturerName":"_TZB210_rkgngb5o","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{}},"genLevelCtrl":{"attributes":{"currentLevel":12}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff077814","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1660159971579,"defaultSendRequestWhen":"immediate"}

### Comments

I got stuck on control from push button.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =[
        {
        fingerprint: [{modelID: 'TS0501B', manufacturerName: '_TZ3000_4whigl8i'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_4whigl8i'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_9q49basr'},
            {modelID: 'TS0501B', manufacturerName: '_TZB210_rkgngb5o'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_grnwgegn'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_wuheofsg'}],
        model: 'TS0501B',
        description: 'Zigbee light',
        vendor: 'TuYa',
        extend: extend.light_onoff_brightness(),
        },
]

module.exports = definition;
```

### Supported color modes

mono/CCT

jacekmarek, 2022-08-12:
Another test:

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0501B',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZB210_rkgngb5o'
        },
    ],
    model: 'WZ1',
    vendor: 'TuYa',
    description: 'ZigBee & RF 2CH LED Controller',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
//		tz.tuya_dimmer_state,
		tz.light_onoff_brightness,
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

What works from Z2M : state, brightness, link quality
Pressing the button is not visible in the logs.
What am i doing wrong with debugging?

Koenkk, 2022-08-13:
Do you get updates for on/off brightness when changing the `configure` code to:

```js
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, [ 'genOnOff', 'genLevelCtrl']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
    },
```

jacekmarek, 2022-08-13:
I get updates now:
off -> on (click button):
```
2022-08-13 12:49:15Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
info 2022-08-13 12:49:15MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T12:49:14+02:00","linkquality":255,"state":"OFF"}'
```
on -> off (click button):
```
debug 2022-08-13 12:51:09Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info 2022-08-13 12:51:09MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T12:51:08+02:00","linkquality":255,"state":"OFF"}'
```
off -> on (push and hold button),  brightness from 0 to ~60%:
```
 2022-08-13 12:56:48Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
info 2022-08-13 12:56:48MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T12:56:47+02:00","linkquality":255,"state":"OFF"}'
debug 2022-08-13 12:56:53Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":220}' from endpoint 1 with groupID null
info 2022-08-13 12:56:53MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T12:56:52+02:00","linkquality":255,"state":"OFF"}'
```
on, push and hold button - brightness from max to min:
```
2022-08-13 12:59:23Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID null
info 2022-08-13 12:59:23MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T12:59:22+02:00","linkquality":255,"state":"OFF"}'
```
on, push and hold button - brightness from min to max:
```
2022-08-13 13:00:59Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
info 2022-08-13 13:00:59MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":254,"last_seen":"2022-08-13T13:00:58+02:00","linkquality":255,"state":"OFF"}'
```

Koenkk, 2022-08-13:
Great, what is left?

jacekmarek, 2022-08-13:
When I publish { "state": "on", "brightness": 1 } (1 is the minimum brightness ):
```
debug 2022-08-13 17:46:09Received MQTT message on 'zigbee2mqtt/0xa4c138f8ec0e9858/set' with data '{ "state": "on", "brightness": 1 }'
debug 2022-08-13 17:46:09Publishing 'set' 'brightness' to '0xa4c138f8ec0e9858'
info 2022-08-13 17:46:09MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":63,"last_seen":"2022-08-13T17:46:09+02:00","linkquality":255,"state":"ON"}'
info 2022-08-13 17:46:09MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":1,"last_seen":"2022-08-13T17:46:09+02:00","linkquality":255,"state":"ON"}'
debug 2022-08-13 17:46:09Received Zigbee message from '0xa4c138f8ec0e9858', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID null
info 2022-08-13 17:46:10MQTT publish: topic 'zigbee2mqtt/0xa4c138f8ec0e9858', payload '{"brightness":1,"last_seen":"2022-08-13T17:46:09+02:00","linkquality":255,"state":"ON"}'
```
I can still turn down the brightness with the button - the LED strip reduces the brightness to new minimum.

>Great, what is left?
Please complete the external converter file with the required parameters.

Koenkk, 2022-08-14:
> Double Click - Turn on at 100% or 10%(night light) and vice-versa.

https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

To add this option the correct data point for this setting has to be found: 

> I can still lower the brightness with the button, the LED strip reduces the brightness to new minimum.

Is it also possible to reduce the brightness lower than the `{ "state": "on", "brightness": 1 }` with the TuYa gateway? Then it might be that TuYa uses a different command for this and reverse engineering via the TuYa gateway is necessary (like the point above).

jacekmarek, 2022-08-14:
I haven't a TuYa gate, please skip these cases.

Koenkk, 2022-08-14:
Can you provide your external converter? Then I can add it.

Koenkk, 2022-08-16:
Thanks, added the device.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-16 by Koenkk

jacekmarek, 2022-08-17:
I confirm, it works. ☕ for you. 
Please delete effects from configuration, it is don't work.

Koenkk, 2022-08-17:
I'm not sure if he got it working completely, he didn't follow up on my latest comment.

mpetro89, 2022-09-25:
I got the same Tuya WZ1 device with a different modelID: "TS0503B" and manufacturerName: "_TZB210_nfzrlz29".

https://www.aliexpress.com/item/1005001884086871.html

Database entry:

{"id":9,"type":"Router","ieeeAddr":"0xa4c13820760a5d22","nwkAddr":51798,"manufId":4417,"manufName":"_TZB210_nfzrlz29","powerSource":"Mains (single phase)","modelId":"TS0503B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c85e59","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c85e59","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1455162343},"lastSeen":1664088690292,"defaultSendRequestWhen":"immediate"}

Please add support for that one.

Koenkk, 2022-09-28:
@mpetro89 `_TZB210_nfzrlz29` will be supported in the next release (1 Oct)

mpetro89, 2022-10-04:
@Koenkk you added {modelID: '**TS0502B**', manufacturerName: '_TZB210_nfzrlz29'}. But if you look my comment, I have different modelID: "**TS0503B**", with the same manufacturerName: '_TZB210_nfzrlz29'

Koenkk, 2022-10-04:
@mpetro89 sorry, added it now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

mpetro89, 2023-02-17:
> Now it's ok, but I don't receiving status updates, when pressing push button (ON/OFF, brightness)...

@Koenkk Can you please help me with that? I'm still not receiving status updates, when I'm pressing physical button...

Koenkk, 2023-02-18:
Try to setup reporting for the `genOnOff` cluster, `onOff` attribute in the z2m frontend -> device -> reporting tab.

mpetro89, 2023-03-02:
@Koenkk I tried to set reporting like this:

[image]

but it's the same result... I'm not recieving status updates.

mpetro89, 2023-03-02:
Sorry my bad! 
On/Off is working now. I also set "Min rep interval" to 1, and now it's working, every time I press the physical button.

Now I would like to get info for brightness too.
If I hold the physical button I can change the brightness, but I can't get new brightness value back to Z2M.

Koenkk, 2023-03-02:
To do this setup reporting for the `genLevelCtrl` cluster, `currentLevel` attribute.

mpetro89, 2023-03-02:
@Koenkk many thanks! It's worked.

finder6600, 2024-05-29:
> Hi, i'm trying to pair the WZ1 with my z2m (ver: 1.35.1 commit: [9eaaa0f](https://github.com/Koenkk/zigbee2mqtt/commit/9eaaa0f9d1725416639ee894fc45542be3a4753f)) but it is being recognized as "ts0503b" https://www.zigbee2mqtt.io/devices/TS0503B.html#tuya-ts0503b . Am i doing something wrong?
> 
> thanks in advance

did you solve this problem? When connecting for the first time, the controller was assigned correctly, but when connecting again it was recognized as "ts0503b"

akadaedalus, 2024-06-11:
I did some experiments with this device.  The documentation is extremely unclear but I think switching it between white and yellow modes changes the device model.  

Also double clicking on the button (which flashes the light ten times) supposedly clears the network, rather than pairing it.  I struggled to understand that for a while.   

The picture for the wz1 white mode seems wrong.  It's rectangular not circular.