Koenkk/zigbee2mqtt #13491: Big battery drain with TUYA TS0043 switch

Gymnae, 2022-08-10:
### What happened?

My white label TUYA TS00043 wall switches with the brand label MOES are burning through environment unfriendly and price cell batteries. In addition, the battery is reported with 100 all the time. I tried two different batches of cells to the same result.

The buttons are pressed maybe 5x a day but each cell battery doesn't last more than two months.

Here are some log files that might help. In this case, I pressed on button of the wall switch twice. 

```
Info 2022-08-10 22:00:00MQTT publish: topic 'zigbee2mqtt/Schalter Küche', payload '{"action":"1_single","battery":100,"device":{"applicationVersion":66,"dateCode":"","friendlyName":"Schalter Küche","hardwareVersion":1,"ieeeAddr":"0x847127fffee2fa1f","manufacturerID":4098,"manufacturerName":"_TZ3000_w8jwkczz","model":"TS0043","networkAddress":28894,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":54}'
Info 2022-08-10 22:00:01MQTT publish: topic 'zigbee2mqtt/Schalter Küche', payload '{"action":"","battery":100,"device":{"applicationVersion":66,"dateCode":"","friendlyName":"Schalter Küche","hardwareVersion":1,"ieeeAddr":"0x847127fffee2fa1f","manufacturerID":4098,"manufacturerName":"_TZ3000_w8jwkczz","model":"TS0043","networkAddress":28894,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":54}'
Info 2022-08-10 22:00:01MQTT publish: topic 'zigbee2mqtt/Schalter Küche/action', payload '1_single'
Info 2022-08-10 22:00:07MQTT publish: topic 'zigbee2mqtt/Schalter Küche', payload '{"action":"1_single","battery":100,"device":{"applicationVersion":66,"dateCode":"","friendlyName":"Schalter Küche","hardwareVersion":1,"ieeeAddr":"0x847127fffee2fa1f","manufacturerID":4098,"manufacturerName":"_TZ3000_w8jwkczz","model":"TS0043","networkAddress":28894,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":54}'
Info 2022-08-10 22:00:07MQTT publish: topic 'zigbee2mqtt/Schalter Küche', payload '{"action":"","battery":100,"device":{"applicationVersion":66,"dateCode":"","friendlyName":"Schalter Küche","hardwareVersion":1,"ieeeAddr":"0x847127fffee2fa1f","manufacturerID":4098,"manufacturerName":"_TZ3000_w8jwkczz","model":"TS0043","networkAddress":28894,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"linkquality":54}'
Info 2022-08-10 22:00:07MQTT publish: topic 'zigbee2mqtt/Schalter Küche/action', payload '1_single'
```

### What did you expect to happen?

According to the product description a CR2032 cell should last a year with normal usage

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20220726

### Adapter

Sonoff Zigbee3.0 USB Dongle

peteh, 2022-08-10:
I have similar experience with the Lidl Zigbee Buttons TS004F. My battery went empty yesterday even though I don't have the device for half a year yet.

Gymnae, 2022-08-11:
Good to hear I'm not the only one. 
What dongle do you use?

peteh, 2022-08-14:
SONOFF Zigbee 3.0 Plus USB

Gymnae, 2022-08-14:
Can you check which firmware and what transmit power you set?
It might be connected to the dongle we use.

peteh, 2022-08-15:
I'm using default settings and haven't configured anything in particular. 

From my config: 
Coordinator type
    zStack3x0

Coordinator revision
    20220219

Transmit power seems to be = 9

Gymnae, 2022-08-17:
Comparable to my setup, so maybe it's something with the dongle

Koenkk, 2022-09-19:
I'm not sure if this can be fixed from the z2m side. In order to investigate this further I need to have a sniff of 1 day usage of this switch. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Gymnae, 2022-09-19:
If I bring the switch back into usage, I will do that. For now, I'm attempting to switch (heh) to "Philips Hue Tag" switches, since they require no battery and can be had used on ebay.

peteh, 2022-09-19:
I unfortunately don't have a compatible stick for sniffing. However since i replaced the battery i haven't had issues so far again. Maybe it was just really low to begin with. 

The button does not report battery though, maybe this can be fixed to track it better without sniffing. If anybody could point me a bit to debug why it doesn't report battery even though it should i could try to fix it.

robertalexa, 2022-10-01:
this feels very similar to what i have described in my initial issue https://github.com/Koenkk/zigbee2mqtt/issues/8072 which in turns in references by other open issues with very similar behaviours.

@Koenkk since the last time the batteries died in Aug 2021 i have not even bothered with these. they just make nice design switches on my walls :( i am hoping to have some more time to continue looking into this and with your help figure out what is different between the tuya gateway and z2m that makes the device fall off the network

Koenkk, 2022-10-02:
@robertalexa could you provide a sniff starting from the point where these still work until they fall off? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

robertalexa, 2022-10-02:
@Koenkk see 8072 where i have provided new sniffs. Maybe close this as duplicate and reopen 8072. L

jan--m, 2022-10-04:
I'm also seeing 1 month battery life on TS0044 although it should last 1 year - so there's probably something z2m-specific draining the battery because I haven't seen any battery complaints in generic forums/reviews like [here](https://www.amazon.com/MOES-Wireless-Powered%EF%BC%88Not-Included%EF%BC%89-Automation/dp/B095LPC57N/)

I've also noticed that the battery indicator is stuck on 100% for 7 out of my 8 TS0044's - except for one where I replaced the battery with a used battery, and that's showing 73%...perhaps this battery indicator (not updating) issue is related to the battery draining, because somebody reported that disabling Powercfg solved the draining issue for all of their switches: https://github.com/Koenkk/zigbee2mqtt/issues/14157#issuecomment-1256959437

Gymnae, 2022-10-08:
Wow, sounds great. I would have not believed that this bug report not only links to a workaround but also leads to a PR. Great stuff :)

Koenkk, 2022-10-09:
This should be fixed now in the latest dev, credits go to @robertalexa 😄 (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-09 by Koenkk

robertalexa, 2022-12-08:
This suggests network instability and is the behaviour of the device. first button will wake the device up, rejoin, second click will actually work.

The battery drain issue was accompanied with the same behaviour, because every hour the wrong reporting configuration would crash the device and take it off the network. But since you are not experiencing any battery drain anymore it suggests that this is not happening, try to pair the switch through router devices that are closer to the switch, also, if you paired it to a device, then moved it into position, it might drop from the network before realising there is a router closer that it can use. Or it could be stubburn and insists to stick to the original device instead of the closer one.

After resolving the issue I have been using 4 such devices daily, multiple times a day with 1 single failure, and that was because i rebooted my home server, so z2m restarted, and because the devices are the sleepy type they didn''t get a chance to reconnect naturally to the network. But this is an exceptional circumstance, apart from this i would call it a 0% failure rate since early October.

Slyke, 2023-01-04:
Just discovered this too. Will the fix be merged into the master branch if not already? I've using the docker image `1.28.0` and just switched to `1.29.0` to see if this helps with the battery life.

robertalexa, 2023-01-04:
Merged in october, been released in the main branch in november

Electronlibre2012, 2023-02-25:
hello, doesnt work for me...

zigbee2mqtt 1.30.1-1
sonoff zigbee 3.0 dongle with last firmware
HaOss 2023.2.5

the batteries ( 2 x CR2032) are dead every month...the batteries cost half the switch lol

is there something i can do? delete and re add the device?

Thanks

swiergot, 2023-03-19:
@rbswift - did you manage to solve the sleep problem?

Electronlibre2012, 2023-03-19:
i have install a ZHA gateway (Ethernet silvercrest hacked) and put my device on it, same problem like with Z2M, battery drain.

So its not only related to Z2M but ZHA too...

Sadly i have buy a bunch of theses...for me its really unusuable...have to change 2 CR2032 battery per device every month...that will cost a lot after a year... 2 x1.50€ x 10 device = 30€ per month, 360€ of battery per year....really not nice for mother earth...

rbswift, 2023-03-20:
> @rbswift - did you manage to solve the sleep problem?

Sadly not. I thought the drain was improved in one version but then it got bad again. I'm not sure if it was after an upgrade. I gave up on the device and stopped using it.

swiergot, 2023-03-20:
@rbswift Thanks. May I ask what did you replace it with?

rbswift, 2023-03-20:
> @rbswift Thanks. May I ask what did you replace it with?

mostly using ikea tradfri buttons now

swiergot, 2023-04-19:
So it's been over a month since I inserted a new battery to one of my TS00043. Before the fix one battery would last around 2 weeks. Now it's still fine! Thanks, @robertalexa!

Unfortunately, I'm still experiencing the sleep issue described by @rbswift. Followed the advice from @robertalexa but it didn't help.

@robertalexa @Koenkk Any ideas how to move forward?

swiergot, 2023-04-20:
Ok, I have just solved it. Looked into logs and saw that the first press after a longer inactivity actually was detected by Z2M but the automation in HA did not work as expected. Turns out the automation had a template condition:

`{{ trigger.from_state.state == '' }}`

Apparently for some reason after a longer inactivity the entity's state is `None`. When a press happens, the state goes from `None` to `1_single` and immediately from `1 single` to an empty string. That's why the second (and subsequent) press works, until... the button goes to sleep again:

```
Zigbee2MQTT:info  2023-04-19 22:46:41: MQTT publish: topic 'zigbee2mqtt/Living room remote/action', payload '1_single'
Zigbee2MQTT:info  2023-04-19 23:36:58: MQTT publish: topic 'zigbee2mqtt/Living room remote', payload '{"action":null,"battery":100,"linkquality":40,"voltage":3000}'
```

Notice the action is `null`.

@Koenkk Is it a bug in Z2M?

Koenkk, 2023-04-20:
@swiergot can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

jan--m, 2023-06-08:
> zigbee2mqtt 1.30.1-1 sonoff zigbee 3.0 dongle with last firmware HaOss 2023.2.5
> the batteries ( 2 x CR2032) are dead every month...the batteries cost half the switch lol

@Electronlibre2012 I'm not sure how your battery is still getting drained - I have two rPi's with Dongle-P running just fine...

one with [1.30.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.30.1) on zStack3x0 20221102 ([TS0044](https://www.zigbee2mqtt.io/devices/TS0044.html#tuya-ts0044) with CR2430 battery at 83% after 6+ months)

another with 1.28.0-dev commit [52a0f68](https://github.com/Koenkk/zigbee2mqtt/commit/52a0f68) on zStack3x0 20220726 (8 of those TS0044 working smoothly)

maybe they don't always work on the first press, but that's not a show stopper

jan--m, 2023-06-14:
@Electronlibre2012 what specific quirk did you install?

Electronlibre2012, 2023-06-14:
> @Electronlibre2012 what specific quirk did you install?

[ts0041 to 46.zip](https://github.com/Koenkk/zigbee2mqtt/files/11745121/ts0041.to.46.zip)

just extract to /config/custom_zha_quirks, and create "custom_zha_quirks" folder first, then reboot HA and this quirk will be used next restart instead of those of Zigbee2Mqtt.

Enjoy

swiergot, 2023-06-14:
@Electronlibre2012 Isn't this already part of Z2M? I have had no issues since the fix in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4742.

atr00, 2023-06-26:
> 

Are these already part of the current ZHA?