Koenkk/zigbee2mqtt #13493: TS130F does not open as wide as in calibration

Chaaaaaaaaarlie, 2022-08-11:
### What happened?

After the TS130F has been calibrated, the shutter does not open completely and a little less after each closing of the shutter. It seems that the shutter closes “faster” than it opens. Therefore, the zero point shifts further and further down.

[You can see it here - YouTube video](https://www.youtube.com/shorts/X5_lqtaQuLE)

### What did you expect to happen?

The shutter should always open and close the same way.

### How to reproduce it (minimal and precise)

Close the shutter and open it again. After several passes, almost not open at all.

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

https://pastebin.com/Vn9QcbYB

colseven, 2022-08-28:
Same Problem here. Thats not fine. What can i do?

convicte, 2022-09-01:
I am fairly certain it's a matter of 'drifting'/desynchronizing time of the opening with the shutter physical position. 

In my case, as I use an automation, the shutter go up/down often throughout the day, but after a few movements it doesn't fully open, until I close it completely for the night.  At that point it closes all the way down, while you can hear the relay switching off a few seconds after it closed completely. This means that the calibrated time was 'offset down' as compared to the position of the shutter. 

In other words, the shutter physical position was, let's say 90% in it's travel, but there was 0s time in the calculated motor time to go up, and full, let's say 25 seconds, to go down, even though there were only 22 seconds needed to close it.

[image]

With every down/up/down/up (but not fully closed) the time of closing and opening desynchronizes more and more with the actual physical position. This is possible if the timer runs a few milliseconds after the motor actually stopped already. Then the more we go up and down, without fully closing, the more the position desynchronizes with the timer.

Chaaaaaaaaarlie, 2022-09-01:
That is exactly the problem I'm having.

colseven, 2022-09-01:
I have the same problem. How can we change that?

Chaaaaaaaaarlie, 2022-09-01:
Guess there should be a different calibration for up and down.

convicte, 2022-09-01:
I am not sure if it's the problem of calibration per se, but I may be wrong.

What it feels like is that the timer doesn't stop the moment the motor stops, which may simply be the limitation of communication. I am not even sure where calibration is stored, such that the time it takes for the timer to stop cannot be shortened. If there is 500 ms between the motor coming to a full stop and the timer stopping, this will mean that you are missing 2-3 seconds after only 4 movements, which during the day for an automation driven shutter is quite normal. 3 seconds is a good 10-15% total motion range of a door shutter, and more for a smaller window.

Not sure if it's intrinsic to the controllers or if something can be done from within Z2m. Not feeling optimistic though.

convicte, 2022-09-05:
@McGiverGim 
Not sure if you came across anything that would pertain to this issue as well in your review of the device and various commits?
It seems it's an omnipresent wove for many users.

Cheers!!

famostwald, 2022-09-05:
since the last update the invert_cover doesn't work anymore either, can someone confirm that?

colseven, 2022-09-05:
since the last update the invert_cover doesn't work anymore either, can someone confirm that?

McGiverGim, 2022-09-05:
The calibration is done by the device itself, not by zigbee2mqtt. Mine works well.

convicte, 2022-09-05:
> The calibration is done by the device itself, not by zigbee2mqtt. Mine works well.

And you don't have the subtle drift when you open and close by 20-30% up/down?
That would be interesting to see what's different - hardware/firmware?

convicte, 2022-09-05:
> since the last update the invert_cover doesn't work anymore either, can someone confirm that?

My invert function works, but it's not resolving the problem, so that's besides the point of the issue at hand.

McGiverGim, 2022-09-06:
> And you don't have the subtle drift when you open and close by 20-30% up/down?
> That would be interesting to see what's different - hardware/firmware?

None that I've noticed. For example I have an automation that moves down the cover with the sun several times a day, and I've not noticed any problem. Maybe it needs move movements for the problem to appear.

The "calibration" is an operation controlled by the device. Zigbee2mqtt only says to the device when to start the calibration, and when to finish. The time is stored by the device, and it maintains the count of time when you open or close the cover.

Maybe is a problem of the motor that does not start instantly, specially in one of the directions? If the motor takes some time it can desynchronize the counter.

I can say what version of firmware I have. I don't see a place where I can consult it.

convicte, 2022-09-06:
Makes perfect sense!
Thanks for the thorough overview.
I guess our motors and delay may simply differ, thus some experience it and some don't.

Cheers!!

Chaaaaaaaaarlie, 2022-09-06:
I'll try to maybe get it working with a Shelly 2.5.
It's ordered and should arrive this week.
Will report back then.

Skysurfer-14, 2022-09-12:
I have the same problem. 

I made the calibration time a few seconds longer than the move up time of the cover, but as I move the cover up and down without hitting the down position, the upper stop position gets more and more down. To reset this failure, I move to the down position. After this the upper stop is correct (for the moment).

I think, that this depends on the firmware of the TuYa TS130F and therefor we are not able to fix this issue.

convicte, 2022-09-12:
These are my findings as well, and I agree with your evaluation.

I wonder if there is any route to bring this to the attention of the OEM, whomever it may be.

closed 2022-10-20 by github-actions[bot]

guim31, 2022-11-11:
I face the same problem here.... Any suggestions ?