Koenkk/zigbee2mqtt #13501: Legrand 067775/741811 does not report Current/Voltage values

CiccioDroid, 2022-08-11:
### What happened?

After device pairing it report only on/off status and power consumption.
I tried using developer console and after setting the values (see image below) i have 0V and 0A for values rmsVoltage and rmsCurrent, that mathing with values used in js file.
[image]
If i add these values to Report tab in frontend
[image]
I get the error below:
 2022-08-11 15:46:03Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'ConfigureReporting  haElectricalMeasurement([{"attribute":"rmsVoltage","minimumReportInterval":60,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')'

### What did you expect to happen?

I expected that values of Current and Voltage is reported.

### How to reproduce it (minimal and precise)

Pair device and see that Voltage and Current is null

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

26660700

### Adapter

Conbee 2

### Debug log

Info 2022-08-11 15:41:20MQTT publish: topic 'zigbee2mqtt/lavastoviglie', payload '{"action":null,"current":null,"last_seen":"2022-08-11T15:41:19+02:00","linkquality":135,"power":19,"state":"ON","update":{"state":"idle"},"update_available":false,"voltage":null}'

Koenkk, 2022-08-11:
`UNREPORTABLE_ATTRIBUTE` indicates that the device does not support reporting voltage

CiccioDroid, 2022-08-11:
Here it's reported that it support Voltage and Current
https://www.zigbee2mqtt.io/devices/067775_741811.html
It's wrong?

Koenkk, 2022-08-12:
I think so, @pirracas77  does the voltage/current work for your Legrand 741811?

pirracas77, 2022-08-13:
> I think so, @pirracas77 does the voltage/current work for your Legrand 741811?

I would say it doesn't as far as I remember , but I'm not at home these days to check it.
I will check when I'm back and I'll comment again

pirracas77, 2022-08-22:
Confired, only power is reported 
[image]

novisys, 2022-08-22:
Thanks for the response!
But Why just W is reported?
How we can try others parameters, because we all use the configuration for old plug, maybe this one has been changed by LEGRAND and need to be created as a new device on z2m ?

Koenkk, 2022-08-22:
Seems the device doesn't support voltage and current, I've removed it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-22 by Koenkk