Koenkk/zigbee2mqtt #13511: Eco-Dim.05 reporting as not supported after pairing

FSJHecker, 2022-08-12:
### What happened?

Tried adding a new EcoDim Eco-Dim.05. Interview = ok. Support status = unsupported.
[image]

After studying the contents of database.db, I noticed that there are no deviceID's registered for the endpoints of the device

After adding this external converter (where I removed the deviceID's from the fingerprint) the device reported as supported.

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const e = exposes.presets;
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');

const definition = {
    fingerprint: [
		{type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
			{ID: 1, profileID: 260, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
			{ID: 2, profileID: 260, inputClusters: [0, 3, 4, 5, 6, 8], outputClusters: []},
			{ID: 242, profileID: 41440, inputClusters: [], outputClusters: [33]},
		]},
	],
	model: 'Eco-Dim.05',
	vendor: 'EcoDim',
	description: 'LED dimmer duo 2x 0-100W',
	extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
	meta: {multiEndpoint: true},
	exposes: [e.light_brightness().withEndpoint('left'), e.light_brightness().withEndpoint('right')],
	configure: async (device, coordinatorEndpoint, logger) => {
		await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
		for (const ep of [1, 2]) {
			const endpoint = device.getEndpoint(ep);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
			await reporting.onOff(endpoint);
			await reporting.brightness(endpoint);
		}
	},
	endpoint: (device) => {
		return {'left': 2, 'right': 1};
	},
};

module.exports = definition;
`

### What did you expect to happen?

I expected that the Eco-Dim.05 was supported without an external converter

### How to reproduce it (minimal and precise)

Pair a Eco-Dim.05

### Zigbee2MQTT version

1.27.0 and Dev

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus–ZBDongle-E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9314835/log.txt)
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/9314950/database.db.txt)

Koenkk, 2022-08-13:
Can you try to force remove the device (from z2m frontend), restart z2m and repair your device?

FSJHecker, 2022-08-13:
Same result. I've tried it several times just to be sure. (Of course I removed my external converter first)

Koenkk, 2022-08-13:
Ok, removed the `deviceId` now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-13 by Koenkk