Koenkk/zigbee2mqtt #13512: [New device support]: Giex QT06 Smart Valve - changed manufacturer (_TZE200_a7sghmms)

hkoci, 2022-08-12:
### Link

https://www.zigbee2mqtt.io/devices/QT06.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13891356bcd62","nwkAddr":19278,"manufId":4417,"manufName":"_TZE200_a7sghmms","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���*\u0011","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":72}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1660233477569,"defaultSendRequestWhen":"immediate"}

### Comments

The Giex QT06 garden sprinkler has seemed to change its manufacturer code for newer batches, I have been able to get it working by following @Batou38 [discussion link](https://github.com/Koenkk/zigbee2mqtt/discussions/13487) in changing the manufacturer code to the new batch and I can confirm Zigbee2Mqtt does pair with the device and I was able to switch the valve on and off.

I was able to get this working on Zigbee2Mqtt by downloading the [giex.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/21a66c05aa533de356a51c8417073f28092c6e9d/devices/giex.js) file and changing the imports and manufacturer code.

In case anyone else has purchased this a7sghmms model, you'll need to manually import the giex.js code for the sh1btabb and then modify the manufacturer to a7sghmms.

- Changed the imports at the top of imported repo js to include zigbee-herdsman-converters as this is taken directly from the repo source.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes'); 
const tuya = require('zigbee-herdsman-converters/lib/tuya');
```

- Modify manufacturer code to support the new product (the functionality seems to all be working, just a change of manufacturer it looks like).

`fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_a7sghmms'}],`

- Insert this js file into the zigbee2mqtt folder (if running HA, this is the /config/zigbee2mqtt folder)
- Add the giex.js external adapter to the zigbee2mqtt configuration (in HA, this is /config/zigbee2mqtt/configuration.yaml)
```
external_converters:
  - giex.js
```

So far, I have been able to test the valve functionality, but I have not yet installed this in my garden yet to test the flow exposures - though it does seem to function all well.

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const dataPoints = {
    // GIEX Water Valve
    giexWaterValveState: 2,
    giexWaterValveMode: 1,
    giexWaterValveIrrigationTarget: 104,
    giexWaterValveWaterConsumed: 111,
    giexWaterValveIrrigationStartTime: 101,
    giexWaterValveIrrigationEndTime: 102,
    giexWaterValveLastIrrigationDuration: 114,
    giexWaterValveBattery: 108,
    giexWaterValveCycleIrrigationNumTimes: 103,
    giexWaterValveCycleIrrigationInterval: 105,
    giexWaterValveCurrentTempurature: 106,
};

const fzLocal = {
    giex_water_valve:
    {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp;
                switch (dp) {
                case dataPoints.giexWaterValveState: {
                    return {state: value ? 'ON': 'OFF'};
                }
                case dataPoints.giexWaterValveMode: {
                    return {mode: value ? 'capacity': 'duration'};
                }
                case dataPoints.giexWaterValveIrrigationTarget: {
                    return {irrigation_target: value};
                }
                case dataPoints.giexWaterValveCycleIrrigationNumTimes: {
                    return {cycle_irrigation_num_times: value};
                }
                case dataPoints.giexWaterValveCycleIrrigationInterval: {
                    return {cycle_irrigation_interval: value};
                }
                case dataPoints.giexWaterValveWaterConsumed: {
                    return {water_consumed: value};
                }
                case dataPoints.giexWaterValveIrrigationStartTime: {
                    return {irrigation_start_time: value};
                }
                case dataPoints.giexWaterValveIrrigationEndTime: {
                    return {irrigation_end_time: value};
                }
                case dataPoints.giexWaterValveLastIrrigationDuration: {
                    return {last_irrigation_duration: value};
                }
                case dataPoints.giexWaterValveBattery: {
                    return {battery: value};
                }
                case dataPoints.giexWaterValveCurrentTempurature: {
                    return; // Do Nothing - value ignored because isn't a valid tempurature reading.  Misdocumented and usage unclear
                }
                default: {
                    meta.logger.warn(`fz:giex_water_valve: NOT RECOGNIZED DP #${dp} with VALUE = ${value}`);
                }
                }
            }
        },
    },
};

const tzLocal = {
    giex_water_valve:
    {
        key: ['mode', 'irrigation_target', 'state', 'cycle_irrigation_num_times', 'cycle_irrigation_interval'],
        convertSet: async (entity, key, value, meta) => {
            switch (key) {
            case 'mode': {
                let mode = 0;
                if (value === 'duration') mode = 0;
                else if (value === 'capacity') mode = 1;
                await tuya.sendDataPointBool(entity, dataPoints.giexWaterValveMode, mode);
                return {state: {mode: value}};
            }
            case 'irrigation_target':
                await tuya.sendDataPointValue(entity, dataPoints.giexWaterValveIrrigationTarget, value);
                return {state: {irrigation_target: value}};
            case 'state':
                await tuya.sendDataPointBool(entity, dataPoints.giexWaterValveState, value === 'ON');
                break;
            case 'cycle_irrigation_num_times':
                await tuya.sendDataPointValue(entity, dataPoints.giexWaterValveCycleIrrigationNumTimes, value);
                return {state: {cycle_irrigation_num_times: value}};
            case 'cycle_irrigation_interval':
                await tuya.sendDataPointValue(entity, dataPoints.giexWaterValveCycleIrrigationInterval, value);
                return {state: {cycle_irrigation_interval: value}};
            default: // Unknown key warning
                meta.logger.warn(`tz.GIEX_water_valve: Unhandled key ${key}`);
            }
        },
    },
};

module.exports = [
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_a7sghmms'}],
        model: 'QT06',
        vendor: 'GiEX',
        description: 'Water irrigation valve',
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fzLocal.giex_water_valve],
        toZigbee: [tzLocal.giex_water_valve],
        exposes: [
            e.battery(),
            exposes.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('State'),
            exposes.enum('mode', ea.STATE_SET, ['duration', 'capacity']).withDescription('Irrigation mode'),
            exposes.numeric('irrigation_target', exposes.access.STATE_SET).withValueMin(0).withValueMax(1440).withUnit('mintues or Litres')
                .withDescription('Irrigation Target, duration in minutes or capacity in Liters (depending on mode)'),
            exposes.numeric('cycle_irrigation_num_times', exposes.access.STATE_SET).withValueMin(0).withValueMax(100).withUnit('#')
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            exposes.numeric('cycle_irrigation_interval', exposes.access.STATE_SET).withValueMin(0).withValueMax(1440).withUnit('min')
                .withDescription('Cycle irrigation interval'),
            exposes.numeric('irrigation_start_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation start time'),
            exposes.numeric('irrigation_end_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation end time'),
            exposes.numeric('last_irrigation_duration', exposes.access.STATE).withUnit('min')
                .withDescription('Last irrigation duration'),
            exposes.numeric('water_consumed', exposes.access.STATE).withUnit('L')
                .withDescription('Water consumed (Litres)'),
        ],
    },
];
```

Koenkk, 2022-08-12:
Has just been added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4524

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-12 by Koenkk

ayasystems, 2024-06-23:
I am trying to get  _TZE204_a7sghmms works with this code and it is rasing the following error:

Exception while calling fromZigbee converter: tuya.getDataValue is not a function}

Can you give me a hand? It seems like this adaptation is old and does not use the tuya.getData function properly

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const dataPoints = {
    // GIEX Water Valve
    giexWaterValveState: 2,
    giexWaterValveMode: 1,
    giexWaterValveIrrigationTarget: 104,
    giexWaterValveWaterConsumed: 111,
    giexWaterValveIrrigationStartTime: 101,
    giexWaterValveIrrigationEndTime: 102,
    giexWaterValveLastIrrigationDuration: 114,
    giexWaterValveBattery: 108,
    giexWaterValveCycleIrrigationNumTimes: 103,
    giexWaterValveCycleIrrigationInterval: 105,
    giexWaterValveCurrentTempurature: 106,
};

const fzLocal = {
    giex_water_valve:
    {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            for (const dpValue of msg.data.dpValues) {
                const value = tuya.getDataValue(dpValue);
                const dp = dpValue.dp;
                switch (dp) {
                case dataPoints.giexWaterValveState: {
                    return {state: value ? 'ON': 'OFF'};
                }
                case dataPoints.giexWaterValveMode: {
                    return {mode: value ? 'capacity': 'duration'};
                }
                case dataPoints.giexWaterValveIrrigationTarget: {
                    return {irrigation_target: value};
                }
                case dataPoints.giexWaterValveCycleIrrigationNumTimes: {
                    return {cycle_irrigation_num_times: value};
                }
                case dataPoints.giexWaterValveCycleIrrigationInterval: {
                    return {cycle_irrigation_interval: value};
                }
                case dataPoints.giexWaterValveWaterConsumed: {
                    return {water_consumed: value};
                }
                case dataPoints.giexWaterValveIrrigationStartTime: {
                    return {irrigation_start_time: value};
                }
                case dataPoints.giexWaterValveIrrigationEndTime: {
                    return {irrigation_end_time: value};
                }
                case dataPoints.giexWaterValveLastIrrigationDuration: {
                    return {last_irrigation_duration: value};
                }
                case dataPoints.giexWaterValveBattery: {
                    return {battery: value};
                }
                case dataPoints.giexWaterValveCurrentTempurature: {
                    return; // Do Nothing - value ignored because isn't a valid tempurature reading.  Misdocumented and usage unclear
                }
                default: {
[... 43 lines cut ...]
        model: 'QT06',
        vendor: 'GiEX',
        description: 'Water irrigation valve',
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fzLocal.giex_water_valve],
        toZigbee: [tzLocal.giex_water_valve],
        exposes: [
            e.battery(),
            exposes.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('State'),
            exposes.enum('mode', ea.STATE_SET, ['duration', 'capacity']).withDescription('Irrigation mode'),
            exposes.numeric('irrigation_target', exposes.access.STATE_SET).withValueMin(0).withValueMax(1440).withUnit('mintues or Litres')
                .withDescription('Irrigation Target, duration in minutes or capacity in Liters (depending on mode)'),
            exposes.numeric('cycle_irrigation_num_times', exposes.access.STATE_SET).withValueMin(0).withValueMax(100).withUnit('#')
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            exposes.numeric('cycle_irrigation_interval', exposes.access.STATE_SET).withValueMin(0).withValueMax(1440).withUnit('min')
                .withDescription('Cycle irrigation interval'),
            exposes.numeric('irrigation_start_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation start time'),
            exposes.numeric('irrigation_end_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation end time'),
            exposes.numeric('last_irrigation_duration', exposes.access.STATE).withUnit('min')
                .withDescription('Last irrigation duration'),
            exposes.numeric('water_consumed', exposes.access.STATE).withUnit('L')
                .withDescription('Water consumed (Litres)'),
        ],
    },
];`

DanielTraining, 2024-08-01:
Hi @ayasystems , did you manage to fix this? I have the same problem

ayasystems, 2024-08-01:
Yes, i did. I'm out of home for holiday, please write me in a fee days to put the code

ayasystems, 2024-08-02:
> Hi @ayasystems , did you manage to fix this? I have the same problem

Try this and let me know please:

```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_a7sghmms',
        },
    ],
    model: 'QT06 NEW',
    vendor: 'GiEX NEW',
    description: 'Water irrigation valve NEW',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        exposes: [
		    e.battery(),
            exposes.binary('state', ea.STATE_SET, 'ON', 'OFF').withDescription('State'),
            exposes.enum('mode', ea.STATE_SET, ['duration', 'capacity']).withDescription('Irrigation mode'),
            exposes.numeric('irrigation_target', exposes.access.STATE_SET).withValueMin(0).withValueMax(240).withUnit('seconds or Litres')
                .withDescription('Irrigation Target, duration in minutes or capacity in Liters (depending on mode)'),
            exposes.numeric('cycle_irrigation_num_times', exposes.access.STATE_SET).withValueMin(0).withValueMax(100).withUnit('#')
                .withDescription('Number of cycle irrigation times, set to 0 for single cycle'),
            exposes.numeric('cycle_irrigation_interval', exposes.access.STATE_SET).withValueMin(0).withValueMax(1440).withUnit('min')
                .withDescription('Cycle irrigation interval'),
            exposes.numeric('irrigation_start_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation start time'),
            exposes.numeric('irrigation_end_time', ea.STATE).withUnit('GMT+8').withDescription('Irrigation end time'),
            exposes.numeric('last_irrigation_duration', exposes.access.STATE).withUnit('min')
                .withDescription('Last irrigation duration'),
            exposes.numeric('water_consumed', exposes.access.STATE).withUnit('L')
                .withDescription('Water consumed (Litres)'),
        ],
        meta: {
            tuyaDatapoints: [
			

				[108,'battery',tuya.valueConverter.raw],
				[2, 'state', tuya.valueConverter.trueFalse1],
				[1, 'mode', tuya.valueConverter.trueFalse1],
				[104,'irrigation_target',tuya.valueConverter.raw],
				[111,'water_consumed',tuya.valueConverter.raw],
				[101,'irrigation_start_time',tuya.valueConverter.raw],
				[102,'irrigation_end_time',tuya.valueConverter.raw],
				[114,'last_irrigation_duration',tuya.valueConverter.raw],
				[103,'cycle_irrigation_num_times',tuya.valueConverter.raw],
				[105,'cycle_irrigation_interval',tuya.valueConverter.raw],
				[106,'Temperature',tuya.valueConverter.raw],

            ],
    },
};

module.exports = definition;
```

DanielTraining, 2024-08-02:
amazing! thanks a lot @ayasystems

Koenkk, 2024-08-03:
@ayasystems if this works, could you submit a pr by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/giex.ts)?

ayasystems, 2024-08-03:
> @ayasystems if this works, could you submit a pr by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/giex.ts)?

[image]

Koenkk, 2024-08-04:
Can you try to create a fork?

ayasystems, 2024-08-07:
> Can you try to create a fork?

I did it but actions checks rise an error... I dont understand how works component definition so i don't know whats the error

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7837

[image]