Koenkk/zigbee2mqtt #13518: Flashing issue?

Varkzor, 2022-08-12:
### What happened?

I flashed my Sonoff Zigbee 3.0 Dongle with the firmware listed here.
After it is flashed successfully i understand it should generate a wifi network with TASMOTA_*** which i can't see anywhere.

Either i didn't intall it correctly, or i did something wrong along the way.
I have no idea, but i'm quite stuck in my Home Automation set-up to read the energy consumption from an Owon device.

### Zigbee2MQTT version

None yet

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219.zip

### Adapter

sonoff zigbee 3.0 dongle

Varkzor, 2022-08-12:
Additional info: when flashing it with Flash Programmer 2 it says it correctly flashed the device.
How-ever i see no wifi tasmota network.
Device manager shows a Silicon Labs CP210x USB to UART Bridge (COM4)

Koenkk, 2022-08-13:
> After it is flashed successfully i understand it should generate a wifi network with TASMOTA_*** which i can't see anywhere.

This is not the case, I'm think you are confused with the tasmota project: https://tasmota.github.io/docs/ . for any further help on setting up z2m I suggest using the discussion section: https://github.com/Koenkk/zigbee2mqtt/discussions

closed 2022-08-13 by Koenkk