Koenkk/zigbee2mqtt #13519: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251

johntdyer, 2022-08-12:
### What happened?

Noticed this in my logs this morning... ti seems to be impacting performance but devices do response, albiet slower then normal

```
022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:38Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]: 2022-08-12 17:03:39.175 INFO ZWAVE: Node 85: value updated: 37-0-currentValue false => false
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]: 2022-08-12 17:03:39.176 INFO ZWAVE: Node 85: value updated: 37-0-currentValue false => false
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]: 2022-08-12 17:03:39.176 INFO ZWAVE: Node 85: value updated: 37-0-currentValue false => false
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:39Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]: 2022-08-12 17:03:40.184 INFO ZWAVE: Node 85: value updated: 37-0-currentValue false => false
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]: 2022-08-12 17:03:40.185 INFO ZWAVE: Node 85: value updated: 37-0-targetValue false => false
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (node:internal/async_hooks:454:18)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at afterDns (node:dgram:660:5)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at processTicksAndRejections (node:internal/process/task_queues:84:21)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]: Encountered MDNS socket error on socket 'wg0' : Error: send Unknown system error -126 224.0.0.251:5353
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at doSend (node:dgram:714:16)
2022-08-12T17:03:40Z zwavejs2mqtt.zwavejs2mqtt[3143839]:     at defaultTriggerAsyncIdScope (`

```

### What did you expect to happen?

not to throw the error

### How to reproduce it (minimal and precise)

zwavejs2mqtt: 6.15.2.5c2434f
zwave-js: 9.6.2
home id: 3410551786
home hex: 0xcb48e3ea

### Zigbee2MQTT version

6.15.2.5c2434f

### Adapter firmware version

these instructions dont work

### Adapter

Aeotec Z-Wave Stick

### Debug log

Lost logs after restart

sjorge, 2022-08-12:
There don't appear to be from zigbee2mqtt but [zwavejs2mqtt](https://github.com/zwave-js/zwavejs2mqtt)

Koenkk, 2022-08-13:
Indeed not z2m related.

closed 2022-08-13 by Koenkk