Koenkk/zigbee2mqtt #13520: [New device support]: Bosch Outdoor Siren

safakaltun, 2022-08-12:
### Link

https://www.bosch-smarthome.com/uk/en/products/devices/outdoor-siren/

### Database entry

{"id":66,"type":"Unknown","ieeeAddr":"0x003c84fffeeda00b","nwkAddr":29598,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1660330204622,"defaultSendRequestWhen":"immediate"}

### Comments

I would like to add support for this device, but I got stuck in the pairing phase. Although the device starts pairing, it cannot complete the interview. Pairing with the original hub was done using the QR code on the back of the siren, which contains the IEEE link key. I am not sure if it has anything to do with it or not. 

I factory reset the device numerous times with no help. I also tried pairing it in both legacy and "smart home controller" modes which is set by the PIN 8 behind the siren. 

Attaching the herdsman logs as well. I am not sure if the length of the log is enough, but the device more or less runs in a loop where z2m tries interviewing, interview fails and starts over again. 

[herdsman.log](https://github.com/Koenkk/zigbee2mqtt/files/9329318/herdsman.log)

danieledwardgeorgehitchcock, 2022-08-22:
Just to confirm, I have recently purchased this siren to try and integrate into Z2M and am experiencing the same as you. It begins to pair, then seems to fail interview.

I have purchased a used Bosch Smart Home Controller to see if I can sniff what is going on with the interview - I am wondering if a key is exchanged given that you have to scan a QR code during adding

danieledwardgeorgehitchcock, 2022-08-23:
So, I received the Smart Home Controller today and bound the siren to it. Points to note:

The app recommends that you switch off the siren and ten switch it on before pairing

The light on the bottom of the siren turns Amber during interview

The switches on the device have no affect in 'Smart Mode' apart from the actual Smart Mode switch 8

I will sort out my sniffing setup and begin capturing device communication during adding to the controllers network

danieledwardgeorgehitchcock, 2022-08-23:
Ok, this is a weird one.

I have captured the join but cannot find a single communication within it that exchanges any sort of Network Encryption Key yet the traffic is definitely encrypted between the devices..?

Here is the capture:

[Outdoor Siren Join](https://drive.google.com/file/d/1vyE3rQILQwFMs_H6hE76qeT387INbqOi/view?usp=sharing)

@Koenkk - any ideas how I can start picking this one apart?

Koenkk, 2022-08-24:
@danieledwardgeorgehitchcock the `Transport Key` is probably packet 23 in the sniff. It seems their gateway uses a custom link key (different the standard one). Can you provide a sniff when pairing to z2m?

danieledwardgeorgehitchcock, 2022-08-25:
Here is the sniff of the device joining Z2M. Not much to report here really, it joins the network, fails the interview, then drops off of the network and the process repeats again:

[Outdoor Siren Join Z2M](https://drive.google.com/file/d/1g-2dLh_wMYJ9L6W2eKobfmpC2xiVjvuy/view?usp=sharing)

safakaltun, 2022-08-26:
@danieledwardgeorgehitchcock I don’t have access to my device at the moment, but have you tried sniffing with PIN8 ON and OFF? I wonder if it changes how the device communicates. I remember that when PIN8 was ON, the device would continuously connect, try interviewing, fail and remove itself from z2m in an infinite loop. When the PIN8 was OFF, the device would connect, start interviewing, fail and stay in z2m database as a device that failed interview.

danieledwardgeorgehitchcock, 2022-08-26:
Interesting concept! I thought pin 8 may have been for switching it between smart I.e. wireless mode, and dumb I.e. wired and controlled with switches mode. I'll also try and do another sniff with the switch flipped this morning

safakaltun, 2022-08-26:
I don’t know if it would help, but below is the content of the QR code of the device

RB01SG0D83101826480040000000000000000000003C84FFFEEDA00BDLKCE5D5C1E3D59D1440ABFE85251205C36ADE4

safakaltun, 2022-08-26:
> Interesting concept! I thought pin 8 may have been for switching it between smart I.e. wireless mode, and dumb I.e. wired and controlled with switches mode. I'll also try and do another sniff with the switch flipped this morning

It says so in the manual, but it still attempts connecting when you start the device up or factory reset it.

danieledwardgeorgehitchcock, 2022-08-26:
> 

The last 36 digits are the install code, then prior to that you have the 16 digit EUI-64 MAC address the not sure what the 40 digits prior to that are, but it looks to be 21 / 22 digits and some 0 padding - the interesting thing is that this part matches mine

danieledwardgeorgehitchcock, 2022-08-26:
Here is the sniff of the device joining Z2M in 'Legacy' mode:

[Outdoor Siren Join Z2M Legacy](https://drive.google.com/file/d/11XNTR6W2nGbHdgfeH0fjHkTElaGnfOT_/view?usp=sharing)

Koenkk, 2022-08-26:
@danieledwardgeorgehitchcock seems the device keeps rejoining, is it possible to pair it to the bosch gateway in legacy mode?

danieledwardgeorgehitchcock, 2022-08-26:
No - it will only pair to the bosch controller in 'smart' mode. Having the device in 'legacy' mode while attempting to pair causes discovery to fail

Koenkk, 2022-08-26:
@danieledwardgeorgehitchcock that is a bummer, I'm not sure if Bosch would be open to helping on this but other that that I see no other way to solve this.

danieledwardgeorgehitchcock, 2022-08-26:
@Koenkk - I have just found this snippet of information:

"The ZigBee implementation of Bosch uses an advanced encryption technology with device specific link keys, which are being used for initial connection to the Controller, where individual encryption keys are being shared. Whilst other manufacturers use publicly available keys for the initial setup, the method of using device-specific encryption keys makes ZigBee a very secure wireless transmission protocol. Presumably, similar technologies are also used for the proprietary 868Mhz-protocol – these devices also have the individual device key printed on it."

Found [here](https://www.iot-tests.org/2017/08/bosch-smart-and-secure-starter-kit/) 

Does this help with any guesses on how the device is added to the controller?

safakaltun, 2022-08-27:
I am not sure if it would be helpful in any way, but found some [documentation related to the bosch smart controller](https://github.com/BoschSmartHome/bosch-shc-api-docs). Although this requires an existing controller, it might still be helpful. 

Here is the [Python library](https://github.com/tschamm/boschshcpy)  for the Rest API.

I have to say though, it is odd that the siren is neither mentioned in the documentation nor the python library they have for the Rest API.

Koenkk, 2022-08-27:
@danieledwardgeorgehitchcock so maybe the devices pairs and expect the `transport key` message to be encrypted with this custom key. This is something we do not support yet and will require some changes in the coordinator fw. I'm not sure if you were able to figure out the specific encryption key of your device already? If yes, then you should also be able to decrypt your sniff with the original gateway.

danieledwardgeorgehitchcock, 2022-08-27:
Thanks, I'll look into it. I'm guessing that it may be something to do with the install code. Hopefully, Bosch don't do anything too clever with it. Any ideas how I would go about decrypting the traffic using the 36 digit key as opposed to the standard 16? Is it as easy as plumbing it into Wireshark in the same fashion as the network key?

Koenkk, 2022-08-27:
> Any ideas how I would go about decrypting the traffic using the 36 digit key as opposed to the standard 16? Is it as easy as plumbing it into Wireshark in the same fashion as the network key?

I have no experience with that but I would expect that you can just paste it in.

danieledwardgeorgehitchcock, 2022-08-27:
That didn't work unfortunately however, I have found the documentation from the Zigbee chip manufacturer on how they implement install codes in their SDK.

I am going to do some further reading on this:

[Silicone Labs Zigbee Install Codes](https://www.silabs.com/documents/public/application-notes/an1089-using-installation-codes-with-zigbee-devices.pdf)

danieledwardgeorgehitchcock, 2022-08-27:
This looks promising:

> To derive a link key from the installation code and store that into the key table on the Z3 Gateway, which acts as the Trust Center
> for the centralized network, enter:
> `option install-code <link key table index> {<Joining Node’s EUI64>} {<installation code + 2-byte CRC>}`
> For example:
> `option install-code 0 {00 0B 57 FF FE 07 A9 E3} {88 77 66 55 44 33 22 11 11 22 33 44 55 66 77 88 D4 90}`

I may be able to pick what I have apart using the chip manufacturers SDK

danieledwardgeorgehitchcock, 2022-08-27:
So,

I feel that I am dangerously close to this, but I cannot quite figure out what I am doing wrong.

Based on the documentation, the install code is a device specific little endian code where the first 16 bits (32 digits) are the network key, and the last 2 bits (4 digits) are a CRC value.

The Silicone Labs router stores the device MAC address (labelled as EUI-64) in an ascending array from 0-254.

The MAC address is used as a lookup key for the device network key.

So, I have done the following:

Decoded my QR code which contains all of the information required:

`RB01SG0D831018264800400000000000000000009035EAFFFE424783DLKAE3B287281CF16F550733A0CEC38AA31E802`

Applying the logic above,

The last 4 digits are the CRC:
`E802`

The next 32 digits are the Network Key:
`AE3B287281CF16F550733A0CEC38AA31`

Then the next 4 digits are unknown (could this be of significance??):
`DLKA`

Then the next 16 digits are the device MAC address:
`9035EAFFFE424783`

After that, the next 34 digits are a 0 padded model identifier:
`0D83101826480040000000000000000000`

Then the remaining 6 digits are the manufacturer identifier:
`RB01SG`

So, all that I should need from the above to decrypt the traffic is the 32 digit network key (as the MAC address is only required for storage in the lookup table), and the remaining fields are a fingerprint to tell the controller what the device is that you are pairing.

I am guessing the byte order should be flipped too as it is little endian but, nothing I try seems to work!

Koenkk, 2022-08-27:
It seems the TI firmware has support for install codes. I need to check how it works, what coordinator are you using?

danieledwardgeorgehitchcock, 2022-08-27:
I have the Sonoff ZBDongle-P running Z-Stack 3

Koenkk, 2022-08-28:
Thanks, I will check it

Koenkk, 2022-08-29:
I've added the install code command; can you:

- Update to the latest z2m dev  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
- Create a folder in the z2m data directory called `extension` and put a file in it named `install.js` with the following contents:

```js
const constants = require('zigbee-herdsman/dist/adapter/z-stack/unpi/constants');
const {Buffer} = require('buffer');
console.log(Buffer);
const Subsystem = constants.Subsystem;

class MyExampleExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;
        this.publishEntityState = publishEntityState;
        this.eventBus = eventBus;
        this.settings = settings;
        this.logger = logger;
    }

    /**
     * This method is called by the controller once Zigbee2MQTT has been started.
     */
    async start() {
        console.log('Adding install code!');
        const payload = {
            installCodeFormat: 1,
            ieeeaddr: '0x9035EAFFFE424783',
            installCode: Buffer.from([0xAE, 0x3B, 0x28, 0x72, 0x81, 0xCF, 0x16, 0xF5, 0x50, 0x73, 0x3A, 0x0C, 0xEC, 0x38, 0xAA, 0x31, 0xE8, 0x02]),
        };
        const k = await this.zigbee.herdsman.adapter.znp.request(Subsystem.APP_CNF, 'bdbAddInstallCode', payload);
        console.log(k);
        console.log('Added install code');
    }
}

module.exports = MyExampleExtension;
```

- Start z2m and try to pair the device in the non legacy mode now

For reference, this is the docs of the install code command:

[image]

danieledwardgeorgehitchcock, 2022-08-30:
I already run the Dev branch in Docker. Tried updating it kust now via docker-compose but it said that it was up-to-date..?

I proceeded to add the extension and I got the following:

```
Zigbee2MQTT:debug 2022-08-30 04:55:57: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'
[Function: Buffer] {
  poolSize: 8192,
  from: [Function: from],
  of: [Function: of],
  alloc: [Function: alloc],
  allocUnsafe: [Function: allocUnsafe],
  isBuffer: [Function: isBuffer],
  allocUnsafeSlow: [Function: allocUnsafeSlow],
  compare: [Function: compare],
  isEncoding: [Function: isEncoding],
  concat: [Function: concat],
  byteLength: [Function: byteLength],
  [Symbol(kIsEncodingSymbol)]: [Function: isEncoding]
}
Adding install code!
Zigbee2MQTT:error 2022-08-30 04:55:57: Failed to call 'MyExampleExtension' 'start' (Error: Command 'bdbAddInstallCode' from subsystem '15' not found
    at Function.createRequest (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:45:19)
    at Znp.request (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:299:34)
    at MyExampleExtension.start (/app/dist/util/externally-loaded.js:27:58)
    at Controller.callExtensions (/app/lib/controller.ts:315:40)
    at Controller.addExtension (/app/lib/controller.ts:187:20)
    at ExternalExtension.loadExtension (/app/lib/extension/externalExtension.ts:91:20)
    at processTicksAndRejections (node:internal/process/task_queues:96:5))
Zigbee2MQTT:info  2022-08-30 04:55:57: Started frontend on port 0.0.0.0:8881
```

I am guessing that if you made changes to the Z2M codebase, they haven't been merged yet, or am I missing something?

I will remove my container and re-add just to make sure it isn't something on my side

danieledwardgeorgehitchcock, 2022-08-30:
So, the above was me being an idiot. I dropped the container and re-created it. The function loaded successfully and....

WE HAVE A SUCCESSFUL JOIN!!!!

```
[Function: Buffer] {
Zigbee2MQTT:debug 2022-08-30 05:04:02: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"8af3ad5","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":21,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_legacy_entity_attributes":false,"last_seen":"disable","legacy_api":false,"legacy_availability_payload":false,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":true,"log_syslog":{"app_name":"Zigbee2MQTT","eol":"/n","host":"localhost","localhost":"localhost","path":"/dev/log","pid":"process.pid","port":123,"protocol":"tcp4","type":"5424"},"output":"json","pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"blocklist":[],"device_options":{"legacy":false},"devices":{"0x680ae2fffe6faf49":{"friendly_name":"0x680ae2fffe6faf49"},"0xa4c1381538445fab":{"friendly_name":"0xa4c1381538445fab"}},"external_converters":["./external_converters/vimar_iot_thermostat.js","./external_converters/tuya_4ch_relay.js"],"frontend":{"host":"0.0.0.0","port":8881},"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://mosquitto:8883"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/ttyUSB0"}},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes with regex from published payload.","examples":["^temperature$","^battery$","^action$"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_cache":{"description":"Filter attributes with regex from being added to the cache, this prevents the attribute from being in the published payload when the value didn't change.","examples":["^input_actions$"],"items":{"type":"string"},"title":"Filtered attributes from cache","type":"array"},"filtered_optimistic":{"description":"Filter attributes with regex from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["^color_(mode|temp)$","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"homeassistant":{"properties":{"name":{"description":"Name of the device in Home Assistant","title":"Home Assistant name","type":"string"}},"title":"Home Assistant","type":["object","null"]},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string"},"optimistic":{"default":true,"description":"Publish optimistic state after set","title":"Optimistic","type":"boolean"},"qos":{"description":"QoS level for MQTT messages of this device","title":"QoS","type":"number"},"retain":{"description":"Retain MQTT messages of this device","title":"Retain","type":"boolean"},"retention":{"description":"Sets the MQTT Message Expiry in seconds, Make sure to set mqtt.version to 5","title":"Retention","type":"number"}},"required":["friendly_name"],"type":"object"},"group":{"properties":{"devices":{"items":{"type":"string"},"type":"array"},"filtered_attributes":{"items":{"type":"string"},"type":"array"},"friendly_name":{"type":"string"},"off_state":{"default":"auto","description":"Control when to publish state OFF for a group. 'all_members_off': only publish state OFF when all group memebers are in state OFF, 'last_member_state': publish state OFF whenever one of its members changes to OFF","enum":["all_members_off","last_member_state"],"requiresRestart":true,"title":"Group off state","type":["string"]},"optimistic":{"type":"boolean"},"qos":{"type":"number"},"retain":{"type":"boolean"}},"required":["friendly_name"],"type":"object"}},"properties":{"advanced":{"properties":{"adapter_concurrent":{"description":"Adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)","requiresRestart":true,"title":"Adapter concurrency","type":["number","null"]},"adapter_delay":{"description":"Adapter delay","requiresRestart":true,"title":"Adapter delay","type":["number","null"]},"cache_state":{"default":true,"description": [... cut ...]
[... 81 lines cut ...]
      },
      "binds": [
        {
          "cluster": 32,
          "type": "endpoint",
          "deviceIeeeeAddress": "0x00124b00257cae15",
          "endpointID": 1
        }
      ],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 2,
  "stackVersion": 7,
  "hwVersion": 32,
  "dateCode": "20211012",
  "swBuildId": "V1.19.00",
  "zclVersion": 3,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1661835961869,
  "defaultSendRequestWhen": "active"
}
```

safakaltun, 2022-08-30:
I also can confirm that I managed to successfully pair the device using hass.io. I think it is important to mention for the people who will do this after us that **ieeeaddr** and **installCode** are both device specific and case sensitive in the example extension @Koenkk  provided. Otherwise the device does not pair.

danieledwardgeorgehitchcock, 2022-08-30:
I am guessing the extension above is just a proof of concept to see if we could successfully join the devices to Z2M.

If I was to see this working in production, I'd probably create a new MQTT topic that accepts the IEEE MAC Adress, Install Code and Install Code Format that in turn adds an entry to the co-ordinator (and one to remove it too)

kirovilya, 2022-08-30:
@Koenkk I think I can implement for the EZSP adapter. just need to figure out where to put the key and how to use it :)
https://www.silabs.com/documents/public/application-notes/an1089-using-installation-codes-with-zigbee-devices.pdf

I have one device with this type of connection - Aqara Knob H1, although you can connect without an install code.

[image]

danieledwardgeorgehitchcock, 2022-08-30:
@safakaltun - do you want to have a go at a herdsman converter? I'm happy to have a go if not, just didn't want to duplicate effort...

safakaltun, 2022-08-30:
@danieledwardgeorgehitchcock Sure thing. I will have a look at it tonight and see how much I can do. Have you seen anything interesting related to the exports when you did the sniff with the Bosch Smart Controller?

safakaltun, 2022-08-30:
Hi again, I am little bit stuck, and can use some help. Below converter works fine to control the siren, set the volume and duration just fine. However, the exposed tamper doesn't work. It seems to be stuck in the "tampered" mode.  Any ideas what could be the problem?

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RBSH-OS-ZB-EU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'BSIR-EZ', // Vendor model number, look on the device for a model number
    vendor: 'Bosch', // Vendor of the device (only used for documentation and startup logging)
    description: 'Wired/Wireless Smart Outdoor Siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
	fromZigbee: [fz.battery, fz.ias_siren],
    toZigbee: [tz.warning],
    exposes: [e.warning(), e.battery(), e.battery_low(), e.tamper()],
    configure: async (device, coordinatorEndpoint, logger) => {
        device.defaultSendRequestWhen = 'immediate';
        device.save();
        await device.getEndpoint(1).unbind('genPollCtrl', coordinatorEndpoint);
	},
};

module.exports = definition;

```

danieledwardgeorgehitchcock, 2022-08-31:
I'm looking into the tamper alarm. When looking at the debug log, it looks as though the tamper is reported on the ssIasZone cluster with the following data `{"extendedstatus":0, "zonestatus":0}` - this message is both the same on a trigger and clear. Going to do some further sniffing to see if that yields anything further

danieledwardgeorgehitchcock, 2022-09-01:
Looking at the sniff events, theres nothing to report with the tamper.

It looks to be an 'action' type response from the device rather than a binary i.e. it reports a tamper event each time the value of the tamper button changes.

That being said, I noticed when setting up the device with the native hub that there is a tamper self-test option in the configuration screen so, I am going to look into that to see if I can see anything - difficulty with this though, is I can't decrypt the sniff

Koenkk, 2022-09-06:
Install codes are now supported (frontend `Settings -> Tools -> Add install code`)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

safakaltun, 2022-09-07:
Thanks @Koenkk! 

@danieledwardgeorgehitchcock have you managed to make any progress on the tamper button? I am in the middle of moving my place, so haven’t had a chance to look into this. I am wondering if the tamper integration is done in a similar way to a button or something like that.

danieledwardgeorgehitchcock, 2022-09-09:
@safakaltun - yeah thats basically what I discovered above when looking at the debug logs.

I know that the Bosch app has the ability to test the tamper feature of the device but, without being able to decrypt the device to controller logs when the device is connected to its native hub, I am just going through a very manual process of figuring out which endpoints I can read from and write to, to see if I can fire a request to read the status when the event is fired

Koenkk, 2022-09-22:
@safakaltun can you try with `RB01SG0D836591B3CC00100000000000000000000000000D6F00179F2BC9DLKD0F471C9BBA2C0208608E91EED17E2B1` ?

danieledwardgeorgehitchcock, 2022-09-22:
I don't think that code will work.

Have a look at how I split out the QR code above.

You should just need the last 36 digits (32 network code and 4 CRC)

safakaltun, 2022-09-22:
I tried adding padding in different places as well, but didn’t work unfortunately. When I try the above suggestion the error I get is: 

`09Request 'zigbee2mqtt/bridge/request/install_code/add' failed with error: 'SREQ '--> APP_CNF - bdbAddInstallCode - {"installCodeFormat":1,"ieeeaddr":"0x0000000D6F00179F","installCode":{"type":"Buffer","data":[157,0,208,244,113,201,187,162,192,32,134,8,233,30,237,23,226,177]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)')' `

The additional zeros are interpreted as a part of the ieee address.

danieledwardgeorgehitchcock, 2022-09-22:
I'd suggest you open up a new thread for this as it is a different device and you wouldn't want to muddy up this thread.

That being said, I have found an image of the label on the device and it looks as though it might be a type 2 (16 bytes for key derived from Install Code) - see ZStack Command Reference page above.

The confusing thing is that there is a 'Device Local Key' which appears to be 26 characters long so, I don't know how you would derive a link key from 26 non-hex characters...

[image]

Koenkk, 2022-09-23:
@safakaltun I've added support for the 16 byte install code now, please check again in the latest dev (with the original key, not the one I added zeros to).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

danieledwardgeorgehitchcock, 2022-10-24:
Would it be worth pushing what you have, then adding to it when you / I have worked out the tamper logic?

safakaltun, 2022-10-24:
I am having a look at it now and I noticed that when I removed the siren to test, its zoneStatus went from 33956 to 32932. This corresponds to bit 10 going to 0 from 1. However, when I put the device back on the wall mount, zoneStatus did not go back to 33956. I am waiting to see if there is an internal timer that needs to be reset for the zoneStatus to go back to its previous value. My other theory is that the device might be waiting to be triggered for bit 10 to go back to its initial value - I will test this tomorrow.

I am not sure what bit 10 corresponds to as it is not explained in the [Zigbee Cluster Library Specification](https://raw.githubusercontent.com/Koenkk/zigbee-herdsman/master/docs/07-5123-08-Zigbee-Cluster-Library.pdf). I am also not sure why Bosch did not use Bit 2 for tamper.

[image]

@danieledwardgeorgehitchcock it might be very helpful if you could also check the zoneStatus before and after removing the device from the wall mount if you have it already installed. If bit 10 changes for you as well, that can at least tell us.

danieledwardgeorgehitchcock, 2022-10-28:
@safakaltun - I experience similar behaviour. I haven't found a timeout or a reset command that sets it back to what is required yet though...

safakaltun, 2022-12-01:
Good news! I have finally managed to get the transport key from the bosch smart home controller through other bosch devices (https://github.com/Koenkk/zigbee2mqtt/issues/14355). I will sniff the traffic over the weekend and try to understand how tamper is implemented.

danieledwardgeorgehitchcock, 2022-12-01:
Any chance you could share it so I can help you out?

safakaltun, 2022-12-01:
As far as I know, the transport key is initially randomly generated by the Bosch Smart Controller (BSC). So that will be device specific. However, I can try describing what I did to get the transport key. 

1- I updated my Wireshark to the most recent version which is 4.0.1. 
2- After that, I needed to find the zigbee channel BSC use. I noticed that after connecting the first device, it tries to find the best channel and switches to that channel. 
3- Add only the installation key as the pre-configured key. The install key should also include the the 2 bit CRC. So, in this case, it should be 18 bits long.
4 - Initiate pairing

For me, these steps revealed the transport key that is used by the BSC.

danieledwardgeorgehitchcock, 2022-12-01:
OK thanks! I'll give it a go.

I remember trying something similar on my siren but didn't get anywhere. Will give it a go with the exact method you described above

safakaltun, 2022-12-02:
I tried the steps above to see if I could get the Transport Key using the install code of the siren, and it did not work. So it might not work for you as well. However, since I already have the Transport Key thanks to the other devices, I did a fairly extensive sniffing while testing the device through the app. 

Here is the sniff from the siren. I am not sure if the encryption keys are embedded into the sniff file or not. If not, so here is the transport key: 79:d3:38:71:27:0f:51:0b:01:20:89:f6:bc:80:eb:3c.

[siren_pairing_and_settings_log.zip](https://github.com/Koenkk/zigbee2mqtt/files/10141077/siren_pairing_and_settings_log.zip)

Below you can find my findings from the sniff. With respect to tamper, something funny might be happening. 

Bindings 
| Source Endpoint | Destination endpoint| Cluster                                      | 
| 1               | 2                   | Cluster: 0x0502 (Intruder Alarm System WD)   |
| 1               | 2                   | Cluster: 0x0500 (Intruder Alarm System Zone) |
| 1               | 2                   | Cluster: 0x0019 (OTA Upgrade)                |
| 1               | 2                   | Cluster: 0x0020 (Poll Control)               |
| 1               | 2                   | Cluster: 0x0001 (Power Configuration)        |

Reporting Configuration
____________________________________________________________________________________________________________________________________________________________
Source Endpoint | Destination Endpoint | Cluster                                      | Attribute                                        | MinRep | MaxRep |
| 2             | 1                    | Cluster: Power Configuration (0x0001)        | Attribute: Unknown (0x003e)                      | 0      | 65534  |
| 2             | 1                    | Cluster: Power Configuration (0x0001)        | Attribute: Battery Percentage Remaining (0x0021) | 0      | 65534  |
| 2             | 1                    | Cluster: Power Configuration (0x0001)        | Attribute: 0xa002                                | 0      | 65534  |
| 2             | 1                    | Cluster: Power Configuration (0x0001)        | Attribute: 0xa000                                | 0      | 65534  |
| 2             | 1                    | Cluster: Intruder Alarm System Zone (0x0500) | Attribute: 0xa001                                | 0      | 65534  |
| 2             | 1                    | Cluster: Power Configuration (0x0001)        | Attribute: 0xa001                                | 0      | 65534  |
____________________________________________________________________________________________________________________________________________________________

----------- IAS ZONE/ IAS ZONE WD Findings ------------

It enrolls to IAS Zone at no 289.
All read/report attributes are done via Cluster: Intruder Alarm System Zone (0x0500)

Write Attributes are done via Cluster: Intruder Alarm System WD (0x0502)
____________________________________________________________________________________________________________________
| Action                                              | Attribute         | Value range        | Wireshark Log No. |
| Change default flashing light alarm  duration (min) | 0xa005            |                    | 1120              |
| Change default siren alarm duration (min)           | 0xa000            |                    | 1180              |
| Siren volume (-)                                    | 0xa002            | 1-low 2-med 3-high | 1225              |
| Siren alarm delay (s)                               | 0xa003            |                    | 1350              |
| Flashing light alarm delay (s)                      | 0xa004            |                    | 1384              |

Report Attributes (Cluster: Intruder Alarm System Zone (0x0500))
___________________________________________________________________________________________________________________
| Action                                                                                     | Attribute         | Value range        | Wireshark Log No. |
| ZCL IAS Zone: Zone Status Change Notification due to tamper on ->> off                     | ZoneStatus        | ------             | 1708, 1716, 1718 (First it sends the current value with tampered=1, second receives an unknown command from the hub, then sends the updated value tampered=0)
| ZCL IAS Zone: Zone Status Change Notification due to tamper off ->> on                     | ZoneStatus        | ------             | 1851, 1859 (Sends the updated value with tamper on (tampered=1), then receives an unknown command from the hub)
| ZCL IAS Zone: Zone Status Change Notification when alarm triggered via wired channel       | ZoneStatus        | ------             | 2420, 2428 (Sends the updated value, then receives an unknown command from the hub)
| ZCL IAS Zone: Zone Status Change Notification when alarm deactivated via wired channel     | ZoneStatus        | ------             | 2555, 2563 (Sends the updated value, then receives an unknown command from the hub)

I am not sure what does the Unknown Command (0xf3) do, but it probably plays a role. 

----------- Power Config Findings -------------

Cluster: Power Configuration (0x0001) is used for read/write/report

Write Attributes
_______________________________________________________________________________________
[... cut ...]
| Power supply selection | 0xa002            | 1 (230v       )    | 1508              |
| Power supply selection | 0xa002            | 0 (solar power)    | 1611              |

Report Attributes
_______________________________________________________________________________________
| Action                    | Attribute         | Value range        | Wireshark Log No. |
| Power supply selection    | 0xa002            | 0-2                | 1444              |
| Battery percentage (0.5%) | 0x0021            | 0-200              | 385               |

danieledwardgeorgehitchcock, 2022-12-02:
This is all great stuff! I'll look into it over the weekend. Out of curiosity, what device allowed you to find the backdoor for the transport key?

safakaltun, 2022-12-02:
Thanks! It was the Twinguards.

danieledwardgeorgehitchcock, 2022-12-04:
I tried with your transport key and it looks like each bosch hub has its own which is unfortunate. I have ordered a used twinguard so I can do my own sniffing but I'm also reviewing the data that you've captured in the above pcap file. Hopefully between us, we can crack it!

safakaltun, 2022-12-04:
I think the BSC generates a random key and I remember reading that it also updates that key periodically (don't remember where, I read a lot of stuff about these devices recently :D).

I am sure we will manage! combination of the outdoor siren and twinguard is also nice actually. One can trigger the other. Anyways, I have had a look at what I captured and compared it a new capture between z2m and the siren. I think I know what the problem is, but I have no clue how to fix that through z2m. 

So whenever the status changes from "tampered" to "not tampered", the device first sends out a change notification with the existing zoneStatus. Then it seems like it expects a command from the hub to send an updated zonestatus. After receiving the command, it sends a new change notification with tamper status 0. These are the 3 selected entries in the below snippet. 

[image]

On the other hand, when the tamper status changes from "not tampered" to "tampered", the siren sends the change notification with an updated value (tamper: 1). It still receives a message from the hub, but this time it does not update the zonestatus since it is already updated. You can see it below as well:

[image]

In both cases, the command sent from the hub is the same: 

[image]

So I think if we can solve this, we can solve the problem with the tamper.

danieledwardgeorgehitchcock, 2022-12-04:
Hmm. Interesting! This sounds like something I have done before with a yale lock - I will see if I can dig the code out and modify it! Good spot!

danieledwardgeorgehitchcock, 2022-12-05:
@Koenkk - any ideas how we can send the above frame?

I have tried with the below in a local toZigbee converter but I  get an `Unsupported Attribute` error...

` await entity.write('ssIasZone', {0xf3: {value: 2, type: 0x20}}, {manufacturerCode: 0x1209})`

safakaltun, 2022-12-05:
A learning from Twinguards is that the attribute names are shown as decimals in the payload instead of hexadecimals for some reason. So, perhaps you can try

`await entity.write('ssIasZone', {243: {value: 2, type: 0x20}}, {manufacturerCode: 0x1209})`

It might be a long shot, but it might work.

danieledwardgeorgehitchcock, 2022-12-05:
I can see that Z2M is converting the hex values correctly, but good thought! I just think my payload is probably incorrect but, I have tried finding a converter that may do something similar in Z2M and have come up empty - the above is an adaption of an Aqara cover payload..

[21 more replies]

closed 2022-12-08 by Koenkk