Koenkk/zigbee2mqtt #13523: SONOFF Zigbee 3.0 USB Dongle Plus V2 - failed to connect SRSP - SYS - ping after 6000ms

clarsen, 2022-08-13:
### What happened?

Getting `Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)`

I have *not* tried flashing new coordinator or source router firmware since the device was newly shipped from iTEAD and they advertised coordinator firmware installed.

### What did you expect to happen?

Expected successful startup

### How to reproduce it (minimal and precise)

Running on Raspberry Pi 4, Ubuntu 22.04.1
$ sudo apt-get purge modemmanager

reboot

start up using 

```
docker run \
  -d \
  --name zigbee2mqtt \
  --restart unless-stopped \
  -p 8080:8080 \
  -e TZ=America/Los_Angeles \
  -e 'DEBUG=zigbee-herdsman*' \
  --device=/dev/ttyACM0 \
  -v $(pwd)/zigbee2mqtt-data:/app/data \
  koenkk/zigbee2mqtt:latest-dev
```
and configuration.yaml:
```
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.5.11

serial:
  port: /dev/ttyACM0
frontend:
  port: 8080
advanced:
  network_key: GENERATE
```

device is /dev/ttyACM0
```
$ la -l /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220707151442-if00
lrwxrwxrwx 1 root root 13 Aug 13 07:44 /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20220707151442-if00 -> ../../ttyACM0
```

### Zigbee2MQTT version

latest-dev (docker image)

### Adapter firmware version

does not start far enough to get this info

### Adapter

QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9331048/log.txt)

clarsen, 2022-08-13:
FYI,
after changing configuration.yaml to:

```
serial:
  port: /dev/ttyACM0
  adapter: ezsp
```

the dongle was recognized.

Turns out the dongle is coordinator type EZSP v8, running coordinator version 6.10.3.0 build 297

Koenkk, 2022-08-13:
`adapter: ezsp` is indeed required, this is now documented on https://www.zigbee2mqtt.io/guide/adapters/#based-on-silicon-labs-efr32mg2x-mgm21x-and-efr32mg1x-mgm1x-series

closed 2022-08-13 by Koenkk

xenblog, 2023-03-26:
Running Version: 1.18.1-1
When adding "adapter:ezsp" to configuration.yaml i get the below error when saving the config:

Failed to save add-on configuration, does not match regular expression **^zstack|deconz|zigate$**. Got {'data_path': '/config/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 'zigbee2mqtt'}, 'serial': {'port': '/dev/ttyACM0', 'adapter': 'ezsp'}, 'advanced': {'log_level': 'warn', 'pan_id': 6754, 'channel': 11, 'network_key': [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13], 'availability_blocklist': [], 'availability_passlist': []}, 'device_options': {}, 'blocklist': [], 'passli...