Koenkk/zigbee2mqtt #13525: Wiser 40/300-Series Module Dimmer - Cannot control via Home Assistant or Zigbee2Mqtt Exposes tab

linkedupbits, 2022-08-13:
### What happened?

Who clicking/changing the "state" on the device "exposes" tab, the switch does not toggle on or off.

The brightness slider respects the current brightness when controlled via the physical button, but cannot be controlled via the UI.

Ballast maximum and minimum level sliders returns an error: 
`Publish 'set' 'ballast_minimum_level' to '0x9035eafffea70623' failed: 'Error: Write 0x9035eafffea70623/21 lightingBallastCfg({"minLevel":52}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

However, I can use the dev console to set these.
**Toggle on/off** endpoint 3, cluster 6, command 2, payload {}
`Invoked '6.2' with payload '{}'`
**Set min level:** endpoint 3, cluster lightingBallastCfg, attribute minLevel value 0=254
`Wrote '{"maxLevel":145,"minLevel":34}' to 'lightingBallastCfg'`
**identify the light** endpoint:3, cluster identify, attribute identifyTime, value 1 
`Wrote '{"identifyTime":1}' to 'genIdentify'`

Endpoint 3
Cluster: 8 
Command: 1
{"startupCurrentLevel":0 ,"movemode":1 ,"rate":40,"transtime":20 }

I am new to typescript, but if I drop a "replacement" definition for the dimmer, will it replace the version from source?
I am happy to make pull requests etc if I can get it working.

There are some other functions it looks like it can support as well, eg:
Dim:
Endpoint: 3, Cluster: 8, Command: 1
`{"startupCurrentLevel":0 ,"movemode":1 ,"rate":40,"transtime":20 }`

### What did you expect to happen?

The user interface in zigbee2mqtt would be able to control the dimmer switch. 

### How to reproduce it (minimal and precise)

Select the dimmer device
Click on the Exposes tab
change the on/off state
** nothing is written to the log **

drag the brightness slider
** nothing is written to the log **

drag the ballast_minimum_level slider
Error:
`Publish 'set' 'ballast_minimum_level' to '0x9035eafffea70623' failed: 'Error: Write 0x9035eafffea70623/21 lightingBallastCfg({"minLevel":31}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

drag the ballast_maximum_level slider
error`Publish 'set' 'ballast_maximum_level' to '0x9035eafffea70623' failed: 'Error: Write 0x9035eafffea70623/21 lightingBallastCfg({"maxLevel":206}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

smartlight CC2652P USB (with Extension cable)

closed 2022-09-20 by github-actions[bot]

thyjukki, 2022-11-15:
Did you ever get this working? I got WDE002388 model and have same problem.

mstjerna, 2023-01-23:
Same issue here with the WDE002388 model.

daandersson, 2023-01-28:
I have WDE002388 as well. It cannot be controlled from UI. Zigbee2Mqtt detects is as the older WDE002334 but there must be some differences. This issue seems closed so I guess it will not be looked at. It is possible to re-open or do we need to create new?

henrikbol, 2023-02-04:
Got the same error on Schneider Electric CCT5010-0001

Zigbee2MQTT ver 1.30.0-1

First it connect and interview correctly:

`info  2023-02-04 22:18:31: Configuring '0xb43522fffe31a2e3'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/light/0xb43522fffe31a2e3/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xb43522fffe31a2e3/set","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"name":"0xb43522fffe31a2e3","schema":"json","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_light_zigbee2mqtt"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/sensor/0xb43522fffe31a2e3/ballast_minimum_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"enabled_by_default":false,"name":"0xb43522fffe31a2e3 ballast minimum level","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_ballast_minimum_level_zigbee2mqtt","value_template":"{{ value_json.ballast_minimum_level }}"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/number/0xb43522fffe31a2e3/ballast_minimum_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xb43522fffe31a2e3/set/ballast_minimum_level","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"max":254,"min":1,"name":"0xb43522fffe31a2e3 ballast minimum level","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_ballast_minimum_level_zigbee2mqtt","value_template":"{{ value_json.ballast_minimum_level }}"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/sensor/0xb43522fffe31a2e3/ballast_maximum_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"enabled_by_default":false,"name":"0xb43522fffe31a2e3 ballast maximum level","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_ballast_maximum_level_zigbee2mqtt","value_template":"{{ value_json.ballast_maximum_level }}"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/number/0xb43522fffe31a2e3/ballast_maximum_level/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xb43522fffe31a2e3/set/ballast_maximum_level","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"max":254,"min":1,"name":"0xb43522fffe31a2e3 ballast maximum level","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_ballast_maximum_level_zigbee2mqtt","value_template":"{{ value_json.ballast_maximum_level }}"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/sensor/0xb43522fffe31a2e3/dimmer_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"manufacturer":"Schneider Electric","model":"Micro module dimmer (CCT5010-0001)","name":"0xb43522fffe31a2e3","sw_version":"002.012.001 R"},"enabled_by_default":false,"name":"0xb43522fffe31a2e3 dimmer mode","state_topic":"zigbee2mqtt/0xb43522fffe31a2e3","unique_id":"0xb43522fffe31a2e3_dimmer_mode_zigbee2mqtt","value_template":"{{ value_json.dimmer_mode }}"}'
info  2023-02-04 22:18:31: MQTT publish: topic 'homeassistant/select/0xb43522fffe31a2e3/dimmer_mode/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xb43522fffe31a2e3/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","command_topic":"zigbee2mqtt/0xb43522fffe31a2e3/set/dimmer_mode","device":{"identifiers":["zigbee2mqtt_0xb43522fffe31a2e3"],"m [... cut ...]
`error 2023-02-04 22:20:25: Publish 'set' 'brightness' to '0xb43522fffe31a2e3' failed: 'Error: Command 0xb43522fffe31a2e3/242 genLevelCtrl.moveToLevelWithOnOff({"level":162,"transtime":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 60119 - 242 - 28 - 8 - 11 after 10000ms)'
`

Any insights or update on the issue?

mstjerna, 2023-02-06:
> I have WDE002388 as well. It cannot be controlled from UI. Zigbee2Mqtt detects is as the older WDE002334 but there must be some differences. This issue seems closed so I guess it will not be looked at. It is possible to re-open or do we need to create new?

Hi

I just updated to release 1.30.0 and it works perfect! 
Dimming, on/off setting timers etc.