Koenkk/zigbee2mqtt #13528: cover HK-ZCC-A sunchiner SR-ZG9080A open close state invert

tomamplius, 2022-08-13:
### What happened?

I can invert position, but open close state is not invert

I can create pull request but the is 2 ways

First is update function 
```
   cover_state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'upOpen', 'close': 'downClose', 'stop': 'stop', 'on': 'upOpen', 'off': 'downClose'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    },
```

to 
```
   cover_state: {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = utils.getMetaValue(entity, meta.mapped, 'coverInverted', 'allEqual', false) ?
                {'open': 'upOpen', 'close': 'downClose', 'stop': 'stop', 'on': 'upOpen', 'off': 'downClose'} :
                {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    },
```

second is create function like bticino_4027C_cover_state

### What did you expect to happen?

Allow invert state on cover_state to_zigbee function

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

1d03

### Adapter

zigate

Koenkk, 2022-08-13:
Changing the standard cover state converter makes the most sense to me.

closed 2022-08-14 by tomamplius

tomamplius, 2022-10-10:
reverse cable between SR-ZG9080A and motor

Hankey71, 2022-10-11:
OK... too bad !