Koenkk/zigbee2mqtt #13531: Livolo switch status change issues

sanychonline, 2022-08-13:
### What happened?

After changing switch status it incorrectly shows in Home app as not change, or does not show real status.
Example: trying to switch off light it shows switch off, then on, then off again. Sometimes it shows status as on, but switch is off. This issue does not exist in version 1.24.0 (still using it, as it works properly)

### What did you expect to happen?

Example: trying to switch off light it shows switch off, then on, then off again. Sometimes it shows status as on, but switch is off. This issue does not exist in version 1.24.0 (still using it, as it works properly)

### How to reproduce it (minimal and precise)

Example: trying to switch off light it shows switch off, then on, then off again. Sometimes it shows status as on, but switch is off. This issue does not exist in version 1.24.0 (still using it, as it works properly)

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20190608

### Adapter

CC2531 ZigBee - Texas Instruments, Inc.

### Debug log

16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/BathroomAccesoriesLights', payload '{"linkquality":49,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/LivingRoomLED', payload '{"linkquality":5,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF"}'
stdout
16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/BathroomLights', payload '{"linkquality":55,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/LivingRoomLights', payload '{"linkquality":7,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":0,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:01
Zigbee2MQTT:info  2022-08-13 13:29:01: MQTT publish: topic 'zigbee2mqtt/ShowerLights', payload '{"linkquality":26,"state_bottom_left_bottom_left":"OFF","state_bottom_right_bottom_right":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:02
Zigbee2MQTT:info  2022-08-13 13:29:02: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":0,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:29:12
Zigbee2MQTT:info  2022-08-13 13:29:12: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":10,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF"}'
stdout
16:29:12
Zigbee2MQTT:info  2022-08-13 13:29:12: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":13,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF"}'
stdout
16:29:12
Zigbee2MQTT:info  2022-08-13 13:29:12: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":13,"state_left":"OFF","state_left_left":"ON","state_right":"OFF"}'
stdout
16:29:12
Zigbee2MQTT:info  2022-08-13 13:29:12: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":15,"state_left":"ON","state_left_left":"ON","state_right":"OFF"}'
stdout
16:29:18
Zigbee2MQTT:info  2022-08-13 13:29:18: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":10,"state_left":"ON","state_left_left":"ON","state_right":"OFF"}'
stdout
16:29:18
Zigbee2MQTT:info  2022-08-13 13:29:18: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":15,"state_left":"OFF","state_left_left":"ON","state_right":"OFF"}'
stdout
16:29:18
Zigbee2MQTT:info  2022-08-13 13:29:18: MQTT publish: topic 'zigbee2mqtt/Wardrobe', payload '{"linkquality":15,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF"}'
stdout
16:29:23
[... 96 lines cut ...]
Zigbee2MQTT:info  2022-08-13 13:33:50: MQTT publish: topic 'zigbee2mqtt/LivingRoomLights', payload '{"linkquality":10,"state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:33:50
Zigbee2MQTT:info  2022-08-13 13:33:50: MQTT publish: topic 'zigbee2mqtt/BathroomLights', payload '{"linkquality":55,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:33:50
Zigbee2MQTT:info  2022-08-13 13:33:50: MQTT publish: topic 'zigbee2mqtt/ShowerLights', payload '{"linkquality":26,"state_bottom_left_bottom_left":"OFF","state_bottom_right_bottom_right":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:33:50
Zigbee2MQTT:info  2022-08-13 13:33:50: MQTT publish: topic 'zigbee2mqtt/KitchenLights', payload '{"linkquality":0,"state_bottom_left_bottom_left":"OFF","state_left":"OFF","state_left_left":"OFF","state_right":"OFF","state_right_right":"OFF"}'
stdout
16:34:21
Zigbee2MQTT:info  2022-08-13 13:34:21: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
stdout
16:34:21
Zigbee2MQTT:info  2022-08-13 13:34:21: Disconnecting from MQTT server
stdout
16:34:21
Zigbee2MQTT:info  2022-08-13 13:34:21: Stopping zigbee-herdsman...
stdout
16:34:22
Zigbee2MQTT:info  2022-08-13 13:34:22: Stopped zigbee-herdsman
stdout
16:34:22
Zigbee2MQTT:info  2022-08-13 13:34:22: Stopped Zigbee2MQTT

ashway83, 2022-08-22:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4550

Koenkk, 2022-08-22:
Fixed by @ashway83 in https://github.com/Koenkk/zigbee-herdsman-converters/pull/4550

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-22 by Koenkk