Koenkk/zigbee2mqtt #13532: JTYJ-GD-01LM/BW doesn't expose "selftest" action to HASS

szymucha94, 2022-08-13:
### What happened?

I wanted to automate calling monthly "self-test" function on both my smoke detectors. Unfortunately "test" action is only available in "exposes" tab of device in z2m, not in the HASS device card.

### What did you expect to happen?

I expect "selftest" button to be available in HASS, either as a switch or as a service.

### How to reproduce it (minimal and precise)

n/a

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8 (Xiaomi Gateway 3 via ser2net)

### Debug log

n/a

szymucha94, 2022-08-13:
Just found out that calling "mqtt: publish" service with applicable payload does the job. Still I'd like to request making this a button in HASS.

Koenkk, 2022-08-14:
Added this to the discovery now, you can call it with e.g.

```yaml
service: select.select_option
data:
  option: ''
target:
  entity_id: select.my_socket_selftest

```

closed 2022-08-14 by Koenkk