Koenkk/zigbee2mqtt #13533: Service Fails to start going 1.22.2 -> 1.27.0

TeslaCLT, 2022-08-13:
### What happened?

Upgrading from 1.22.2 to 1.27.0 via [this doc](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version) led to `zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE`.

Restored backup and this time upgraded Node to 16 first via `sudo curl -fsSL https://deb.nodesource.com/setup_16.x | sudo -E bash - sudo apt-get install -y nodejs git make g++ gcc`. This failed to start 1.22.2.

Then performed `git pull` and `npm ci` , but failed with the same error message. After snooping around, saw a [2019 thread](https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-524729152) were @Koenkk recommended `rm -rf node_modules && npm install`. After running this, zigbee2mqtt started successfully. 

In between those steps, I restored my backups maybe 20 times, trying different variations. I hope this helps someone in the future so they don't spend as much time trying to figure out what happened.

### What did you expect to happen?

DIdn't expect needing to delete node_modules and reinstall, effectively a clean rebuild.

### How to reproduce it (minimal and precise)

From NodeJS 12.2.2, upgrade to 16.x. From there, all instances of zigbee2mqtt fail until node_modules deleted and npm install redone.

### Zigbee2MQTT version

1.22.2 -> 1.27.0

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee3.0 USB Dongle

Koenkk, 2022-08-14:
Do you still have logs of the issue?

TeslaCLT, 2022-08-16:
I do not, but I can restore the backup and recreate the problem if you wish. I will tell you that when it would fail to initialize above, it would NOT write a new entry to /opt/zigbee2mqtt/data/log , so I would need a pointing on which logs to gather.

Koenkk, 2022-08-16:
The output of executing `npm start` should be enough

closed 2022-09-24 by github-actions[bot]