Koenkk/zigbee2mqtt #13534: [New device support]:  Woolley Smart Plug SA-029

Sineos, 2022-08-13:
### Link

https://de.aliexpress.com/item/1005004505697978.html

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0x00124b00257e1789","nwkAddr":61101,"manufId":4742,"manufName":"SONOFF","powerSource":"Mains (single phase)","modelId":"Z111PL0H-1JX","epList":[1,13,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,64599,64672],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"Z111PL0H-1JX","manufacturerName":"SONOFF","powerSource":1,"zclVersion":3,"hwVersion":1,"dateCode":"20211223","swBuildId":"1.0.1"}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237dfd02","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":1,"dateCode":"20211223","swBuildId":"1.0.1","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1660422249267,"defaultSendRequestWhen":"immediate"}

### Comments

16A version

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Z111PL0H-1JX'],
    model: 'SA-029',
    vendor: 'Woolley',
    description: 'Smart Plug',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2022-08-14:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-14 by Koenkk

TungstenE2, 2022-11-29:
did anybody manage to toggle the LED as well? Mine stays on all the time.

treemonkey83, 2023-12-25:
Anybody still using these plugs? My plugs are constantly publishing

"Received Zigbee message from '#device name#', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0".

These messages are flooding the coordinator and causing instability with the plugs dropping off the network or even causing the bridge to go off line. 

Anybody have a solution? I'm assuming this is software related rather than firmware.