Koenkk/zigbee2mqtt #13536: WL-SD001-9W doesn't work with scenes

Interloping, 2022-08-14:
### What happened?

Light stays on when recalling a scene that defines the light state as off. (More broadly, light does not change power state as defined by scenes, nor colour. Brightness seems to maybe work.)

### What did you expect to happen?

I expected the light to turn off.

### How to reproduce it (minimal and precise)

Turn the light off. Store a scene for the group the light is part of
`mosquitto_pub -h HOST -t 'zigbee2mqtt/driveway_lights/set' -m '{"scene_store": 1}'
`
Turn the light on.

Try to recall the scene.
`mosquitto_pub -h HOST -t 'zigbee2mqtt/driveway_lights/set' -m '{"scene_recall": 1}'

Observe that the light is still on. (Also reported as on still is zigbee2mqtt)

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

0x26780700

### Adapter

Deconz Conbee II

### Debug log

Zigbee2MQTT:debug 2022-08-14 13:01:46: Received MQTT message on 'zigbee2mqtt/driveway_lights/set' with data '{"scene_recall": 2}'
Zigbee2MQTT:debug 2022-08-14 13:01:46: Publishing 'set' 'scene_recall' to 'driveway_lights'
Zigbee2MQTT:debug 2022-08-14 13:01:46: Received Zigbee message from 'Coordinator', type 'commandRecall', cluster 'genScenes', data '{"groupid":12,"sceneid":2}' from endpoint 1 with groupID 12, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-08-14 13:01:46: Requested color_temp 577 is higher than maximum supported 370, using maximum!
Zigbee2MQTT:info  2022-08-14 13:01:46: Successfully recalled group scene
Zigbee2MQTT:info  2022-08-14 13:01:46: MQTT publish: topic 'zigbee2mqtt/front_entrance', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"state":"OFF"}'
Zigbee2MQTT:info  2022-08-14 13:01:46: MQTT publish: topic 'zigbee2mqtt/Driveway spot 2', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"linkquality":108,"state":"OFF"}'
Zigbee2MQTT:info  2022-08-14 13:01:46: MQTT publish: topic 'zigbee2mqtt/driveway_lights', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"state":"OFF"}'
...
Zigbee2MQTT:debug 2022-08-14 13:03:03: Received MQTT message on 'zigbee2mqtt/driveway_lights/get' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2022-08-14 13:03:03: Publishing get 'get' 'state' to 'driveway_lights'
Zigbee2MQTT:debug 2022-08-14 13:03:03: Received Zigbee message from 'Coordinator', type 'read', cluster 'genOnOff', data '["onOff"]' from endpoint 1 with groupID 12, ignoring since it is from coordinator
Zigbee2MQTT:debug 2022-08-14 13:03:03: Received Zigbee message from 'Driveway spot 2', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-08-14 13:03:03: MQTT publish: topic 'zigbee2mqtt/front_entrance', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2022-08-14 13:03:03: MQTT publish: topic 'zigbee2mqtt/Driveway spot 2', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2022-08-14 13:03:03: MQTT publish: topic 'zigbee2mqtt/driveway_lights', payload '{"brightness":61,"color":{"x":0.393,"y":0.24},"color_mode":"xy","color_temp":370,"state":"ON"}'

ahd71, 2022-08-29:
I had a similar problem with Gledopto lights. After contacting the Vendor I got new firmware which silver the problem.

Tried to convince them to offer automatic OTA firmwarw updates via z2m but apperently did not succeeded. Did it manually for my devices and problem solved.

Interloping, 2022-08-31:
> I had a similar problem with Gledopto lights. After contacting the Vendor I got new firmware which silver the problem.
> 
> Tried to convince them to offer automatic OTA firmwarw updates via z2m but apperently did not succeeded. Did it manually for my devices and problem solved.

Good to know, thank you. I'm looking into getting new firmware but it isn't looking very promising. Can't find any sign of a website for the product; have tried reaching out to the seller on Amazon.

Interloping, 2022-10-01:
Have found that setting a transition setting of 0.1 for these bulbs allows scenes to work probably 90% of the time. Longer transition settings > 1 seem to lock up the bulb.