Koenkk/zigbee2mqtt #13539: Devices turn to Unavailable in HA in 1.27.0-1, even though they work fine in Z2M

Creat, 2022-08-14:
### What happened?

Running Home Assistant OS (2022.8.3 currently), and having Z2M running as an addon with Mosquito as the broker, I never had any issues until I upgraded to 1.27.0-1. I was coming from 1.26.0-1, and this was around the time HA 2022.8 was released, so I went through .0 to .3 trying to figure out the problem.

After a while (many hours or days), Z2M all Zigbee devices in HA turn to "Unavailable". The devices show up and work perfectly fine in Z2M (I can toggle lights, it shows LQ, dashboard is filled, ...). Restarting HA fixes the issue, restarting Z2M seems to as well, but not alyway I think? The availability-feature in Z2M used to be active, but isn't anymore (turned it off cause I thought it might be the cause).

Downgrading Z2M to 1.26.0-1 fixes the issue (had it running at this version for 5+ days with no issues), and upgrading it again to 1.27.0-1 causes it to reappear.

### How to reproduce it (minimal and precise)

Just upgrading to 1.27.0-1 causes this issue 100% of the time (after like a day or two)

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20210120

### Adapter

Slaesh's CC2652RB stick (USB-A)

Koenkk, 2022-08-14:
Probably a duplicate of #13382, please check if it is fixed in the dev branch.(https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Creat, 2022-08-14:
Well apparently I suck at searching, cause I did (try to) find similar existing issues and didn't find that one. Sarry about that.

Yes, seems to be a duplicate, with the only exception being that it happens after much longer periods for me. I assume it's related to the 'availability' feature and mine just happens to fail at the long interval, and for most people it fails at the short interval? In any case, following the other issue instead, since it is just a duplicate.

closed 2022-08-14 by Creat