Koenkk/zigbee2mqtt #13541: Wired issue with the Aqara presence detector FP1

wenjiavv, 2022-08-14:
### What happened?

I am trying to integrate the Aqara presence detector FP1 to the Xiaomi Mijia Smart
Multi-Mode Gateway ( which is in Zigbee2mqtt mode and works perfectly with other sensors).

Once I pair the FP1 to the gateway, FP1 just leave the Zigbee network automatically.

### What did you expect to happen?

Join the device to z2m.

### How to reproduce it (minimal and precise)

Once I pair the FP1 to the gateway, FP1 just leave the Zigbee network automatically.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8  (Xiaomi Mijia Smart Multi-Mode Gateway in Z2M mode)

### Debug log

Starting interview of '0x54ef441000dddddd'
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef441000dddddd","ieee_address":"0x54ef441000dddddd","status":"started"},"type":"device_interview"}'
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef441000dddddd"},"type":"pairing"}'
MQTT publish: topic 'zigbee2mqtt/0x54ef441000dddddd', payload '{"last_seen":"2022-08-14T16:58:53+08:00"}'
Warning: Device '0x54ef441000dddddd' left the network

Yuke-hd, 2022-08-15:
additional logs:
```
Failed to configure '0x54ef44100053d61e', attempt 2 (Error: Read 0x54ef44100053d61e/1 aqaraOpple([324], 
{
  "sendWhen": "immediate",
  "timeout": 10000,
  "disableResponse": false,
  "disableRecovery": false,
  "disableDefaultResponse": true,
  "direction": 0,
  "srcEndpoint": null,
  "reservedBits": 0,
  "manufacturerCode": 4447,
  "transactionSequenceNumber": null,
  "writeUndiv": false
})
failed (Timeout - 7543 - 1 - 41 - 64704 - 1 after 10000ms) 
at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) 
at listOnTimeout (node:internal/timers:559:17) 
at processTimers (node:internal/timers:502:7))
```

closed 2022-09-22 by github-actions[bot]

Foristkirito, 2023-05-10:
same issue, any solution

wenjiavv, 2023-05-10:
用了新的Zigbee网关解决了，估计是芯片问题。

With new zigbee gateway the problem solved，maybe it was related to the
zigbee chip.