Koenkk/zigbee2mqtt #13543: [New device support]: Philips Hue Centura Recessed spot light 929003045401

treb33, 2022-08-14:
### Link

https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-centura-inbouwspot/8719514342927

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x001788010c4bdcee","nwkAddr":2633,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"929003045401","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"929003045401","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20220502","swBuildId":"1.93.11"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20220502","swBuildId":"1.93.11","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1660496800723,"defaultSendRequestWhen":"immediate"}

### Comments

Same light spots as 3 pack https://www.philips-hue.com/nl-nl/p/hue-white-and-color-ambiance-3-pack-centura-inbouwspot/8719514342903.

See https://www.zigbee2mqtt.io/devices/929003045501.html#philips-929003045501 for more info.

treb33, 2022-08-14:
added the device manualy with a external converter file, like https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/externalConvertersExample/light.js

Koenkk, 2022-08-14:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-14 by Koenkk

treb33, 2022-08-14:
That’s fast! Thanks!

treb33, 2022-08-15:
It gives still not supported in dev branch.
Received Zigbee message from '0x001788010c4bdcee', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784402,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0.

Zigbee2MQTT version 1.27.0-dev

treb33, 2022-08-15:
`extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}), `
doesn't work, after rebooting i get the error: hueExtend is not defined

`extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),`
works fine

Koenkk, 2022-08-16:
@treb33 make sure to remove your external converter, can you make a screen of how the device appears in the z2m frontend? Can you also make a screenshot of the "settings -> about" page?

treb33, 2022-08-16:
the external converter was deleted, screenshots with external converter:
https://pasteboard.co/k0JPsTILnvCV.png
https://pasteboard.co/zEYVFifwP55a.png
https://pasteboard.co/y6ykrxdxIqYM.png

Koenkk, 2022-08-16:
This looks good, in your first screenshot you see "Support status: supported". Note that the product image will be added with the next release on 1 September.

treb33, 2022-08-16:
this is with the external converter. But it's not working without the external converter. I will make a screenshot later today