Koenkk/zigbee2mqtt #13547: Not able to control devices suddenly (still able to read their state)

aoberoi, 2022-08-14:
### What happened?

I've suddenly lost the ability to control any of my paired devices. I was away from my home for about a week, so I'm not sure when exactly it occurred, or what might have changed to cause the issue.

I was running v1.24.0 and my devices were working fine (before I left home). Once I returned, I noticed that I wasn't able to control the devices (such as turning on and off lights). I tried restarting the server, which didn't fix things. I noticed that when I used the frontend to load the network map, none of the devices appeared connected. The logs showed messages about failing to load the LQI for every device. When I tried to turn a light off via the Dashboard in the frontend, it failed with the following log output:

```
Debug 2022-08-14 15:16:11Received MQTT message on 'zigbee2mqtt/Kitchen lights/set' with data '{"state":"OFF"}'
Debug 2022-08-14 15:16:11Publishing 'set' 'state' to 'Kitchen lights'
Error 2022-08-14 15:16:31Publish 'set' 'state' to 'Kitchen lights' failed: 'Error: Command 0x5c0272fffece35e8/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
Debug 2022-08-14 15:16:31Error: Command 0x5c0272fffece35e8/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205)) at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:414:23) at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
Info 2022-08-14 15:16:31MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Kitchen lights' failed: 'Error: Command 0x5c0272fffece35e8/1 genOnOff.off({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Kitchen lights"},"type":"zigbee_publish_error"}'
```

However, the current state of each device seems to be accurately shown on the Dashboard (e.g. the 'Kitchen lights' device was correctly showing "on"). Also I have a switch that is bound to a light in a group, and when I used that switch I can see the state of the light change on the Dashboard (and the physical light also changes). It seems, although the server doesn't see any links between itself and the devices (or amongst the devices), that the server is in fact able to read the state of the devices.

I updated to the latest version (v1.27.0) and all the behavior is exactly the same.

### What did you expect to happen?

I expected to be able to control the devices, as I was able to before.

### Zigbee2MQTT version

1.27.0 commit: a9b88082

### Adapter firmware version

20200211

### Adapter

ZigDobby USB dongle

### Debug log

[zigbeemqtt-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9334209/zigbeemqtt-log.txt)

aoberoi, 2022-08-14:
I just realized that the current state is only being read correctly for that one device, 'Dining room light', which belongs to the binding group. For all other devices, the state is not accurate. I know this by toggling the state manually on the device (turning a light on, for one of the lights not in that binding group) and then restarting the server, and the state is not accurate.

Idan37S, 2022-08-15:
Same thing happens to me now.

Idan37S, 2022-08-15:
Tried to downgrade to 1.26.0 or 1.25.2 but the UI didn't load successfully, not sure why.
On 1.27.0 I just can't control devices.

Idan37S, 2022-08-20:
@aoberoi Were you able to figure it out?

aoberoi, 2022-08-28:
Some additional data:

Last weekend I came home, and all my devices we working. I was pleasantly surprised but I didn't make the time to go check the logs to see if there was anything I could trace it back to. I didn't change anything in a week at that point. Things continued to work all week until yesterday. At some point I tried to control one of my lights and noticed that none of the devices we working. It was exactly as I describe in the original description. It's still not working in the same way. I'm not sure what to do but to wait.

Idan37S, 2022-08-28:
I'm not sure if that's what solved my issues, but I got it working again -

Even though I used a USB cable extension with the Conbee 2,
I switched to a longer USB cable extension and also changed the physical location of the stick and now it's working again,
And although I still see the error in the logs occasionally, now everything works and hopefully it won't happen again.

closed 2022-10-06 by github-actions[bot]