Koenkk/zigbee2mqtt #13548: Xiaomi JY-GZ-01AQ `linkage_alarm` not working

Danit2, 2022-08-15:
### What happened?

I have two Xiaomi [JY-GZ-01AQ](https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html#xiaomi-jy-gz-01aq) smoke detectors.

On both i have switch on the function `linkage_alarm` on. 

Now i have tested both of the smoke detectors and the function `linkage_alarm` is not working.
Only on one detectror the alarm comes.

Is this a problem of Z2M or of the firmware from the detector?
Have someone else the same problem?

What i have to do, tht this is working?

Thanks. 

### What did you expect to happen?

When one of the detectros have smoke, that all others also sound the alarm. 

### How to reproduce it (minimal and precise)

Switch the `linkage_alarm` on and simulate smoke. 

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20211217

### Adapter

Sonoff USB Stick CC2652P

Otnow, 2022-08-22:
Yes, the `linkage_alarm` option only sends a notification, but does not trigger an alarm on another similar device, so I created a [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4552) that adds the ability to manually activate/deactivate the buzzer.

Danit2, 2022-08-22:
Cool, thanks for your work. 
I will test it after the next relase.

closed 2022-08-22 by Koenkk

elmasryahmed, 2022-10-13:
@Otnow  How can I use the manual activation to trigger the alarm on other detectors if one detects smoke?!

Otnow, 2022-10-13:
> linkage_alarm Is already enabled in all 3 sensors, but when I tested it it didn’t sound on the other sensors, I’m using HomeKit via HB, can’t find a way to enable alarm sound for other sensors.

As I said earlier, for this you need to create automation in your smart home system:

https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html#buzzer-enum

elmasryahmed, 2022-10-19:
@Otnow I'm using HomeKit, the Buzzer_manual_alarm is not showing to create any automation!!!!

Otnow, 2022-10-19:
> I'm using HomeKit, the Buzzer_manual_alarm is not showing to create any automation!!!!

I don't use HomeKit, so unfortunately I can't help 🤷‍♂️

elmasryahmed, 2022-10-22:
> > I'm using HomeKit, the Buzzer_manual_alarm is not showing to create any automation!!!!
> 
> I don't use HomeKit, so unfortunately I can't help 🤷‍♂️

Thank you for your help, I guess I've to wait for an update that can expose the Manual alarm to Homebridge/Homekit

lost-carrier, 2022-10-28:
@Otnow Thanks for the option to set the buzzer.

However: I wonder what's the purpose of `linkage_alarm` (and `linkage_alarm_state`) then? If it's anyway to be done in NodeRed, one can just as well read `smoke` and trigger `buzzer`, no? So why `linkage_alarm` then?

lost-carrier, 2022-10-28:
@Otnow a maybe worth mentioning experiment I just did: 
- Removed 2 sensors from Zigbee2mqtt and connected to Aqara E1 Hub
- Then turned on the linked alert for both
- Created smoke 
-> 1st sensor raised alert, a few seconds later the linked one also alerted
- Waited air was clean (and "mute" timed out)
- Unplugged Aqara E1 Hub
- Again created smoke
-> 1st sensor raised alert, linked one stayed quiet

I still wonder what's the purpose of linkage_alarm (and linkage_alarm_state), but I guess the original Aqara stuff does the same as NodeRed would do.

...anyone happen to know any Z2M compatiblant smoke sensor that could directly trigger its neighbours? (without any Hub/Zigbee2Mqtt online)

Otnow, 2022-10-28:
@lost-carrier, that's right, unfortunately, these sensors cannot communicate directly with each other and the Aqara hub controls the buzzer.

`linkage_alarm` must be enabled in order to be able to manually control the buzzer, otherwise it and `linkage_alarm_state` are generally not useful, but they are implemented in the code, because they are part of the functionality of the sensor.

lost-carrier, 2022-10-29:
Ah, I see... makes sense (...at least to some extend... ;-) )

Thanks for that explanation.

Danit2, 2022-11-21:
@Otnow 
I got a new smoke alarm.
This has a new firmware. (Firmware date: 20220617)

Is it possible for you to get this firmware so I can update my other smoke alarms?
I would also be interested in the changelog. Perhaps something has changed in the linkage_alarm?

Many Thanks.

Otnow, 2022-11-21:
> Is it possible for you to get this firmware so I can update my other smoke alarms?
> I would also be interested in the changelog. Perhaps something has changed in the linkage_alarm?

The URL of the firmware file can be [retrieved from the Aqara app](https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html), but unfortunately there are no updates available for this sensor at the moment, so all that remains is to wait for it 🤷‍♂️

ciB89, 2023-01-12:
I have created a blueprint in home assistant to achieve this functionality:

https://community.home-assistant.io/t/aqara-smoke-detector-linkage-alarm/517656

Danit2, 2023-01-12:
@ciB89 
Thanks for your blueprint. 
I do this at the moment over Node-Red but i will try your blueprint.
But can you explain a little bit more what you bluepint does? 
The actions that can be triggered? The notification that will come? Some Pictures?

And one thing that i have found now: Why i can only notif one device? i would like to notife my wife also?

ciB89, 2023-01-12:
Once the alarm is triggered (smoke) it will manually set all the alarms on all the connected devices and will send a notification to a mobile device, telling it which room (as long as each device has its area assigned to it) has triggered the alarm and giving the option to deactivate the alarm again (in case of a false positive).

There are two possible triggers, the other one is when the alarm is manually disabled it disables all the alarms on all the other detectors. It's basically another automation, but the way I set it up you only need the one blueprint.

There is only one notify device because you cannot properly use multiple devices here. The better way is to actually create a notify group and use that in the blueprint.

I need to update the description, I know, it's very basic ;)

bdwilton, 2023-05-15:
My Zigbeemqtt has the following 

Firmware build date
20220617
Firmware version
2019www.
Manufacturer
[Xiaomi](https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi)
Model
[JY-GZ-01AQ](https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html#xiaomi-jy-gz-01aq)

I only have one sensor for testing, the test mode triggers correct from HA but manual on does nothing for me.

Turning that on should it actually trigger ?

Cossey, 2023-07-02:
> During the absence of a smoke alarm, the buzzer can be manually alarmed ("alarm") and disalarmed ("mute"), but for this "linkage_alarm" option must be enabled
I can confirm that when `Linkage Alarm` is `OFF`, neither the alarm or mute options will work - but when `ON` they will work. So technically it does have a linkage to other alarms, but the zigbee coordinator must trigger these other smoke alarms. I was also hoping this could directly notify other devices in the case the coordinator goes offline.