Koenkk/zigbee2mqtt #13549: lumi.switch.n0agl1 (SSM-U01) stops reporting power readings

szymucha94, 2022-08-15:
### What happened?

lumi.switch.n0agl1 after pairing with coordinator running under z2m works fine until some other device is added into the network. At this point t1 relay stops reporting power readings. Similar devices like SP-EUC01 report all the values reliably for weeks.

Example of working SSM-U01:
debug 2022-08-15 12:27:01Received MQTT message on 'zigbee2mqtt/0x54ef4410002581cd/set' with data 'ON'
debug 2022-08-15 12:27:01Publishing 'set' 'state' to '0x54ef4410002581cd'
info 2022-08-15 12:27:01MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.14,"current":0,"device_temperature":35,"energy":0.14,"last_seen":1660559222833,"linkquality":220,"power":0,"power_outage_count":1,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"state":null},"update_available":null,"voltage":228.2}'
info 2022-08-15 12:27:01MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.14,"current":0,"device_temperature":35,"energy":0.14,"last_seen":1660559222833,"linkquality":220,"power":0,"power_outage_count":1,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":228.2}'
debug 2022-08-15 12:27:01Received Zigbee message from '0x54ef4410002581cd', type 'attributeReport', cluster 'genOnOff', data '{"245":117440749,"onOff":1}' from endpoint 1 with groupID 0
info 2022-08-15 12:27:01MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.14,"current":0,"device_temperature":35,"energy":0.14,"last_seen":1660559222852,"linkquality":220,"power":0,"power_outage_count":1,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":228.2}'
debug 2022-08-15 12:27:01Received Zigbee message from '0x54ef4410002581cd', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2022-08-15 12:27:01MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.14,"current":0,"device_temperature":35,"energy":0.14,"last_seen":1660559222863,"linkquality":220,"power":0,"power_outage_count":1,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":228.2}'
debug 2022-08-15 12:27:05Received Zigbee message from '0x54ef4410002581cd', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":22.364463806152344}' from endpoint 21 with groupID 0
info 2022-08-15 12:27:05MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.14,"current":0,"device_temperature":35,"energy":0.14,"last_seen":1660559226717,"linkquality":220,"power":22.36,"power_outage_count":1,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":228.2}'

### What did you expect to happen?

I expect z2m to properly process messages coming from SSM-U01 relays so that energy and power readings are reliably passed to HASS.

### How to reproduce it (minimal and precise)

Add SSM-U01 to z2m
Change it's description and home assistant name to more friendly one
Add another zigbee device to network
Restart both z2m and home assistant

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8 (Xiaomi Gateway 3 via ser2net)

### Debug log

debug 2022-08-15 12:25:09Received MQTT message on 'zigbee2mqtt/0x54ef44100025d95a/set' with data 'OFF'
debug 2022-08-15 12:25:09Publishing 'set' 'state' to '0x54ef44100025d95a'
info 2022-08-15 12:25:09MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559111407,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'
info 2022-08-15 12:25:09MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559111407,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'
debug 2022-08-15 12:25:09Received Zigbee message from '0x54ef44100025d95a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info 2022-08-15 12:25:10MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559111461,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'
debug 2022-08-15 12:25:13Received MQTT message on 'zigbee2mqtt/0x54ef44100025d95a/set' with data 'ON'
debug 2022-08-15 12:25:13Publishing 'set' 'state' to '0x54ef44100025d95a'
info 2022-08-15 12:25:13MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559115118,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'
info 2022-08-15 12:25:13MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559115118,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'
debug 2022-08-15 12:25:13Received Zigbee message from '0x54ef44100025d95a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info 2022-08-15 12:25:13MQTT publish: topic 'zigbee2mqtt/0x54ef44100025d95a', payload '{"consumption":0.09,"current":0,"device_temperature":28,"energy":0.09,"last_seen":1660559115136,"linkquality":220,"power":0,"power_outage_count":0,"power_outage_memory":null,"state":"ON","switch_type":null,"update":{"state":null},"update_available":null,"voltage":227.5}'

szymucha94, 2022-08-15:
Checking the value in dev console works.
[image]

Still, HA gets "0W" readings, also 0A current and 0kWh readings despite device working for long hours. Device temperature is reported normally.

donagrejja, 2022-08-15:
Mine doesn't report any power reading from initial pairing. I just tried to pair it yesterday though so it's recently added to the network.
Temperature, linkquality, switch mode, power outage memory and turning it on and off works.
But everything else is "none or n/a".

donagrejja, 2022-08-15:
There's an OTA available for this device. I tried updating it last night but still the same.

szymucha94, 2022-08-15:
these values are "none or n/a" in HA or z2m "exposes" tab?

donagrejja, 2022-08-15:
[image]

szymucha94, 2022-08-15:
Another relay just stopped reporting power. I didn't even touch it and I was out of home.
There are some automations that turn the relay on and off periodically but nothing that changes any settings. Please urgently fix this.

donagrejja, 2022-08-15:
I've added 2 more relays that are functioning without issue so my comment might have been a broken relay.

szymucha94, 2022-08-15:
This is not a broken relay (assuming you're responding to my issue) This applies to 3 separate relays under two different firmwares (I upgraded all of them to the fresh FW from dec 2021 because of this issue - zero improvement). And they work just fine when gateway is running in lumi mode instead of z2m. So this is clearly z2m issue.
Also please note that I am able to get power readings from dev console. For example:
debug 2022-08-16 00:57:26Received Zigbee message from '0x54ef4410002581cd', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"**activePower":212**}' from endpoint 1 with groupID 0
info 2022-08-16 00:57:26MQTT publish: topic 'zigbee2mqtt/0x54ef4410002581cd', payload '{"consumption":0.16,"current":0,"device_temperature":36,"energy":0.16,"last_seen":1660604246525,"linkquality":220,"**power":0**,"power_outage_count":1,"power_outage_memory":false,"state":"ON","switch_type":"toggle","update":{"state":null},"update_available":null,"voltage":227.2}'

donagrejja, 2022-08-15:
Yep. First I thought that my issue was related but since my other 2 relays are working fine I thought that I would inform you that my initial conclusion that I was affected by the same issue was wrong.

szymucha94, 2022-08-15:
Sorry, I removed my comment because I totally misread yours. For some reason I thought someone else responded in this ticket :P

szymucha94, 2022-08-19:
I'm sorry, z2m is totally unusable with this gateway with more than 50 devices. I've decided to stop using z2m and fallback to lumi implementation of mqtt within the gateway. z2m causes very strange routing rules (for example almost all endpoints are connected to gateway despite chipset limit of <=32endpoints while for all of them at least 3 routers are available - under lumi mqtt maybe 4 out of 39 devices are connected directly). At least once every 3 days I'm forced to repair devices, z2m fails to handle power readings from relays, OTA checks notoriously fails, etc. This may be related to aqara not following zigbee standard properly.

closed 2022-08-19 by szymucha94