Koenkk/zigbee2mqtt #13552: Zigbee2mqtt HA addon is not working on fresh HA install, RPi3 and RPi4 with Conbee 2

outlying, 2022-08-15:
### What happened?

On fresh HA OS instance I installed zigbee2mqtt addon, with standard mosquitto broker installed before that, and zigbee2mqtt is not running due to "Error while starting zigbee-herdsman"

### What did you expect to happen?

Work

### How to reproduce it (minimal and precise)

Get RPi 3/4
Install fresh HA OS (now latest is 8.4)
Do basic setup
Install mosquitto broker
Install zigbee2mqtt using provided guide

### Zigbee2MQTT version

1.27.0-1

### Adapter firmware version

 deCONZ_ConBeeII_0x26780700

### Adapter

Conbee 2

### Debug log

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[13:33:59] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[13:34:00] INFO: Zigbee Herdsman debug logging enabled
[13:34:02] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:debug 2022-08-15 13:34:08: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  2022-08-15 13:34:08: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-15.13-34-08' filename: log.txt
Zigbee2MQTT:debug 2022-08-15 13:34:08: Removing old log directory '/config/zigbee2mqtt/log/2022-08-15.13-30-57'
Zigbee2MQTT:info  2022-08-15 13:34:09: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-15 13:34:09: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:debug 2022-08-15 13:34:09: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2252655-if00"}}'
2022-08-15T11:34:09.444Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2022-08-15T11:34:09.449Z zigbee-herdsman:deconz:driver Set delay to 0
2022-08-15T11:34:09.450Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2252655-if00","adapter":"deconz"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-08-15T11:34:09.452Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2252655-if00
2022-08-15T11:34:09.469Z zigbee-herdsman:deconz:driver Serialport opened
2022-08-15T11:34:09.471Z zigbee-herdsman:controller:log Started with result 'resumed'
2022-08-15T11:34:09.472Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2022-08-15T11:34:09.487Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2022-08-15T11:34:12.452Z zigbee-herdsman:deconz:adapter checking coordinator simple descriptor
2022-08-15T11:34:12.462Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x0 EP:0 SeqNr. 2 request id: 1
2022-08-15T11:34:19.489Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xa seqNr: 1
Zigbee2MQTT:error 2022-08-15 13:34:19: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-15 13:34:19: Failed to start zigbee
Zigbee2MQTT:error 2022-08-15 13:34:19: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-15 13:34:19: Exiting...
Zigbee2MQTT:error 2022-08-15 13:34:19: undefined
2022-08-15T11:34:19.519Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 3
[13:34:19] INFO: Zigbee Herdsman debug logging enabled

outlying, 2022-08-15:
What I already did:
- install it on RPi4 and RPi3, problem is on both
- use USB extension cable
- check that Conbee 2 is connected to USB 2.0 port
- use different USB ports
- restart devices
- unplug and plug Conbee 2 while RPi is running

I moved to new apartment and I wanted to start from scratch, just a month ago I was running HA on RPi3 with zigbee2mqtt and the same Conbee 2 adapter without any issues, now I cannot make it work on fresh HA instance

outlying, 2022-08-15:
So one of those rare working moments occurred - but as soon as I tried to add a device whole zigbee2mqtt stopped working

Log

```
2022-08-15T17:10:06.708Z zigbee-herdsman:deconz:driver query aps data indication
2022-08-15T17:10:06.715Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 54
2022-08-15T17:10:06.716Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 54 srcAddr: 0xe93b destAddr: 0x0 profile id: 0x104 cluster id: 0x19 lqi: 255
2022-08-15T17:10:06.717Z zigbee-herdsman:deconz:frameParser response payload: 1,136,1,0,95,17,28,21,22,11,0,0
2022-08-15T17:10:06.721Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":136,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4447,"imageType":5404,"fileVersion":2838},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":59707,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-08-15T17:10:06.721Z zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '59707', skipping...
2022-08-15T17:10:06.722Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-08-15T17:10:14.229Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0xb seqNr: 50
2022-08-15T17:10:14.230Z zigbee-herdsman:deconz:driver too many timeouts - restart serial connecion
2022-08-15T17:10:14.231Z zigbee-herdsman:deconz:driver Opening with /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2252655-if00
UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "TIMEOUT".
[19:10:14] INFO: Zigbee Herdsman debug logging enabled
[19:10:16] INFO: Handing over control to Zigbee2MQTT Core ...
```

peteh, 2022-08-15:
Just a far fetch, Is your power supply for the pi itself stable or do you get power warnings?

outlying, 2022-08-16:
@peteh I think I checked for warnings on RPi4 and I've seen no warning, power supply was bought as dedicated for RPi4, with 3.1A if I remember correctly

Still, I have the same issue on RPi3 - different power supply - but I didn't check for power issues, since they are usually typical for RPi4

outlying, 2022-08-18:
So I decided to go to extremes and buy 2nd Conbee 2 ... which is also not working on fresh install, with same errors

outlying, 2022-08-18:
OK, I was wrong, error remained for the while but it seems that things are working now, however, HA version is not `8.6` so I'm unsure how this affected addon

outlying, 2022-08-18:
Not working again, I was able to setup NodeRed with few automation but it seems now it's again in an endless loop of not working

peteh, 2022-08-20:
Does the same happen when you install the docker versions of zigbee2mqtt and homeassistant on clean raspberrypi os?

outlying, 2022-08-22:
@peteh I'm not doing any docker, just straight HA OS + Z2M addon

I see that this issue is actually happening all over the Internet, not a single person gets real help, this issue kinda disappeared in each case I read.

For the last 3 days, Z2M kinda worked ... as long as I didn't touch it much, failed on an occasion or two, I had to restart it a few times to get it work but it is hardly a fix.

Started happening when I tried to connect 2nd Philips bulb and now it's down again.

> It might also be interferences, have a look at the last post in this thread: https://community.home-assistant.io/t/zigbee2mqtt-addon-not-connecting-to-conbee-ii-usb-stick/360616/9

Like I stated above, extension cable, FW upgrades, USB port switched, etc. ... on the new Conbee 2 I didn't upgrade FW, because it worked! for last 3 days, I didn't move it an inch and now while connecting another bulb it failed.

So my observation is:
- different RPi (3 and 4), same problem
- new Conbee 2, same problem
- typical issues, extension cables, USB ports verified

Only the software remains the same, the logs are useless, and I read through other threads and the logs are the same, ppl have no idea why it is happening, see:

https://github.com/Koenkk/zigbee2mqtt/issues/11075 

This is annoying, I wish someone would address this issue properly, and add some extra logs to the code so we could actually figure out why is it failing.

peteh, 2022-08-22:
Well, from my point you have several options: 
 - downgrade to a version that doesn't have these problems (not sure how to do that, would be easy with docker) and find out from which version the issues start occurring to pinpoint down to change that causes the issue as you said your setup was working previously
 - Try HA's own zigbee addon ZHA to see if the same issues occur there (would exclude hardware as a problem)
 -  Try another stick to exclude conbee itself as the source, e.g. due to unknown incompatibility with your devices
 - Try different channels in zigbee2mqtt: https://home-assistant-guide.com/2020/10/29/choose-your-zigbee-channel-wisely/

outlying, 2022-08-22:
For the record (for anyone who found this thread):

> Try HA's own Zigbee addon ZHA to see if the same issues occur there (would exclude hardware as a problem)

This is excluded, I tried desktop deConz software (Windows), and the stick worked right away

> Try different channels in zigbee2mqtt: https://home-assistant-guide.com/2020/10/29/choose-your-zigbee-channel-wisely/

I already reconfigured Wi-Fi so my Zigbee network is using default channel 11 and Wi-Fi is using its channel 11, on top of extension cable ofc. FYI those channels are not colliding (same channel number but different frequency)

And for the record and reminder, I was able to use my first Conbee 2 adapter successfully for 2+ years straight without such an issue on my previous HA instance, I updated software from time to time but never had this issue before I started with fresh HA install.

I strongly believe It is not a hardware issue

peteh, 2022-08-22:
Then go the software route and downgrade till you find a version that works and then people can look in what changed to the next version when it stopped working.

outlying, 2022-08-22:
@peteh the only problem I have with this approach is I won't have the same environment as I have now, now it is HA OS + Z2M addon, if I go Docker (on the same RPi4) then it probably will be Raspbian running HA + Z2M on Docker, it is not the same as what I have now.

I'm not saying I won't do it, I'm just saying I might not get the same result, and even if I get the same result it is easy to contest the result.

Unluckily I don't see a way to downgrade the addon, I guess it's strongly dependent on the HA OS version, so Docker seems like the only option

outlying, 2022-08-22:
For the record:

A similar thread mentions that 1.17.1 worked while 1.18.1

https://github.com/Koenkk/zigbee2mqtt/issues/5854#issuecomment-788337825

Not the same setup but might the moment when this became a problem

outlying, 2022-08-23:
In the morning (shortly after 07:00) things got shortly back to normal, I was able to add that 2nd bulb, I thought that maybe things are working again (spoiler, they don't), but I noticed some CPU temperature ramping up (same 07:00). At `07:48:28` my ZigBee devices became unavailable. OFC temperature dropped because Z2M stopped working properly

I'm using a Flirc aluminum case, I bought it 2 years ago, just in case I'm going to replace thermal pads to see if maybe the issue is somehow temperature related - but the temperature is way under thermal throttling (around 80C)

[image]

outlying, 2022-08-30:
I shamefully have to admit that the problem was that my HA instance configured ZHA during setup, I never used ZHA I had no idea that I'm installing it since I did not recognize the icon, I simply clicked next to get to the forms I had to fill and then start using HA, installing Zigbee2MQTT, etc. in my mind this icon represented some sort of "driver" required by the coordinator, and since my RPi4 had Conbee 2 connected during setup I never bothered checking what this thing actually is.

closed 2022-08-30 by outlying

peteh, 2022-08-30:
No worries, that means both zha and zigbee2mqtt sent commands to the coordinator and therefore commands are either answered by one or the other extension which caused your timeouts.