Koenkk/zigbee2mqtt #13553: Sonoff Router Firmware Version not Updating

sumur1au, 2022-08-15:
### What happened?

I am trying to update my router firmware from 20210128 to 20220125 on my Sonoff dongle using router [link](https://github.com/Koenkk/Z-Stack-firmware/raw/master/router/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_router_20220125.zip) but the firmware version in zigbee2mqtt is still the old version.  I am using the Smart RF Flash Programmer 2 and I get a success message after the end of the firmware upgrade. 
[image]

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20210128

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

Koenkk, 2022-08-16:
Since the device has been updated out of z2m (not via an OTA) z2m will not reread the firmware version. You do a manual update via the z2m frontend -> click on your device -> dev console and read the `dateCode`:

[image]

After that restart z2m.

closed 2022-08-16 by Koenkk

sumur1au, 2022-08-17:
Thanks @Koenkk