Koenkk/zigbee2mqtt #13555: Zigbee2MQTT fails to start install docker-compose

gurues, 2022-08-15:
### What happened?

Hi, I try to install zigbee2mqtt with docker but it won't start. I have read and applied Zigbee2MQTT fails to start but still the same. Home assistant autodetect sonoff zigbee 3.0 usb in /dev/ttyUSB0 to install ZHA. I have not installed ZHA.

Thanks for your help I'm desperate, I've read everything I've found about this problem but I'm not able to fix it

Path Adapter:
`gurues@domo-HA:~/domotica/zigbee2mqtt$ ls -l /dev/serial/by-id`
`total 0
lrwxrwxrwx 1 root root 13 ago 14 19:56 usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_28471632d69cec11a6d4614d73138bba-if00-port0 -> ../../ttyUSB0`

`gurues@domo-HA:/dev$ ls -ls ttyUSB0`
`0 crw-rw---- 1 gurues dialout 188, 0 ago 14 19:56 ttyUSB0
`

Zigbee2MQTT fails to start (response -> success):
`gurues@domo-HA:~/domotica/zigbee2mqtt$ test -w /dev/ttyUSB0 && echo success || echo failure`
`success`

Permissions and user groups "gurues":
`gurues@domo-HA:~/domotica/zigbee2mqtt$ id
uid=1000(gurues) gid=1000(gurues) grupos=1000(gurues),5(tty),10(uucp),20(dialout),24(cdrom),25(floppy),27(sudo),29(audio),30(dip),44(video),46(plugdev),101(systemd-journal),108(netdev),113(bluetooth),118(lpadmin),121(scanner),998(docker)`

Pach install:
`gurues@domo-HA:~/domotica/zigbee2mqtt$ ls`
`data`
`gurues@domo-HA:~/domotica/zigbee2mqtt$ cd data`
`gurues@domo-HA:~/domotica/zigbee2mqtt/data$ ls`
`configuration.yaml  log`

configuratiión.yaml:
`homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  user: *****
  password: ******
  server: mqtt://192.168.1.xxx:1883
  version: 5
serial:
  port: /dev/ttyUSB0
frontend:
  port: 8099`

docker-compose
`version: '3.8'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    group_add:
      - dialout
    user: 1000:1000
    volumes:
      - /home/gurues/domotica/zigbee2mqtt/data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8099:8080
    environment:
      - TZ=Europe/Madrid
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyUSB0:/dev/ttyACM0`

### Zigbee2MQTT version

1.27.0

[... 85 lines cut ...]
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-08-15 17:04:05: Logging to console and directory: '/app/data/log/2022-08-15.17-04-05' filename: log.txt
Zigbee2MQTT:info  2022-08-15 17:04:05: Starting Zigbee2MQTT version 1.27.0 (commit #a9b8808)
Zigbee2MQTT:info  2022-08-15 17:04:05: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:error 2022-08-15 17:04:06: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-15 17:04:06: Failed to start zigbee
Zigbee2MQTT:error 2022-08-15 17:04:06: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-15 17:04:06: Exiting...
Zigbee2MQTT:error 2022-08-15 17:04:06: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-08-15 17:05:07: Logging to console and directory: '/app/data/log/2022-08-15.17-05-07' filename: log.txt
Zigbee2MQTT:info  2022-08-15 17:05:07: Starting Zigbee2MQTT version 1.27.0 (commit #a9b8808)
Zigbee2MQTT:info  2022-08-15 17:05:07: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:error 2022-08-15 17:05:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-08-15 17:05:07: Failed to start zigbee
Zigbee2MQTT:error 2022-08-15 17:05:07: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-08-15 17:05:07: Exiting...
Zigbee2MQTT:error 2022-08-15 17:05:07: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12

comleoita, 2022-08-15:
I've got the same problem.

peteh, 2022-08-15:
Your debug log says it cannot find ttyUSB0 which is mapped to ttyACM0 on the host. 

Which makes sense as you stated your adapter identifies as ttyUSB0 also on the host. Thus fixing your docker compose file for the device mapping like this should work: 

```
    devices:
      # Make sure this matched your adapter location
      - /dev/ttyUSB0:/dev/ttyUSB0
```

gurues, 2022-08-15:
Thanks a million peteh, it is solved and zigbee2mqtt is working.

closed 2022-08-16 by Koenkk